maniacx/Battery-Health-Charging #3: Battery Health Service sometimes stops working after startup

yukina3230, 2023-02-11:
Thank you for your work, the extension worked fine on my asus laptop. But sometimes when I startup, it shows this error and I have to reinstall the service.

[image]

maniacx, 2023-02-13:
It is possible that the kernel is changing the permission of the file, after the service has changed it.
Anyways I wont be using systemd services any more on the next update
I am going to implement polkit. but its going to take somedays.

yukina3230, 2023-02-14:
Thank you, looking forward for the next update.

maniacx, 2023-02-18:
Hello Yukina.

Can you test the extension updloaded in the github main branch.

1. Use git clone or Download and extract the zip using code button (green color) on the website 
2. go to the extracted folder and right click and **open in terminal**
3. Install the extension using command **./install.sh**
4. logout after installing and login.
5. Only if you have not remove the earlier version3 extension the extension will automatically detect it and prompt you to remove the service files and it will ask you to reboot the system.
6. extension will ask you to install polkit since you have Asus laptop and need root privileges. 
7. After installing polkit script logout/login again . 

**Test and Report back please**

Thank you.
Mac

yukina3230, 2023-02-18:
Thanks, I will test it and report back soon.

yukina3230, 2023-02-18:
Hi Mac.

Thank you for the update

- I have installed the extension using "install.sh" and logout/login again.
- Then I install polkit via Extention Setting, it prompts me to type in root password and restart the system.
- After that the extension kept asking me to install polkit, even though i logged out/login several times, rebooting didn't work.

maniacx, 2023-02-18:
I usually tested this on my system, which is ubuntu 22.10 and it works.
I just realised that ubuntu uses an older version of polkit (which the script detects as legacy).
In legacy polkit the /tool/installer.sh "check" command doesnt need root privileges.

I install the latest manjaro OS, which uses the lastest version of polkit.
It seems like the new version of polkit, /tool/installer.sh "check" command needs privilege access. (denied access to "/etc/polkit-1/rules.d" without root privileged.)
So the /tool/installer.sh "check" command  fails to check if polkit is installed or not, stopping the extension from initializing.

I already have a solution in mind, but need some time to implement and properly test it.

Just for info sake
Need to know which OS are you running?
And what is the polkit version it is using?
to check polkit version in terminal
`pkaction --version`

Thanks for testing.

yukina3230, 2023-02-18:
Hi, I'm using Arch Linux, and my polkit version is 122.

Thank you.

maniacx, 2023-02-19:
Hey Yukina.
I have updated and test on both systems. Ubuntu (pkversion = 0.105) and Majora (pkversion = 0.122) and it is working without any issue.
Please download and test again and report please.

yukina3230, 2023-02-20:
Hi Mac.

I have reinstalled it, the extension worked fine without any errors. But I think I have to wait a few days to see if it is stable or not, then I will report back to you here.

Thank you.

maniacx, 2023-02-20:
Thank you for testing it. 
And report back if there are any issues

maniacx, 2023-02-20:
> Thank you for testing it. 
> And report back if there are any issues 

Also I just push so more changes an hour ago.
Last commit 9a9d6ce91030ee5517b04f8a54a19221a5f835f2

Hopefully you are testing the latest upload.

yukina3230, 2023-02-20:
Thank you, I just updated to the latest commit, no issues so far.

maniacx, 2023-02-20:
Thank you Yukina. I will close this issue.

yukina3230, 2023-02-21:
Hi Mac.
Today the extension stopped working for me, it keeps returning back to version 3, even if I reinstall it again using git clone.

maniacx, 2023-02-22:
Yes.  It happened to me as well. That is because the extension manager keeps updating to the version that is uploaded on gnome extension website which is still version 3

If you want to use the latest version temporary until I upload version 4 on gnome extensions, you can rename it to version 3 again. 
In the downloaded/extracted folder, before installing. Open the file **metadata.json** and the line that says "    **"version": 4,** change it to 3.. it should look like   **"version": 3,** . and then install the extension.

maniacx, 2023-02-24:
I have uploaded the extension at gnome extension website.
You extesnion should automatically update.
Let me know if you faced any issue. I will close this issue for now.

closed 2023-02-24 by maniacx