maniacx/Battery-Health-Charging #8: Charging limit current status and notification is incorrect (Fedora 37, Acer Swift X SFX14-51G)

anzigo, 2023-03-16:
The extension's functionality is working as it should. However, the "Charging limit is set to 80%" notification remains even when  I've switched back to Full Capacity Mode.

maniacx, 2023-03-16:
I believe you are using this "https://github.com/frederik-h/acer-wmi-battery" with your laptop.
It has an issue with reading threshold from sysfs which this extension depends on.

However I have already created a patch for this, to fix reading from sysfs.
 https://github.com/maniacx/acer-wmi-battery/commit/4747abd90d1e8135a87302bd91177ab2431eec89

A user asant confirmed the patch is working. See discussion here.
https://github.com/frederik-h/acer-wmi-battery/issues/14#issuecomment-1465337052

I have already asked @frederik-h to merge this fix but he seems to be away and not reponding.

anzigo, 2023-03-16:
@maniacx Thanks for the update/explanation. I'll close this issue and wait for the update to be pushed to the acer-wmi-battery code.

maniacx, 2023-03-16:
Alternatively you could try the patch

https://github.com/maniacx/acer-wmi-battery

Its a fork of fredirick's repo with the patch.
But at your own risk. :)

anzigo, 2023-03-16:
I tested the updated acer-wmi-battery code and it works as expected. Thanks again!

maniacx, 2023-03-26:
Hello anzigo. Looking for some testing of new version if interest please contribute

Link here https://github.com/maniacx/Battery-Health-Charging/issues/11

closed 2023-03-27 by maniacx