maniacx/Battery-Health-Charging #9: After the update does not work with Lenovo ThinkPad T480s any longer

kir-93, 2023-03-18:
Msg: Unsupported device. This extention is not compatible  with your device

maniacx, 2023-03-18:
Sorry about that.
Could be the additional check I introduced distinguish between devices. I guess not all thinkpad have it.
It checks if the file/folder exist in sysfs `/sys/devices/platform/smapi`
Can you check if this path exist on your device from your terminal?
`ls -l /sys/devices/platform/smapi`

maniacx, 2023-03-18:
Additional check for this too
`ls -l /sys/class/power_supply/BAT0/charge_control_end_threshold`
`ls -l /sys/class/power_supply/BAT0/charge_control_start_threshold`
`ls -l /sys/class/power_supply/BAT1/charge_control_end_threshold`
`ls -l /sys/class/power_supply/BAT1/charge_control_start_threshold`

maniacx, 2023-03-18:
Could you try, the new updated version 9 on gnome extensions?

kir-93, 2023-03-18:
> Could you try, the new updated version 9 on gnome extensions?

tried a new version ... seem to work + new threshold ranges have been introduced ) nice!! All works for me again.
would it still help if i make checks proposed above?

maniacx, 2023-03-18:
Yes. If possible, make the checks. Just curious why version 8 didnt work.

kir-93, 2023-03-18:
ls -l /sys/devices/platform/smapi
ls: cannot access '/sys/devices/platform/smapi': No such file or directory

$  ls -l /sys/class/power_supply/BAT0/charge_control_end_threshold
-rw-r--r--. 1 root root 4096 18. Mär 22:18 /sys/class/power_supply/BAT0/charge_control_end_threshold

$  ls -l /sys/class/power_supply/BAT0/charge_control_start_threshold
-rw-r--r--. 1 root root 4096 18. Mär 22:18 /sys/class/power_supply/BAT0/charge_control_start_threshold

$  sudo ls -l /sys/class/power_supply/BAT1/charge_control_end_threshold
ls: cannot access '/sys/class/power_supply/BAT1/charge_control_end_threshold': No such file or directory

$  ls -l /sys/class/power_supply/BAT1/charge_control_start_threshold
ls: cannot access '/sys/class/power_supply/BAT1/charge_control_start_threshold': No such file or directory

maniacx, 2023-03-18:
Yes if failed because of this.
`ls: cannot access '/sys/devices/platform/smapi': No such file or directory`
Thanks for reporting it. I will close this issue.

maniacx, 2023-03-18:
Just need one more check from you. If in future , some other device conflict with thinkpad and I need to implement vendor check back again, then this will be helpful to me.
Check the result of this
`ls -l /sys/devices/platform/thinkpad_acpi`

maniacx, 2023-03-18:
Just need one more check from you. If in future , some other device conflict with thinkpad and I need to implement vendor check back again, then this will be helpful to me.
Check the result of this
ls -l /sys/devices/platform/thinkpad_acpi

kir-93, 2023-03-18:
here it is:

ls -l /sys/devices/platform/thinkpad_acpi
total 0
-rw-r--r--.  1 root root 4096 19. Mär 00:30 bluetooth_enable
--w-------.  1 root root 4096 19. Mär 00:30 cmos_command
lrwxrwxrwx.  1 root root    0 19. Mär 00:27 driver -> ../../../bus/platform/drivers/thinkpad_acpi
-rw-r--r--.  1 root root 4096 19. Mär 00:30 driver_override
-r--r--r--.  1 root root 4096 19. Mär 00:30 hotkey_adaptive_all_mask
-r--r--r--.  1 root root 4096 19. Mär 00:30 hotkey_all_mask
-r--r--r--.  1 root root 4096 19. Mär 00:30 hotkey_bios_enabled
-r--r--r--.  1 root root 4096 19. Mär 00:30 hotkey_bios_mask
-rw-r--r--.  1 root root 4096 19. Mär 00:30 hotkey_enable
-rw-r--r--.  1 root root 4096 19. Mär 00:30 hotkey_mask
-rw-r--r--.  1 root root 4096 19. Mär 00:30 hotkey_poll_freq
-r--r--r--.  1 root root 4096 19. Mär 00:30 hotkey_radio_sw
-r--r--r--.  1 root root 4096 19. Mär 00:30 hotkey_recommended_mask
-rw-r--r--.  1 root root 4096 19. Mär 00:30 hotkey_source_mask
drwxr-xr-x.  3 root root    0 19. Mär 00:27 input
drwxr-xr-x. 10 root root    0 19. Mär 00:27 leds
-r--r--r--.  1 root root 4096 19. Mär 00:30 modalias
drwxr-xr-x.  2 root root    0 19. Mär 00:30 power
drwxr-xr-x.  3 root root    0 19. Mär 00:27 rfkill
lrwxrwxrwx.  1 root root    0 19. Mär 00:27 subsystem -> ../../../bus/platform
-rw-r--r--.  1 root root 4096 19. Mär 00:27 uevent
-r--r--r--.  1 root root 4096 19. Mär 00:30 wakeup_hotunplug_complete
-r--r--r--.  1 root root 4096 19. Mär 00:30 wakeup_reason

maniacx, 2023-03-18:
Great thank you for the info. If needed I may use it next time.

maniacx, 2023-03-26:
Hello kir-93. Looking for some testing of new version if interest please contribute

Link here https://github.com/maniacx/Battery-Health-Charging/issues/11

closed 2023-03-28 by maniacx