maniacx/Battery-Health-Charging #11: Test: New version test before launcher

maniacx, 2023-03-26:
Created this issue for testing newer version before uploading on extension.
Share your result and your opinion here

maniacx, 2023-03-26:
[test_version10.zip](https://github.com/maniacx/Battery-Health-Charging/files/11073294/test_version10.zip)

anzigo, 2023-03-27:
The test version is working for me. I ran the reset and install scripts. Successfully enabled and disabled Maximum Lifespan Mode and Full Capacity Mode without issue.

monethass, 2023-03-27:
Confirm working on Dell

mascherm, 2023-03-27:
test_version10 works fine.

mascherm, 2023-03-29:
i get the same results as you, in case 1 it shows charging and in case 2 it shows not charging. 

i have a question, what exactly is the use case of the start charging treshold?

mascherm, 2023-03-31:
thanks for the explaination!

maniacx, 2023-04-02:
Hello @mascherm @monethass @anzigo 
Download [test_version_11.zip](https://github.com/maniacx/Battery-Health-Charging/files/11130865/test_version_11.zip) and test please.

Changes:
1. Added an option that changes system icon behavior as explained in the post https://github.com/maniacx/Battery-Health-Charging/issues/11#issuecomment-1489200854

Currently when charger is powered and plugged-in, if the threshold is active and battery reaches threshold limit, the battery indicator icon switches to **on-battery/charger-unplugged**.
This for me, creates a problem as I do not know if the charger is powered and plugged-in or not.
Dont know if  **on-battery/charger-unplugged** is displayed because threshold is reached or charger is unplugged or powered off.

That why i introduced this option.
With this option enabled, the battery indicator will display **charging** icon and never display  **on-battery/charger-unplugged** icon with charger powered and plugged-in. Similar behaviour as in Windows OS.
This way we know that if charger is power and plugged-in or not.
[image]

2. This is for users with custom threshold. Added addition text (currently active) showing the current mode, as shown in pic.  Depending on which mode is selected this text will be displayed besides the mode indicating current mode.
Although it is informative, in my opinion doesn't look fitting. Please test and let me know if I should keep this option or remove it.
[image]

monethass, 2023-04-02:
Just tried on Dell, and both feature worked as expected.

However, a feedback on the indicator icon, is that if it's enabled, it can't be distinguished between the two state of "charging" or "holding". For example, I set my threshold to 60/80. Now I see my battery is at 70%, and having the charging icon, but I do not know if it's at the cycle of charging from 60% to 80%, or it already has been charged to 80%, and currently it's discharging from 80% to 60% because my laptop is using more power than the power brick can supply.

A suggestion is that to use this extension's icon to distinguish between those two states instead of changing the original battery icon.

maniacx, 2023-04-02:
Like the idea. Initially i thought of using extensions icon to display the charging status instead of the system icon. But no space on the icon (it is a very small icon) and I ran out of ideas if how to modify the icon to provide more information about charging and holding (unless I add another icon which I dont want to).  One way of implementing was to change to color the icon, but it look bad and inconsistent as rest of the system icons are symbolic.
But i will give it another shot, and see if i can come up with something for the next  release.

ViBE-HU, 2023-04-17:
could you please add more gnome version support?

maniacx, 2023-04-17:
Gnome 43 and 44 are supported. Which Gnome version are you using?

maniacx, 2023-04-17:
@ViBE-HU I had a look to support gnome 42 and the changes are massive. I don't think can support any gnome versions earlier than 43.

ViBE-HU, 2023-04-21:
@maniacx good to know. bad news for compatibility cause bunch of major distros still on 42x.

maniacx, 2023-04-21:
No confirming, but I will give it a go when I have free time.

ViBE-HU, 2023-04-21:
that would be cool. translation is much more easier if i can test the extension.

maniacx, 2023-04-21:
> that would be cool. translation is much more easier if i can test the extension.

Can you please give me the exact gnome version used on pop os? I can see that latest release of gnome42 is 42.9. Is pop-os using 42.9 or an older version of 42?

ViBE-HU, 2023-04-21:
i did not mention pop os. at the moment i use ubuntu 20.04 and 22.04 machines.

22.04 has gnome 42.5

ViBE-HU, 2023-04-21:
meanwhile i created a pop os 22.04 vm if the information helps. most ubuntu based distros are use the same update schemes and no surprise: it is using gnome 42.5 too.
[image]

maniacx, 2023-04-23:
Hey @ViBE-HU
I have ported the extension for Gnome42.
But because of some string dont fit on Gnome42 quick panel, I had create a seperate extension. I have tested for Asus, but didnt test for other device. Please install an test it.

https://github.com/maniacx/Battery-Health-Charging-Gnome42

Let me know how it goes.

ViBE-HU, 2023-04-23:
@maniacx here are the results.

the machine is a Dell Latitude E5470 with ubuntu 22.04:
[image]

and i also tested the created pop os vm. the host machine is a Lenovo Legion Y520-15IKBN and as far as i know VirtualBox reports battery natively for the guest system. but maybe i'm wrong and MAY the results are useless:

> ./install.sh
> Packing extension...
>  
> (gnome-extensions pack:6930): GLib-GIO-CRITICAL **: 15:13:27.906: g_subprocess_wait: assertion 'G_IS_SUBPROCESS (subprocess)' failed
> Failed to execute child process ?msgfmt? (No such file or directory)
> Installing extension...
> Error opening file /home/user/Downloads/Battery-Health-Charging-Gnome42-main/Battery-Health-Charging-Gnome42@maniacx.github.com.shell-extension.zip: No such file or directory
> Extension installed Battery Health Charging succesfully. Restart the shell (or logout) to be able to enable the extension.

maniacx, 2023-04-23:
**_the machine is a Dell Latitude E5470 with ubuntu 22.04:_**
What command in terminal are you using to change battery threshold? 
For Dell you need to install additional libraries.
Are you using **libsmbios** or **cctk**? 
If you are using **cctk**, please let me know. 
If you havent install **libsmbios** or **cctk**. Install **libsmbios** (Ubuntu **smbios-utils**)

You can install libsmbios battery module using
 `sudo apt install smbios-utils`

Then try changing threshold values using terminal
`sudo smbios-battery-ctl --set-charging-mode='custom'`
`sudo smbios-battery-ctl --set-custom-charge-interval=55 60`

If changing threshold works without errors, you will see results.
`Custom charge interval has been set to (55, 60)`

If this works, you can try changing threshold using extension.

-----------------------------------------------------------------------------------------------

 _**Failed to execute child process ?msgfmt? (No such file or directory)**_ For lenevo
Install gettext. Then you will be able to install the extension.
`sudo apt install gettext`

--------------------------------------------------------------------------------------------------

**Lenovo Legion Y520-15IKBN virtual box**

If VM supports change threshold, you should be able to change threshold using command line.
`echo "1" | sudo tee /sys/bus/platform/drivers/ideapad_acpi/VPC2004:00/conservation_mode`

I do not think VM supports changing threshold. Send me the results of this command
`ls /sys/bus/platform/drivers/ideapad_acpi/VPC2004:00/conservation_mode`

maniacx, 2023-04-23:
@ViBE-HU 
I have decided to include Gnome42 in this extension and not create a seperate version.
So please remove the **Battery-Health-Charging-Gnome42** extension from extension manager.
and run the following command to remove all settings.
`gsettings --schemadir /home/$USER/.local/share/gnome-shell/extensions/Battery-Health-Charging-Gnome42@maniacx.github.com/schemas reset-recursively org.gnome.shell.extensions.Battery-Health-Charging-Gnome42`

And install **Battery-Health-Charging**

maniacx, 2023-04-26:
Hello @ViBE-HU . Any luck?
Are you using cctk by any chance?

ViBE-HU, 2023-04-27:
@maniacx no worries. i did not forget the project. but im a bit busy now and need a little time to test a few machines. i also want to test a dual battery machine to translate properly the related strings.

so the dell with ubuntu  22.04 is tested. you was right. i had to install `smbios-utils` first. then the plugin installation went well. polkit added successfully. looks like the plugin working well.

and i have a question. i explain it first cause i'm not a developer. i'm a dual boot user and i have windows too on most machines. on windows you can use Dell Power Manager to manage battery/charging. this program controls U/EFI(?) directly. but as i understand your plugin does not and it stores the configuration locally. so if i change it on linux it will won't affect permanently. am i right? if yes, is it possible to control the same way as windows does?

maniacx, 2023-04-27:
When you set the threshold using this extension, the only think the extension does is executes 1 or both commands below depending upon the mode. Also on every reboot to linux it will enable the previous settings
```
sudo smbios-battery-ctl --set-charging-mode='mode'
sudo smbios-battery-ctl --set-custom-charge-interval=low high
```
Rest everything is done by libsmbiosutils (using smbios-battery-ctl) which I think interacts with the UEFI bios (I may be wrong).
So I dont know if it will effect permanently. You will have to test it.

I didnt read much into it as I do not have dell laptop, and my only goal was to set battery threshold.
@monethass helped me understand how it works on dell laptop see discussion here https://github.com/maniacx/Battery-Health-Charging/issues/7

From what I have read, now Dell has two ways of changing charging threshold. 
1. libsmbios https://github.com/dell/libsmbios
 opensource and available on distros package manager.
 but on some dell laptop such as XPS  some features doesnt work such as charging threshold

2.  CCTK which is now Dell command configure .(https://www.dell.com/support/kbdoc/en-us/000178000/dell-command-configure)
This is dell official version . I think you can also set various bios options such battery threshold but I don't know for sure. I dont have much information on this.

monethass, 2023-04-27:
@ViBE-HU `smbios-battery-ctl` does interact with UEFI. It changes the config in UEFI when you set the config with `--set-charging-mode`.

maniacx, 2023-04-27:
> meanwhile i tested Lenovo too. sadly you're right. it does not recognise the host natively. but is there any dependencies for Lenovo machines? cause i tried to test it with the command but nothing happened. the host OS is 20.04. and if i set conservation mode via Lenovo Vantage On Windows 10 Ubuntu will follow the rule.

No. there are no dependencies required for Lenovo, it is natively supported by the kernel but it should be installed and running on lenovo hardare and not on Virtual Machine/Virtual Box.

ViBE-HU, 2023-04-27:
> @ViBE-HU `smbios-battery-ctl` does interact with UEFI. It changes the config in UEFI when you set the config with `--set-charging-mode`.

now i see. you are right. but something still wrong. i did a more detailed test on my E5450. i tried first the mode menu and _Express Mode_ is the only one which really interacts with my machine's UEFI. the method i used:
1. set a mode in ubuntu
2. then return to windows and check via Dell Power Manager

only _Express Mode_ worked. any other mode does not have permanent effect.

then i tried to configure `smbios-battery-ctl` directly via terminal and only `express` and `standard` worked. these are truly permanent. any other modes are not somehow. i also checked the UEFI and looks like `express` and `standard` are legacy(?) modes cause any other modes are not listed but can be set via Dell Power Manager. in general if i set a mode with Dell Power manager Windows keep using it.

@maniacx i also noticed that if i set `express` or `standard `via Windows or Linux terminal the plugin:
- will overwrite it after restart if i set a different mode
- not checks the mode when system menu open up
[Képernyőfelvétel ekkor: 2023-04-28 01:22:22.webm](https://user-images.githubusercontent.com/640109/235011738-5ab74acd-983c-4b88-9d33-a8533c1faa49.webm)

maniacx, 2023-04-28:
1. Yes. On reboot extension will restore the mode/threshold to whatever was previously set on linux before reboot.

2. When mode is set using extension, It will set using --set-charging-mode=mode and then verify if it actually set using --get-charging-cfg once

To actually monitor if user have change the mode t using command line, I will have to continuously run
Command --get-charging-cfg. Which will be use lot of resources. So cannot use that.

maniacx, 2023-04-28:
@ViBE-HU Did you ever try changing threshold using cctk in Ubuntu? Are you familar with it?
Are you willing to try?
https://www.dell.com/support/kbdoc/en-us/000178000/dell-command-configure

But My extension doesnt support it. So was thinking about including it.

ViBE-HU, 2023-04-28:
@maniacx 

> 1. Yes. On reboot extension will restore the mode/threshold to whatever was previously set on linux before reboot.
>
>     2. When mode is set using extension, It will set using [...]

and no way to switch on/of this forcing and simply just read UEFI on boot? also i'm not familiar how system menu works but isn't it possible to check the actual config when the user open the exact submenu? gnome also checks somehow charging time when and bunch of plugins have sliders. are those different?

btw as i mentioned i'm not a developer. i'm a kinda advanced user with a light clue about programming. so i don't do random things and if you think cctk can improve the plugin and it's functionality then i can take a try. just need a clear guide what exactly should i check. of course i don't wanna kill my machine but at the moment i'm using it for testing cases and i can take a look.

maniacx, 2023-04-28:
> and no way to switch on/of this forcing and simply just read UEFI on boot?

To implement this feature.
- All modes must be  reported correctly by libsmbios on reboot.
- All threshold values must be reported correctly on reboot

> then i tried to configure smbios-battery-ctl directly via terminal and only express and standard worked.

But as per your test libsmbios on reports only "express" and "standard" correctly on reboot. 
So it is pointless to implement this if all modes and threshold are not reported correctly by libsmbios on reboot.

> it possible to check the actual config when the user open the exact submenu?

Sorry, I dont want to implement.

>  if you think cctk can improve the plugin and it's functionality then i can take a try. just need a clear guide what exactly should i check. of course i don't wanna kill my machine but at the moment i'm using it for testing cases and i can take a look.

Not sure,  But I think in regards to batthery threshold there is no benefits.  On some laptops libsmbios doesnt work, but cctk works so I thinking of adding support.
There was one  comment on arch forum that cctk prevented arch linux from booting as it overwrote a file libcrypto.so during update. So better not to try. 
I actually installed dell command configure (cctk) on Ubuntu 20.04 om my asus just to try. but it said something like libcrypto missing. For me it is not big deal because ubuntu 20.04 is use for testing only, and if any problem I will just format it and reinstall.
I mainly use Windows11 (laptop internal hardrive) and Fedora38 (portable usb hardrive). I have debian bookworm and ubuntu 20.04 just also on portable usb hardrive for testing only..

ViBE-HU, 2023-04-29:
i would like to test the plugin on my ThinkPad R61. it is from ~2008. not sure is it supported. the plugin starts with unexpected error.

maniacx, 2023-04-29:
Wow its way old. Not all thinkpad have this feature. it will work if kernel supports it. you can check if any of the following files exist.
`ls- l /sys/class/power_supply/BAT0/charge_control_end_threshold`
`ls -l  /sys/class/power_supply/BAT1/charge_control_end_threshold`
`ls -l /sys/devices/platform/smapi/BAT0/stop_charge_thresh`
`ls -l /sys/devices/platform/smapi/BAT1/stop_charge_thresh`

ViBE-HU, 2023-04-29:
yeah it's pretty old but one of my favourite machine for home office. i thought the ThinkPad legacy rules are also support this machine.

it has only one removable battery. here's the only working output:
`-rw-r--r-- 1 root root 4096 ápr   29 17:21 /sys/class/power_supply/BAT0/charge_control_end_threshold`

maniacx, 2023-04-29:
Seem like it has it. can you give me the output in this specific order
`ls -l /sys/devices/platform/thinkpad_acpi`
`echo "80" | sudo tee /sys/class/power_supply/BAT0/charge_control_end_threshold`
`cat /sys/class/power_supply/BAT0/charge_control_end_threshold`
`echo "75" | sudo tee /sys/class/power_supply/BAT0/charge_control_start_threshold`
`cat /sys/class/power_supply/BAT0/charge_control_start_threshold`

ViBE-HU, 2023-04-29:
sadly only the firs command gives detailed output.

```
ls -l /sys/devices/platform/thinkpad_acpi
total 0
--w------- 1 root root 4096 ápr  29 18:08 cmos_command
lrwxrwxrwx 1 root root    0 ápr  29 16:58 driver -> ../../../bus/platform/drivers/thinkpad_acpi
-rw-r--r-- 1 root root 4096 ápr  29 18:08 driver_override
-r--r--r-- 1 root root 4096 ápr  29 18:08 hotkey_adaptive_all_mask
-r--r--r-- 1 root root 4096 ápr  29 18:08 hotkey_all_mask
-r--r--r-- 1 root root 4096 ápr  29 18:08 hotkey_bios_enabled
-r--r--r-- 1 root root 4096 ápr  29 18:08 hotkey_bios_mask
-rw-r--r-- 1 root root 4096 ápr  29 18:08 hotkey_enable
-rw-r--r-- 1 root root 4096 ápr  29 18:08 hotkey_mask
-rw-r--r-- 1 root root 4096 ápr  29 18:08 hotkey_poll_freq
-r--r--r-- 1 root root 4096 ápr  29 18:08 hotkey_radio_sw
-r--r--r-- 1 root root 4096 ápr  29 18:08 hotkey_recommended_mask
-rw-r--r-- 1 root root 4096 ápr  29 18:08 hotkey_source_mask
drwxr-xr-x 3 root root    0 ápr  29 16:58 input
drwxr-xr-x 7 root root    0 ápr  29 16:58 leds
-r--r--r-- 1 root root 4096 ápr  29 18:08 modalias
drwxr-xr-x 2 root root    0 ápr  29 18:08 power
lrwxrwxrwx 1 root root    0 ápr  29 16:58 subsystem -> ../../../bus/platform
-rw-r--r-- 1 root root 4096 ápr  29 16:58 uevent
-r--r--r-- 1 root root 4096 ápr  29 18:08 wakeup_hotunplug_complete
-r--r--r-- 1 root root 4096 ápr  29 18:08 wakeup_reason
```
others just reply with "no device" message or something like that however the files are present in BAT0.

maniacx, 2023-04-29:
```
echo "80" | sudo tee /sys/class/power_supply/BAT0/charge_control_end_threshold
cat /sys/class/power_supply/BAT0/charge_control_end_threshold
echo "75" | sudo tee /sys/class/power_supply/BAT0/charge_control_start_threshold
cat /sys/class/power_supply/BAT0/charge_control_start_threshold
```
Unfortunately if these commands dont work, then too extension wont work

ViBE-HU, 2023-04-29:
@maniacx i think i found something. as i mentioned i'm not a programmer but i checked the scripts maybe i can figure out what's going on and your ThinkPad legacy script contains a `tp_smapi` module(?). not clear which exact models use that but i guess this handles the pre-uefi machines and if i install `tp_smapi` i can set thresholds via terminal. i followed [this](https://askubuntu.com/questions/34452/how-can-i-limit-battery-charging-to-80-capacity) instruction (first answer) and looks like it works like a charm. looks like it's permanent too but not sure yet.

maybe your script is not complete or something different? i tested this machine with Ubuntu 22.04 and i guess most oddware IBM/Lenovo ThinkPad (starting with T4x/R4x/X4x series) might work with this method.

maniacx, 2023-04-29:
So how are you setting threshold now?

Can you run this again and give me the output.
```
ls- l /sys/class/power_supply/BAT0/charge_control_end_threshold
ls -l  /sys/class/power_supply/BAT0/charge_control_start_threshold
ls -l /sys/devices/platform/smapi/BAT0/stop_charge_thresh
ls -l /sys/devices/platform/smapi/BAT1/start_charge_thresh
```

ViBE-HU, 2023-04-29:
i did exactly what the commenter said:
```
sudo apt-get install tp-smapi-dkms
sudo modprobe tp_smapi
```

then these example values for testing:
```
echo 40 | sudo tee /sys/devices/platform/smapi/BAT0/start_charge_thresh
echo 80 | sudo tee /sys/devices/platform/smapi/BAT0/stop_charge_thresh
```

and then restarted the laptop and voilà. it's working.

maniacx, 2023-04-30:
As this extension support several device it search for files at certain path do know what type of laptop it is installed on.
For example. If these 3 files exist, it will know that is it thinkpad
```
/sys/devices/platform/thinkpad_acpi
/sys/class/power_supply/BAT0/charge_control_end_threshold
/sys/class/power_supply/BAT0/charge_control_start_threshold
```
If these 2 files exist, it will know it is legacy thinkpad tp-smp
```
/sys/devices/platform/smapi/BAT0/stop_charge_thresh
/sys/devices/platform/smapi/BAT0/start_charge_thresh
```
Strange You laptop has all 5 files.

 Just out of curiosity does this files still exist in your system after installing tp-smp-dkms and restarting. or do they get removed? You can check again with
```
ls -l /sys/class/power_supply/BAT0/charge_control_end_threshold
ls -l  /sys/class/power_supply/BAT0/charge_control_start_threshold
```

ViBE-HU, 2023-04-30:
```
-rw-r--r-- 1 root root 4096 ápr   30 20:24 /sys/class/power_supply/BAT0/charge_control_end_threshold
-rw-r--r-- 1 root root 4096 ápr   30 20:24 /sys/class/power_supply/BAT0/charge_control_start_threshold
```

maniacx, 2023-05-29:
@ViBE-HU , @monethass 
I am having an issue reported by a user on lenovo , where the firmware updater install libsmbios automatically on their device. 

Currently in the extension , it only checks if **smbios-battery-ctl** is present and report device as  dell devices. Now since knowing that other brand laptops may also install this program, I will have to add additional checks for detecting if the laptop is a dell laptop or not.

On your dell laptop, please and comfirm if this file exist on your device. According to the kernel code, all dell device register this file/folder

`ls /sys/devices/platform/dell-laptop`

ViBE-HU, 2023-06-08:
> `ls /sys/devices/platform/dell-laptop`

just checked:

ls /sys/devices/platform/dell-laptop
driver  driver_override  leds  modalias  power  rfkill  subsystem  uevent

btw i have no time to test at the moment but looks like settings are still not permanent on dell somehow. is this a bug or it's normal?

maniacx, 2023-06-08:
Not permanent?
You mean it changes after sometime?
Does the settings change on reboot?
Either way , extension should reapply threshold on reboot, so might be a bug.
When you have time please test and let me know, when does the threshold change

I will also check on my end too.

ViBE-HU, 2023-06-25:
now finally my E5470 is back and i tested the plugin again. if i change the mode it does not permanent. i thought your recent update targeted this thing. it's working but it is only permanent if i change the mode via Windows. on Ubuntu it is temporary. is this normal?

maniacx, 2023-06-25:
> now finally my E5470 is back and i tested the plugin again. if i change the mode it does not permanent. i thought your recent update targeted this thing. it's working but it is only permanent if i change the mode via Windows. on Ubuntu it is temporary. is this normal?

I am not understanding what do mean by permanent. I am assuming you talking about charging threshold  from windows and ubuntu. When you boot from windows to Ubuntu it should restore your charging threshold values what was ubuntu.

This is an example of behavior
* If you are in Ubuntu and set threshold custom:55-60 for example. 
 
The extension will apply the mode and threshold to SMBIOS. it will also save the mode and threhsold schema gsettings **/org/Gnome/shell/extensions/Battery-Health-Charging**

*  Then you boot windows and set threshold custom:75:80  for example.

* Now , if you boot again to Ubuntu again, the extension will apply threshold custom:55-60, during boot.  It will restore the mode and threshold from schema gsettings that was last saved in Ubuntu.

This an example of normal behavior of the extension.  Is it behaving differently for you?

Locke, 2023-10-29:
I understand this issue to be for general discussions too, so I am posting it here instead of a new issue.

I am using this extension on my Lenovo ThinkPad L13 Yoga and are happy with it.

In addition to the charge thresholds, my device supports changes to the charge behavior:

```
$ cat /sys/class/power_supply/BAT0/charge_behaviour
[auto] inhibit-charge force-discharge
```

I am using these currently from the console, and would love to see support for that via gnome, just like the charge thresholds. I think it would be best to have it in the same extension, or do you think that should be a separate one?

My use case:
I am using the device mostly as a desktop working from home. I have the peripherals on a KVM switch, which also delivers power over the same cable.
I normally have the charge thresholds low to extend the battery life. When I anticipate to use it as a notebook/tablet, I charge it up (e.g. for an office day with meetings). When I connect it back to the KVM at home, I often have more charge left than the threshold I'd like to have at home. In this case, I need to drain the battery - either via above `charge_behaviour` or by not connecting the device to the KVM (I cannot simply unplug the power to the KVM as it powers the KVM and the USB devices).

Alternative use case:
Some energy providers are offering a dynamic electricity pricing; in central Europe typically low (sometimes negative) on sunny or windy hours, and high on evenings. Some users may want to manually control when their devices are charging or discharging due to that.

maniacx, 2023-10-31:
Lets continue this discussion here https://github.com/maniacx/Battery-Health-Charging/issues/67#issue-1969706683

maniacx, 2023-10-31:
> I understand this issue to be for general discussions too, so I am posting it here instead of a new issue.
>
> I am using this extension on my Lenovo ThinkPad L13 Yoga and are happy with it.
>
> In [...]

Lets continue our discussion here https://github.com/maniacx/Battery-Health-Charging/issues/67#issue-1969706683

ponychicken, 2024-08-04:
Thanks for this beautiful extension ::))