maniacx/Battery-Health-Charging #24: Dark Mode

abhshkdsgh, 2023-04-09:
Dark Mode on Ubuntu 22.10 doesn't get applied to this gnome-shell-extension.
Theme: Dracula (https://www.gnome-look.org/p/1687249) (Shell & Legacy Applications).
I tried changing from light mode to dark mode also.

abhshkdsgh, 2023-04-09:
[image]

maniacx, 2023-04-09:
Can you post some screenshot. You can just drag or attach the screenshot file in the comment

maniacx, 2023-04-09:
Ok I just tried the theme and the issue is not with my extension but with the theme. All other extension have the same issues and behave the same as my extension. The installation method suggested in the instructions is not enough to apply the theme properly, for gnome 40 and above. A quick search in the Dracula's themes issue I found the solution.

For Preference you need to copy all contents from gtk4 in ~/config/gtk4 folder as mentioned here https://github.com/dracula/gtk/issues/203#issuecomment-1486034571

For gnome shell fix is copy/replace files from v40 folder as mentioned here https://github.com/dracula/gtk/issues/124#issuecomment-875036340

But I am just assuming your problem,  since you havent mentioned what is the exact issue and I will leave this open.
Please give me more info and continue with your issue and close if the themes fixes works for you.

[image]
[image]

abhshkdsgh, 2023-04-10:
In quick settings only dark mode didn't get applied. 
I am sorry to raise the issue here. I should have noticed this is a problem with theme not with this extension.
The solution pointed out by @maniacx worked for me.
Thank you for helping me even when you could just delete this issue once you noticed it's not a problem with this extension.

closed 2023-04-10 by abhshkdsgh