maniacx/Battery-Health-Charging #25: Extensions is not compatible

girishpal0070, 2023-04-19:
On pop os extension is not compatible laptop is FX506LI Asus tuf but this extension works on ubuntu and fedora properly but not on pop os

girishpal0070, 2023-04-19:
No JS module 'quickSettings' found in search path

error

maniacx, 2023-04-19:
I haven't used pop os. But by the error it seems like it is not using gnome43. My extension only supports gnome 43 and above (44). I tried looking at popos website and I couldn't find which Gnome version is currently used on popos. But some Reddit post suggest popos is still using gnome42 and may possibly not move to gnome43/44 as they are developing own cosmic version os

maniacx, 2023-04-23:
Hey @girishpal0070 
I have ported the extension for Gnome42.
But because of some string dont fit on Gnome42 quick panel, I had create a seperate extension. I have tested for Asus, but didnt test for other device. Please install an test it.

https://github.com/maniacx/Battery-Health-Charging-Gnome42

Download. Extract, Run ./install.sh 
Log out/ Login

Let me know how it goes.
Also mention which device you are testing

maniacx, 2023-04-23:
Extension got approved by Gnome extension. You can start using it on Pop-os
https://extensions.gnome.org/extension/5724/battery-health-charging/

closed 2023-04-23 by maniacx