maniacx/Battery-Health-Charging #26: Unify selected item with Energy Mode

ai, 2023-04-24:
GNOME settings dropdown already have a similar list with modes, Energy More. But it uses `✓` mark instead of `•` to highlight the current element.

Should we use the same marker?

[image]

[image]

maniacx, 2023-04-25:
I changed my extension to checkmark. However I don't know if it looks the same as in **power mode**.
I have on two modes of powerprofile so no popupmenu.
Could you install this **test** version and verify it looks the same as in power profile.
[test.zip](https://github.com/maniacx/Battery-Health-Charging/files/11315526/test.zip)

If during install you get **msgfmt** error. install gettext
sudo apt install gettext
or
sudo dnf install gettext

That check mark looks ugly and little out of place though. 
But for consistency I will include it in my next update, if you confirm it looks the same as your **power mode**.

[image]

I wish they would make it look more like the wifi check mark.
[image]

Let me know if look the same, i will include it

maniacx, 2023-04-25:
Yes. Hope so gnome does implement something to fix this.

 I have already released a version 14 on gnome extension with ornament changed from DOT to CHECK. I will close the issue.
Thanks for reporting.

closed 2023-04-25 by maniacx