maniacx/Battery-Health-Charging #27: Feature request: Support MacBook with Apple Silicon (Asahi Linux)

teohhanhui, 2023-04-25:
See https://social.treehouse.systems/@AsahiLinux/110073752180666718 and https://github.com/francku/gnome-shell-extension-battery-threshold/pull/11

maniacx, 2023-04-25:
Interesting. I am just trying to understanding somethings
1. Is Asahi Linux is a kernel? or A linux distro? 
2. And how to do you set battery threshold? using command line?

teohhanhui, 2023-04-25:
1. Is Asahi Linux is a kernel? or A linux distro?

    Asahi Linux is a project to support Apple Silicon devices on Linux. There is an official distro by the same name which is currently based on Arch Linux ARM. See https://asahilinux.org/about/

2. which gnome version are you using?

    GNOME 44

3. And currently how to do you set battery threshold? using command line?

    Yeah, or udev rules.

maniacx, 2023-04-25:
What is the command?
Do you have to install additional packages or work around to get this command working?

teohhanhui, 2023-04-25:
Just the standard `charge_control_start_threshold` / `charge_control_end_threshold`:

```
[teohhanhui@han-macbook-air ~]$ ls -la /sys/class/power_supply/macsmc-battery/charge_control_*
-rw-r--r--. 1 root root 16384 Apr 26 06:21 /sys/class/power_supply/macsmc-battery/charge_control_end_threshold
-rw-r--r--. 1 root root 16384 Apr 26 06:21 /sys/class/power_supply/macsmc-battery/charge_control_start_threshold
```

maniacx, 2023-04-25:
Can you check is  "/sys/module/applesmc" file is present on your system.?
Cause Apple-Intel uses it and it might conflict with Asahi linux.
`ls -l /sys/module/applesmc`

teohhanhui, 2023-04-25:
It is not present:

```
[teohhanhui@han-macbook-air ~]$ ls -la /sys/module/applesmc
ls: cannot access '/sys/module/applesmc': No such file or directory
```

teohhanhui, 2023-04-25:
May I suggest splitting into `devices/AppleIntel.js` and `devices/AppleArm.js`?

maniacx, 2023-04-25:
> May I suggest splitting into `devices/AppleIntel.js` and `devices/AppleArm.js`?
Oh I just named it Asahi.js
Ok test this. 
[origin.zip](https://github.com/maniacx/Battery-Health-Charging/files/11327342/origin.zip)

Download and Extract and run ./install.sh
If during installation if you run "msgfmt" error. install gettext
`sudo apt install gettext`
`sudo dnf install gettext`

Let me know if it works.
if it doesnt send me the logs
`journalctl -f -o cat /usr/bin/gnome-shell`

maniacx, 2023-04-25:
Sorry about that it was wrong zip file, so I edited comment and removed it
[test.zip](https://github.com/maniacx/Battery-Health-Charging/files/11327381/test.zip)

maniacx, 2023-04-25:
Oh also I need one more information
`ls -l /sys/class/power_supply`
Does it have the a filenode for charger? usually it is AC or AC0/1 or ADP0/1

teohhanhui, 2023-04-25:
It's `macsmc-ac`

maniacx, 2023-04-26:
ok. so probably it should have an file "online" to check if charger is plugged or not.
Can you check if the file is present if it changes state to "1" or "0" when charger is plugged or unplugged?
`cat /sys/class/power_supply/macsmc-ac/online`

This would be required for the **`Change battery indicator icon behavior`** settings in extension preference to work properly.
Please check and let me know if the online file exist and reads charger status.
Also let me know about the test.zip results.

teohhanhui, 2023-04-26:
Hmm... I'm trying your modified code, but I don't understand why this extension needs to install polkit policy and rules, whereas the *other* extension works with just plain `pkexec tee`:

https://github.com/search?q=repo%3Afrancku%2Fgnome-shell-extension-battery-threshold+pkexec&type=code

maniacx, 2023-04-26:
> Hmm... I'm trying your modified code, but I don't understand why this extension needs to install polkit policy and rules, whereas the _other_ extension works with just plain `pkexec tee`:
> 
> https://github.com/search?q=repo%3Afrancku%2Fgnome-shell-extension-battery-threshold+pkexec&type=code
> 
> Does it have anything to do with the way you're executing a separate shell script (`batteryhealthchargingctl`)? But why do that when you have the full expressiveness of JavaScript already? thinking

Most of the device need root privilged to change threshold.  So if I implement the same method as in francku extension (without polkit) there are 2 drawbacks.

1. Everytime you need to change threshold, it will prompt for a password with a pkexec dialog box
2. Most device reset threshold setting on reboot.  So on every reboot, the extension will ask the pkexec password to set it back to previous.

Using polkit solves this issue with ease of usabilty.

If Asahi linux can change the threshold without root privilegdes (such as sudo, su or pkexec)  out of the box (without any additional workaround such as daemon / systemd script) please let me know. I will remove to option of installing polkit for Asahi linux and modify my code not to use pkexec.

maniacx, 2023-04-26:
Also I will need more information on the charger by running the below command twice. one with charger plugged-powered and second with charger disconnected.
`cat /sys/class/power_supply/macsmc-ac/online`

This will help me to make this feature working.
[image]

To better understand what this setting does read these 2 post below

https://github.com/maniacx/Battery-Health-Charging/issues/11#issuecomment-1493185059

maniacx, 2023-04-26:
> Hmm... I get what you mean, but I think this is trying to solve the problem at the wrong level.

Until devs at gnome, upower and device kernel maintainers get on the same page, we have no option.

For me, changing threshold once I wouldn't mind to enter password, but I hated the fact that I had to enter password on my Asus laptop to change threshold on every reboot. So this extension was born.
Luckily I found this 2 extension using polkit and I just modified my way around it change threshold

https://extensions.gnome.org/extension/4372/shutdowntimer/
https://extensions.gnome.org/extension/945/cpu-power-manager/

maniacx, 2023-04-26:
@teohhanhui 
I have included the charging info. Please do this final test for me regarding charging icon as described in part1 here https://github.com/maniacx/Battery-Health-Charging/issues/11#issuecomment-1493185059

[test2.zip](https://github.com/maniacx/Battery-Health-Charging/files/11335681/test2.zip)

Take your time and reply, if and when every  is working as it should, so that I can submit it for gnome extension review.

teohhanhui, 2023-04-26:
The "Change battery indicator icon behavior" option is *really* confusing. I had to look at the code to understand what it's meant for.

If I understand correctly, it's the opposite of what the explanation text under the option says:

> Battery indicator icon switches to **charging** in system tray and quick settings panel when charging threshold is enabled and device is powered by charger.

The option in fact forces the use of the "charging" icon even when the device is *not* charging because of the charge control threshold.

May I suggest changing the wording to:

> Switch the battery indicator icon to **charging** in system tray and quick settings panel when device is connected to AC power, even when charging is disabled due to charge threshold.

teohhanhui, 2023-04-26:
Otherwise everything is working as expected. :tada:

maniacx, 2023-04-26:
Thanks. Your description makes more sense. It was very hard for me to describe what it does.
I will add it.