maniacx/Battery-Health-Charging #29: Add systemd service

AbrarSL, 2023-05-03:
See: https://github.com/tshakalekholoane/bat

The above program installs a systemd service that sets the battery charging level on boot. Would it be possible to incorporate something similar in this extension?

maniacx, 2023-05-03:
If this extension can directly set the threshold, I don't see a reason to depend on a 3rd party module (unless it is really requireds such as dell libsmbios and cctk). 

Another reason, my first release version 1, 2 and 3 of this extension, used a systemd service. There were users having issues where on boot, extension already before  systemd service and extension would throw a error as it was unable to set threshold. With pollkit there are no such issues.

So no it doesn't make sense for me to use this 3rd party systemd service.

Are you having issues setting threshold? Which laptop and operating system are you using?

AbrarSL, 2023-05-03:
I'm using an Asus Vivobook M513. My OS is Fedora Silverblue 38.

The extension is generally reliable but it seems to take far too long to reapply the threshold after waking from suspend. The balanced settings also doesn't seem to apply sometimes?Another issue with a purely GJS extension is that these run in the user session. These settings don't get applied until I actually log in to my account.

maniacx, 2023-05-03:
Which version of Gnome? 44?

AbrarSL, 2023-05-03:
Yes

maniacx, 2023-05-03:
Seems like you have 3 issue.
> The extension is generally reliable but it seems to take far too long to reapply the threshold after waking from suspend.

Can you give me more details.  As soon as you log-in what happens?
1. Is the Extension itself starting late? Is it only this extension or other extensions too starting late?
or
2. Extension has started but threshold applies very late?

Also do you have any other scripts or 3 partly modules setting / controlling threshold?

> The balanced settings also doesn't seem to apply sometimes?

How are you verifing balance settings is not applied?
In quick settings does show `✓` on balance after changing? or Is the `charging limit set to` displaying wrong threshold? or are you using `cat` command to verify?

> Another issue with a purely GJS extension is that these run in the user session. These settings don't get applied until I actually log in to my account.

Yes. This i cannot solve. One way is to have to utilize systemd but Gnome extension reviewer wont approve. But I think I will look into it after I solve the above 2 issues.

AbrarSL, 2023-05-03:
I managed to reproduce the issue.

[image]

maniacx, 2023-05-03:
> Extensions, but wouldn't the extension be able to reset and set the limit whenever the extension is initialized? I'm guessing it's reinitialized everytime the shell is locked and unlocked?

Everytime you logout extension gets disabled
Everytime you lock your screen extension gets disabled.
Everytime use log-in extension gets enabled.
Everytime unlock the screen, extension is enable.

When this extension is enabled, it will apply the threshold value stored in gsettings (what ever value was set before previous logout or before entering lockscreen)

Now if you are using `bat` application also together with the extension. It will could be that the extension apply threshold before the bat application, and the bat application overwrites it.
I would suggest you to remove the bat application temporarily and just use only the extension, to see if issue exist.

maniacx, 2023-05-03:
So I am running fedora38 on asus viwobook M401QC. I was trying to experiment your case.

I currently suspended my laptop to 60. And resume and found the kernel still showing 60. But that is because my extension sets threshold back to 60

So I did another test
I uninstall my extension and manually set threshold.
echo 60 | sudo tee /sys/class/power_supply/BAT0/charge_control_end_threshold
Suspended the device. After 20 mins resume back.
And still it is set to 60 and not charging.
cat /sys/class/power_supply/BAT0/charge_control_end_threshold

Even without the extension the kernel still holds charge threshold to 60 after suspend and resume
Only when I reboot my laptop the kernel sets, threshold back to 100.

So there is some external factor that is changing your threshold during sleep. If you have the `bat` module, could be it.
Or else the kernel. This extension only applies the command **echo 60 | sudo tee /sys/class/power_supply/BAT0/charge_control_end_threshold** and reads the threshold value by reading **charge_control_end_threshold** . Rest everything is handled by kernel

AbrarSL, 2023-05-05:
I don't have any other program that sets the threshold installed. I only had `bat` and I removed that before installing your extension.

In my case the threshold holds even after suspend, however the batttery keeps charging. I suspect this is more of a kernel bug, however `bat` had a service that would reset the threshold after a suspend. It's possible that reverting 514ffff might fix the issue. I'll give it a try and report back.

AbrarSL, 2023-05-05:
I tested this out and reverting that commit fixes the issue.

AbrarSL, 2023-05-05:
> Strange that kernel is showing wrong values after suspend.

Just to be clear. The sys interface shows the correct value after suspend but the behaviour is wrong.

maniacx, 2023-05-07:
@AbrarSL 
Can you please download this extension from the **main** branch on github and install/test using the updated code.
I have removed the checks, threshold should apply everytime.
Let me know . If everything works, I will submit it for review on gnome-extension

AbrarSL, 2023-05-07:
I'm having trouble installing the extension from source.

The packaging is failing with exit code 1

maniacx, 2023-05-07:
Hmm Silverblue....

[Battery-Health-Charging@maniacx.github.com.shell-extension.zip](https://github.com/maniacx/Battery-Health-Charging/files/11415005/Battery-Health-Charging%40maniacx.github.com.shell-extension.zip)

Download open terminal in the folder where zip file is present and run command
`gnome-extensions install Battery-Health-Charging@maniacx.github.com.shell-extension.zip --force`

Logout and Login, enable in extemsions/extension-manager app

Am wondering how did you manage to install and test earlier, after reverting the commit?

AbrarSL, 2023-05-07:
It works after suspend now!

> Am wondering how did you manage to install and test earlier, after reverting the commit?

I packaged it in an Arch container. It worked last time, it didn't this time. Silverblue is missing some program and I didn't want to layer it.

maniacx, 2023-05-07:
I apologize it was my mistake. I just checked. One of the language transalation file would fail and wont allow it to compile if I delete the zip. Throws the same`Child process exited with code 1`

closed 2023-05-08 by maniacx