maniacx/Battery-Health-Charging #33: Fedora 38 error message on login "This extension is not compatible with your device"

xuanvinhvu, 2023-05-05:
I tried to install the Battery Health Charging through https://extensions.gnome.org/ but it failed because my device is not compatible. Since then whenever I log in I'm greeted with this error.

[image]

It's not listed as a gnome extension to uninstall.

maniacx, 2023-05-05:
Please give me details?
Make and model? Example Asus viwobook  
Which operating system? Example fedora/Ubuntu
Which gnome version?

xuanvinhvu, 2023-05-05:
Thanks for the quick reply and your suggestion works. I installed the Extensions app and was able to remove it.

In case others might benefit here is the details you requested.

Make and model: Maingear Vector Pro 15 MG-VCP15
OS: Fedora 38
Gnome Version: 44.1

Are you using extension manager app or extension app? Neither if I understand it correctly, I went to https://extensions.gnome.org/ and when I click on the install button it asked me to install browser extension.

maniacx, 2023-05-05:
It is better to install extension manager app or extension, to 

You can install and remove the apps. 

Also you can enable and disable apps if any extension causes any issues.helps troubleshooting .

if the devs post any update it will automatically update to the latest version of extension.

As for your laptop. My extension doesn't support it.

But if your laptop can set threshold in Linux any other way such as using commandline, let me know.

closed 2023-05-11 by maniacx