maniacx/Battery-Health-Charging #37: Add intelligent charging

jeffconrad18, 2023-05-08:
Add intelligent charging, like with smartphones: 
-if the computer is connected overnight, it should not be at 100% until e.g. 8 o'clock (best found out by usage time)
-if the computer is connected during the day, it should charge full
-if the computer is connected over 48 h, it should charge until 60%

This allows a longer battery lifetime.

ViBE-HU, 2023-05-08:
lol.

maniacx, 2023-05-08:
> Add intelligent charging, like with smartphones: -if the computer is connected overnight, it should not be at 100% until e.g. 8 o'clock (best found out by usage time) -if the computer is connected [...]

It would difficult to implement. This can be done by manufacturer on bios level (i think some dell laptops has this). As for extension has no way to change threshold during sleep or power-off. (unlike smartphone)

jeffconrad18, 2023-05-09:
> It would difficult to implement. This can be done by manufacturer on bios level (i think some dell laptops has this). As for extension has no way to change threshold during sleep or power-off. [...]

Or maybe we can just add a simple if-then condition: if it's after 10 p.m. and before 6 a.m., then the charging limit should be 80%

maniacx, 2023-05-10:
> Or maybe we can just add a simple if-then condition: if it's after 10 p.m. and before 6 a.m., then the charging limit should be 80%

I dont see any benefit that this will improve battery health.  Lets take an example,
At 10pm my battery is 100%, and threshold is automatically switched to 80%. On my laptop when threshold is set to 80% and my current battery level lets say 100%. battery starts discharging at a very low rate, (1% per 1.5 hours), At 6am the battery level would be approx 92-94% then charging threshold switches back to 100%, battery starts charging at 100%.
Battery level will never reach to 80% and from 10pm-6am the battery average at 96%. I dont think this will help improving battery life.

In smartphones world it is a bit different scenario, AFAIK. the smart charging controls the rate of charging and not the level .
Most smartphone charger use fast charging which is beneficial for recharging the phone quickly which most user want.
Specially during the day when the phone is used constanly and you need to quickly charge your battery.
But fast charging also harms the batteries reducing it lifespan. 
So smartphone manufactures have come up with smart/intelligent charging, when if charged at night, (since most user leave it plugged in at new until morning) it will **prevent the use of fast charging** and charge at a slower rate, specially when the batter reaches above 90% to prevent battery aging due to fast charging.

I hope this makes sense, and help clarify why I wont be implementing this feature.

closed 2023-10-13 by maniacx