maniacx/Battery-Health-Charging #42: Menu does not adjust to selected mode

olebole, 2023-05-20:
On my system (Thinkpad Z13, upcoming Debian 12, Gnome 43.4, Wayland), the selection in the menu is not changed when I select a different mode. The checkmark and the caption always remain at 100%. The only place where the value changes is the icon, which makes the display inconsistent:

[image]

My specific problem here is that I want to disable the icon (because the info is not needed day-to-day), but this is the only place where the information is written properly.

maniacx, 2023-05-20:
It seems like you threshold is not getting applied by the extension for some reason.
**Batterieladeschwellen: 100/95%**  displays current threshold read  just after changing theshold.

Can do some test for me. Apply "Maximale" threshold and check what is your actual threshold by typing this in terminal.
```
cat /sys/class/power_supply/BAT0/charge_control_end_threshold
cat sys/class/power_supply/BAT0/charge_control_end_threshold
```
Cat command should output 60 and 55 (or if you are using custom value it will show that)

If the cat command output is 100 and 95, try to manually update it using terminal to check if it updates. Just for test we willd apply 70/65 
```
echo 70 | pkexec tee /sys/class/power_supply/BAT0/charge_control_end_threshold
echo 65 | pkexec tee  sys/class/power_supply/BAT0/charge_control_end_threshold
```
Now read again. using command below
```
cat /sys/class/power_supply/BAT0/charge_control_end_threshold
cat sys/class/power_supply/BAT0/charge_control_end_threshold
```
It should read 70/65
Let me know what result you get by testing.

Also did you install using any custom scripts or systemd service or apps to change charging threshold?

olebole, 2023-05-21:
I had tlp installed before, but this is stopped for the moment. tlp worked fine (using the same intefrace). Also, the commands you proposed work fine.
However, when setting via the extension, the end threshold is always set to 100, not to the desired value. Start threshold changes accordingly:
* "100" --> 95/100
* "80" --> 75/100
* "60" --> 55/100

The thresholds are set to default (and are displayed as 100/80/60 in the settings).

maniacx, 2023-05-21:
> i.e. when going down, one needs to change the start threshold first.

Thanks for finding the issue, I wasnt aware about it. I read TLP's code and it does mention some devices having issue, and it the same, if going down, change the start threshold first.

I have made a test.zip. Can you please test it for me.

[test.zip](https://github.com/maniacx/Battery-Health-Charging/files/11524781/test.zip)

extract and run **./install.sh**
If you come across an error about **msmgmt**, install **gettext**.

One question. Looking at you system you are using German language. There was no contributer for german language so I used google translate for translation.
How are the translation? Are they correct? Are the text in the settings and the panel understandable / grammatically correct?

olebole, 2023-05-21:
The updated version works for me; the thresholds are now always set accordingly. Thank you for the quick fix!
The German translation is mostly fine; there is a small Glitch in  "Einstellungen/Symbolaussehen bestimmen": 
> … wechselt der Symboltyp to "**Symbole nur**" und …

should be, as in the selection:
> … wechselt der Symboltyp to "***Nur Symbole***" und …

Also, in the main menu I would remove the word "Modus" in each item: In German one would write compound nouns like these with a dash, but this doesn't look well. As the caption already mentiones "Modus", I would limit this to

> ### Batterielademodus
> * Volle Kapazität
> * Ausgeglichen
> * Maximale Lebensdauer

That's all I found.

maniacx, 2023-05-21:
Great. I will submit an updated version in a few days. Thanks.

closed 2023-05-22 by maniacx