maniacx/Battery-Health-Charging #44: Changes not kept + Battery Health Charging encountered an unexpected error on Toshiba laptop

bpstuder, 2023-05-23:
On a Toshiba Portégé A30-C-158, I tried to change Battery Charging Mode from Ful Capacity to Maximum Lifespan, the icon change but it's rollbacked as soon as the menu closes. The icon show 80% but in the menu it's back to full capacity.
In addition, I get Battery Health Charging encountered an unexpected error

_Tests as requested_
`~$ echo 80 | sudo tee /sys/class/power_supply/BAT1/charge_control_end_threshold `
`80`
`~$ cat /sys/class/power_supply/BAT1/charge_control_end_threshold`
`100`

It seems the modification are not kept.

_System infos :_ 
OS: Pop!_OS 22.04 LTS x86_64
Kernel: 6.2.6-76060206-generic
DE: GNOME 42.5

maniacx, 2023-05-23:
Oh strange. For some reason the kernel is not applying the threshold. When echo 80, the cat should reply back to 80.
Do you have any other extension, scripts or app that controls sets charging threshold? I would suggest to remove all of them, (even my extension) and repeat the command
**echo 80 | sudo tee /sys/class/power_supply/BAT1/charge_control_end_threshold 
cat /sys/class/power_supply/BAT1/charge_control_end_threshold**

maniacx, 2023-05-26:
Hey @bpstuder 
Any results. Did you dicharge you battery level to 70% and then try changing threshold using **echo** and reading back using **cat**?

bpstuder, 2023-05-26:
Hello
Sorry I had no time to do tests, will do this weekend

bpstuder, 2023-05-27:
hello
So I tried after discharging my battery and change never applies. I tried cat 10 times in a row, and it never changes.
Also, nothing in dmesg about the threshold.

maniacx, 2023-05-27:
Ok. Reading charging threshold from kernel is bug.

What about threshold? Does it apply?
If you discharge battery  is at 70% and you to echo command
`echo 80 | sudo tee /sys/class/power_supply/BAT1/charge_control_end_threshold 
`
And then connect your charger. Does it charge to 100% or does it hold charging to 80%?

bpstuder, 2023-05-28:
So I've discharged my battery to 69%, set the threshold and now it's charged up to 80%, not going more than that.
`cat /sys/class/power_supply/BAT1/charge_control_end_threshold` gives me 80, so it looks good

maniacx, 2023-05-28:
Good. At least you can set the threshold.

I will change the extension to only set threshold only. It will not read and verify if it is change.

bpstuder, 2023-05-31:
so I installed the git version. Here are some tests I did :

bpstuder, 2023-05-31:
so I've installed the git version and here are some tests : 
- battery at 60%. `cat /sys/class/power_supply/BAT1/charge_control_end_threshold
` reads 80. I plug the adapter, icon shows 80. I switch to full capacity, icon change to 100, but the charge_control_end_threshold variable doesn't change. I go back to the extension, I'm in full capacity and icon shows 100.
- battery at 85%. I change the mode to max lifespan, charging stops.

maniacx, 2023-05-31:
I am bit confused now, about threshold set for toshiba.

We have confirmed that reading for charge_control_end_threshold does not work.

And according to the kernel commit, it said that threshold cannot be set if battery level is above 80%.
if battery level is above 80%, you cannot switch threshold from 100% to 80%

But it seems like on your laptop it works. I just want to confirm again before and redo the changes the extension again.

Since we cannot rely `cat /sys/class/power_supply/BAT1/charge_control_end_threshold` command (kernel bug), we will check the actual status on battery itself. Follow exact direction

**Keep the battery level at 85% and try on terminal**

`echo 80 | sudo tee /sys/class/power_supply/BAT1/charge_control_end_threshold  
`
wait 5 seconds

`cat /sys/class/power_supply/BAT1/status`

`upower -i /org/freedesktop/UPower/devices/DisplayDevice`

`echo 100 | sudo tee /sys/class/power_supply/BAT1/charge_control_end_threshold `

wait 5 seconds

`cat /sys/class/power_supply/BAT1/status`

`upower -i /org/freedesktop/UPower/devices/DisplayDevice`

`echo 80 | sudo tee /sys/class/power_supply/BAT1/charge_control_end_threshold  
`
wait 5 seconds

`cat /sys/class/power_supply/BAT1/status`

`upower -i /org/freedesktop/UPower/devices/DisplayDevice`

`echo 100 | sudo tee /sys/class/power_supply/BAT1/charge_control_end_threshold  
`
wait 5 seconds

`cat /sys/class/power_supply/BAT1/status`

`upower -i /org/freedesktop/UPower/devices/DisplayDevice`

Paste the output

bpstuder, 2023-06-04:
Results are attached here : [test.txt](https://github.com/maniacx/Battery-Health-Charging/files/11645845/test.txt)

maniacx, 2023-06-04:
The charging threshold seems to apply even above 80%.
However, since the kernel developer mentioned, threshold applies and take effect only below 80%, i think it is better  to use this way. Thank you for all the testing and information. I will leave the issue open.

closed 2023-10-13 by maniacx