maniacx/Battery-Health-Charging #45: Incorrect device detection

hensnenenej, 2023-05-29:
As I see in `devices/Dell.js`, detection of Dell laptops is done by simply checking if `smbios-battery-ctl` installed.
However, it can be installed on machines other than Dell, i.e. as a dependency for another package:
```
❯ pacman -Qo $(which smbios-battery-ctl)| tee >(awk '{print $5}' | xargs -I{} pacman -Qi {} | grep '^Required')   
/usr/bin/smbios-battery-ctl is owned by libsmbios 2.4.3-6
Required By     : fwupd
```
As a result, my Lenovo laptop is not detected correctly, and extension fails to init with `Battery Health Charging: Error threshold values not updated`. 
Also it would be nice to log the result of device detection and/or give user an option to select device manually, it took me some time to figure out what's happening.

maniacx, 2023-05-29:
> As I see in devices/Dell.js, detection of Dell laptops is done by simply checking if smbios-battery-ctl installed.
However, it may be installed on machines other than Dell, e.g. as a dependency for another package:

I had not idea why would any other brand laptop  other than dell would install libsmbios. as far as I know libsmbios would be useless on any other device, but seem like it is required by fwupd in your case. Thank you for finding the actual cause and bringing it to my attention. I will add additional checks for dell devices in the next update.

> As a result, my Lenovo laptop is not detected correctly, and extension fails to init with Battery Health Charging: Error threshold values not updated.
Also it would be nice to log the result of device detection and/or give user an option to select device manually, it took me some time to figure out what's happening.

Gnome extension guideline suggest to use minimal logs, so I havent included any.   My list of device supported is a log (22 devices) and giving an option to choose wrong device would only result in more user errors. The correct way to fixing dell detection. I will work on it.

Just curios. Which lenevo laptop do you have? thinkpads or ideapad/legion series?

hensnenenej, 2023-05-29:
> Gnome extension guideline suggest to use minimal logs, so I haven't included any.

IMO it would be helpful and not really contrary to guidelines to log _errors_ a little more verbose, something like `Error (filename:line) ...` for example.

> Just curios. Which lenevo laptop do you have? thinkpads or ideapad/legion series?

Yoga slim, sometimes referred as ideapad as well (ideapad_laptop driver)

maniacx, 2023-05-30:
I haven't add any logs, but when an error notification comes up it will display the device it has detected, so at least the user has an idea that wrong device is detected. I would never guess that libsmbios gets installed on other devices other than dell, thanks for debugging that.

Can you please try the updated version I push on github in the main branch. 
Download/extract and ./install.sh
Install gettext if you get any error related to msgfmt
Also let me know which gnome version are you using.

hensnenenej, 2023-05-30:
> Can you please try the updated version I push on github in the main branch.

Yep, it works.

> Also let me know which gnome version are you using.

```
❯ gnome-shell --version     
GNOME Shell 44.1
```

closed 2023-06-02 by maniacx