maniacx/Battery-Health-Charging #47: Tranalation Update

maniacx, 2023-05-30:
Tranalation Updates

maniacx, 2023-05-30:
Few more string added. Please assist.
@Valeria 
@ViBE-HU 
@Vistaus  
@sabriunal 
@viksok

maniacx, 2023-12-14:
@Valeria
@ViBE-HU
@Vistaus
@sabriunal
@viksok
@dalz
@ludrol

Hello everybody. 
Thank you for the contribution in translation. 
I think I may have covered support for most laptops and features available, probably finalize with the extension.
Some common translation have been updated by me using a script for translation that is already existing in repos like gnome-shell, libadwaita and some other project.
Some complex translation are still missing.
I would appreciate if you can contribute once again translation.

Some explanation to the complex strings for context

## Settings Apple Laptops
This is for charging LED / light on apple laptop
When button enabled: Indicator light glows green when charger on hold. (battery level reaches charging threshold). Indicator light glows amber while charging.
When button disabled (system default behavior): Indicator light glows green only when battery is fully charged (battery level 100%) Indicator light glows amber when charger on hold. (battery level reaches charging threshold). Indicator light glows amber while charging. Below are string for title of switch that enables this settings
```
msgid "Change the behavior of charging indicator light"
```
[image]

## Dell Laptops
Some dell laptop when using library Dell command configure require bios password to change bios configuration and charging threshold / mode. Below are string displayed in settings
```
# Title of group of settings that as related to bios password required by Dell command configure 
msgid "Bios Authentication"

# Title for switch dell command center requires Bios password to change charging threshold
msgid "Requires Bios password validation to change mode/threshold"
# Description of switch informing to enable switch if Dell command configure required bios password to change settings
msgid "Enable if Dell command configure uses Bios (system) password to change mode/threshold"

#Title for Gtk Entry box informing to enter bios password
msgid "Enter Bios Password"

# Small description Note provided on the Gtk entry box informing that this password will store in gnome keyring
msgid "Note: Bios password will be stored in Gnome Keyring"

# Notification informing the password needs to be correct to change charging threshold
msgid "Apply correct Bios Password to set threshold."
```
[image]
[image]

## Thinkpad Laptop
These are settings enabled a force discharge feature in Thinkpad laptop. In normal case when laptop battery level is higher than charging threshold it will disconnect, the charger from the battery and power the laptop and the battery will discharge slowly (floating charge) until the battery level reaches the charging threshold level and may take days to discharging to charging threshold level.
Enabling "Force discharge" will disconnect the charger from the battery and laptop, forcing the laptop to discharge faster (few minutes or hours) and when the battery level reaches charging threshold the charger connects and powers the laptop whilst still disconnected to the battery. This feature is provided by enable force discharge in settings 

```
#A title for switch to enable force discharge button feature. Enabling this will add a button named "Force discharge" in quick settings to turn Force Discharge ON or OFF
msgid "Enable force discharge battery feature (Experimental)"

#Small description for the above GTK button
msgid "Enable option to force discharge battery when battery level higher than current threshold"

# A button in quick settings to turn ON/OFF force discharge
msgid "Force Discharge"
```
[image]
[image]

sabriunal, 2023-12-14:
Tranalation > Translation

ViBE-HU, 2023-12-16:
@maniacx i'm done with both builds.

maniacx, 2023-12-16:
Great, Thanks @sabriunal @ViBE-HU

Vistaus, 2023-12-16:
@maniacx Done! 🙂

maniacx, 2023-12-20:
Hello @sabriunal 
I use the below command to generate pot file.

```
xgettext \
    --add-comments="TRANSLATORS:" \
    --from-code=UTF-8 \
    --copyright-holder="maniacx@github.com" \
    --package-name="Battery Health Charging" \
    --output="po/Battery-Health-Charging.pot" \
    lib/*.js \
    preferences/*.js \
    ui/*.ui
```
However it get the following 7 warning, and after test it seem the warning shows up for each ui files
```
xgettext: warning: a fallback ITS rule file '/usr/share/gettext-0.22/its/gtkbuilder.its' is used; it may not be in sync with the upstream
```

Although nothing serious as all, as al strings in UI get added to pot file, i was just curious , how i can I fix this warning. Do you have any idea about this?

sabriunal, 2023-12-21:
I have no idea.

https://github.com/autotools-mirror/gettext/blob/master/gettext-tools/src/xgettext.c#L941C6-L954C26

maniacx, 2024-04-01:
Gnome 46. Translation. I think only one string.

@Valeria
@ViBE-HU
@Vistaus
@yakushabb
@viksok
@dalz
@ludrol

[image]

ViBE-HU, 2024-04-01:
@maniacx does this only shows up when the limit changed or it also shows up when changing the charging mode?

maniacx, 2024-04-01:
It shows while Changing mode and limit as well.

ViBE-HU, 2024-09-28:
just reviewing my translation and got stuck. i am on an actual lenovo legion machine but i have zero clue about the new _"Skip Threshold Verification"_ feature. could you give me some context or reference what exactly happen when _"verification failures"_ happen?

maniacx, 2024-09-28:
@ViBE-HU 
The way I built the extension, is it writes/applies threshold value, and then it reads back the threshold value, to verify if it applied correctly.

Some thinkpad, (also Asus) models. reads incorrect values due to bugs in firmware / kernel drivers.
For some thinkpad users
Writing threshold
`echo '60' | sudo tee /sys/class/power_supply/BAT0/charge_control_end_threshold
`
But after applying 60, reading back
`cat /sys/class/power_supply/BAT0/charge_control_end_threshold`
would read 200 or some random values.

The couple of user mentioned when the apply threshold, it works, battery stops charging if above 60.
Just reading the threshold value is a problem. and since the extension read and verifies, it notifies an error.

Enabling `Skip Threshold Verification` extension only writes the threshold, but doesnt read back and verify.

To features are difficult to explain using short description. Any suggestion in English too would be helpful

**Skip Threshold Verification**
and 
**Change battery indicator icon behavior**

ViBE-HU, 2024-09-28:
so as i understand _"Skip Threshold Verification"_ just ignores (i mean it does not even read/check) the false value and force overwrite it when the extension loads. am i right?

did not notice _"Change battery indicator icon behavior"_ before. but as i understand it is for showing when the charger is connecting. any other case it works the same, right?

maniacx, 2024-09-28:
> so as i understand _"Skip Threshold Verification"_ just ignores (i mean it does not even read/check) the false value and force overwrite it when the extension loads. am i right?
>
> did not notice [...]
Yes. you understood correctly on both

I have some few additional changes I have made, that I didnt push to Github yet. Do not translate and submit for now.
I will message you here, when it is ready and updated on Github. Still doing some test on the next version.

maniacx, 2024-10-04:
which one is better

**Skip Threshold Verification**
Certain models may read threshold values incorrectly, causing verification failures. Enabling this option will write new threshold values without verifying if they were applied correctly.

or

**Skip Threshold Verification**
"Enable if writing the threshold succeeds, but reading it fails, or if thresholds reset after sleep/hibernate"

Which one is better?

maniacx, 2024-10-04:
Ok I pushed it. I made slight changes to your suggestion. Thanks. You can update the strings.
Tomorrow I will add same feature for Asus as well, I didn't have the chance to  worked on it. But  string used will be the same. so there wont be any translation changes.

ViBE-HU, 2024-10-04:
cool. i still not see the new pot files in the repo. or you added via crowdin?

maniacx, 2024-10-04:
Nah. I forgot to update the pot.

ViBE-HU, 2024-10-04:
one more question. are these polkit related?

_Install check failed.
Install check timed out._

also where these appear?

_Sysfs node
Ectool
Framework Tool
Select configuration for charging mode
Select configuration used for changing charging mode/threshold._

my guess that these are all related with one feature and also vendor specific. but i have only old dells, so i never met with this feature. i guess a single screenshot could help me understand.

maniacx, 2024-10-04:
> Install check failed.
> Install check timed out.

yes this are notifiication, regarding extension checking installation of polkit and batteryhealthchargingctl, checking if they are updated. So whenever check fails or doesnt respond in time. It will notify about errors.

```
Sysfs node
Ectool
Framework Tool
Select configuration for charging mode
Select configuration used for changing charging mode/threshold.
```
Yes. **Sysfs** node are file path un sysfs like `/sys/class/power_supply/BAT0/charge_control_end_threshold ` so extension will just write threshold on these values. In latest kernel Chromebook, Framework Laptop and even Dells latest kernel will have the ability to change threshold using sysfs node. While earlier when kernel was not supporting sysfs node, package like Libsmbios, Dell Command Configure, Ectool, Framework were used

Similar to Libsmbios and Dell Command Configure
**Ectool** - Tool for Chromebook
**Framework** - Tool for Framework Laptops

And example for chromebook.
Similar option will be available for dell and framework as well
[image]

ViBE-HU, 2024-10-05:
i'm not sure but looks like you missed the GNOME42-44 branch :D

maniacx, 2024-10-05:
Yes still have to work on it. Changes are huge. You can commit the GNOME45 branch if you want now. I will merge it with GNOME42-44 branch when ready, or You might have to wait another week and do both when its ready.

closed 2026-11-06 by maniacx