maniacx/Battery-Health-Charging #53: Lenovo legion - show only main settings

Lukasz87, 2023-06-27:
Hello,

On Lenovo Legion 5 15ARH7H show only main setting without possibilities to change: 

[image]

System info: 
Fedora Linux 38 (Workstation Edition)
Gnome version: 44.2
Kernel: Linux 6.3.8-200.fc38.x86_64

maniacx, 2023-06-28:
Does you laptop supports threshold customization?

I think Lenovo IdeaPad/legion can be only enable and disable charging threshold (conservation mode) which is 60% on some older models or 80% on never models.

In Lenovo only Thinkpad supports customization of threshold.

maniacx, 2023-06-28:
Check the charging threshold paths, if you want to dig deeper.

Using ls commands.
For Lenovo IdeaPad/legion it is the.
`ls -l /sys/bus/platform/drivers/ideapad_acpi/VPC2004:00/conservation_mode`

If this path exist, you can set threshold using command line in terminal, echo 1 to enable/0 to disable
`echo 1 | sudo tee /sys/bus/platform/drivers/ideapad_acpi/VPC2004:00/conservation_mode`

All the charging threshold paths for different devices are mentioned in the Readme on GitHub.
https://github.com/maniacx/Battery-Health-Charging#lenovo-ideapad-legion

The extension automatically checks the charging threshold path, give customization options based on device.

closed 2023-07-04 by maniacx