maniacx/Battery-Health-Charging #54: smbios, cctk, BIOS password required

MasinAD, 2023-07-31:
When the BIOS is locked with a password, neither smbios nor cctk can change the battery settings unless the BIOS password is provided.

It would be nice for the extension to have it ask for the password and use it.
```
smbios-battery-ctl --password=PASSWORD
cctk --ValSetupPwd=PASSWORD
```
Of course, if there are reasons for locking the BIOS it might not be a good idea to just provide the password in cleartext somewhere in the filesystem. The extension could ask for the password on changes and offer to store it. Usually, such secrets get stored in the GNOME Keyring. The keyring itself is locked by the user password until login.

maniacx, 2023-07-31:
Sorry Masin. Unfortunately I don't have the time to add it this feature.
If you are using CCTK you can try this extension I think it has this feature of bios password.

https://extensions.gnome.org/extension/4545/dell-command-configure-menu/

maniacx, 2023-10-01:
Hello @MasinAD 
I was just looking into how to implement this.

I would like some information.
For cctk. 
Setting mode to express would be
/opt/dell/dcc/cctk --ValSetupPwd=<password> --PrimaryBattChargeCfg=express

What is the command line output displayed when you set express mode using the above command?
What is the output displayed if the bios password is incorrect?

Can I read the current threshold without the bios password?
/opt/dell/dcc/cctk  --PrimaryBattChargeCfg
or do I have to use the below command to read
 /opt/dell/dcc/cctk  --ValSetupPwd=<password>  --PrimaryBattChargeCfg

MasinAD, 2023-10-01:
Hi @maniacx, thanks for taking a look into this. I'll gladly provide the answers:

# cctk
> What is the command line output displayed when you set express mode using the above command?
```
$ sudo cctk --ValSetupPwd=correctpassword --PrimaryBattChargeCfg=Express
PrimaryBattChargeCfg=Express
```

> What is the output displayed if the bios password is incorrect?
```
$ sudo cctk --ValSetupPwd=wrongpassword --PrimaryBattChargeCfg=Express

The setup password provided is incorrect. Please try again.
```
Yes, there's a blank line before the response.

> Can I read the current threshold without the bios password?
> /opt/dell/dcc/cctk --PrimaryBattChargeCfg

Yes.
```
$ sudo cctk --PrimaryBattChargeCfg
PrimaryBattChargeCfg=Custom:50-80
```

> or do I have to use the below command to read
> /opt/dell/dcc/cctk --ValSetupPwd=password --PrimaryBattChargeCfg
No, that's not necessary.

# smbios-battery-ctl
While I am at it, I wanted to provide the corresponding answers for `smbios-battery-ctl` but although it queries the current mode (`smbios-battery-ctl --get-charging-cfg`) without issues, setting the mode with `smbios-battery-ctl --set-charging-mode=express --password=correctpassword` achieves nothing. The command reports back it had changed the mode but querying it afterwards still shows the previous mode.

For the sake of completeness here the output of `smbios-battery-ctl`:
Querying the current mode:
```
$ sudo smbios-battery-ctl --get-charging-cfg
Charging mode: express
```
Changing with correct password:
```
$  sudo smbios-battery-ctl --set-charging-mode=express --password=correctpassword
Charging mode has been set to: express
```
Changing with wrong password:
```
$ sudo smbios-battery-ctl --set-charging-mode=express --password=wrongpassword
Charging mode has been set to: express
```
As you can see, there's no difference.
```
$ sudo smbios-battery-ctl --set-charging-mode=express
Charging mode has been set to: express
```
And even without password parameter it's still the same response.

MasinAD, 2023-10-01:
@maniacx 
`cctk` outputs the mode names thus:
* Standard
* Express
* PrimAcUse
* Adaptive
* Custom

While `smbios-battery-ctl` outputs them like this:
* primarily_ac
* adaptive
* 'custom
* standard
* express

Maybe support for the latter works for someone else.

maniacx, 2023-10-01:
Thanks for the detailed update...
I will look into how to implement this.

maniacx, 2023-10-04:
Hello @MasinAD 
Which gnome version are you using?

MasinAD, 2023-10-04:
> Hello @MasinAD 
> Which gnome version are you using?

Hello @maniacx
I use gnome 44 on Arch Linux and gnome 42 on Ubuntu 22.04.3 or 42.5 as I've read it is called.

maniacx, 2023-10-04:
I have push a new branch https://github.com/maniacx/Battery-Health-Charging/tree/GN44-dell-bios-validation
You can install it from there.

* This should be compatible with Gnome42-44.
* I have implement Bios validation for cctk only (since libsmbios is kinda broken and nobody seems to be using it with bios password)
* The Bios password entered in settings will be stored in Gnome Keyring. 
* Created a seperate Tab for dell in settings and it will only appear if cctk is installed
* The extension uses exit codes to detect is bios password is incorrect or not provided (thanks for the info)

I have not tested it for Gnome 42-44, so incase of any silly errors, I apologise in advance.
let me know how it goes.

maniacx, 2023-10-04:
@MasinAD  Added a minor gui change for dell settings.
Try the latest commit

MasinAD, 2023-10-05:
@maniacx many thanks.

I tried the latest commit on my private Arch Linux Dell laptop. It works. I switched around a little and always checked with the CLI `cctk` if the changes had been applied, and they always were. What I haven't tried yet is changing the thresholds but I will test that in a few hours.

On Monday I'll be back in office, where I have some Dell laptops with Ubuntu 22.04. I'll test it there then, too.

I had some trouble getting the latest commit to work. Nothing serious, I figured it out. But to recap:
1. I saw the zip file and thought it would contain a distributable version of the branch. It didn't.
2. I copied all files and folders to the correct location, but Extensions complained about a missing schema file when I wanted to enable it.
3. I copied the gschemas.compiled from the zip file but then the settings dialog complained about the key `detected-libsmbios` not being part of the schema or something along the line.
4. Finally, I googled how to compile the schema … well, it was the easiest solution :-D. `glib-compile-schemas .` in the schemas directory.

I think the most annoying part of testing extensions is logging off and in again if I use Wayland sessions. Not your fault, though.

maniacx, 2023-10-05:
My apologies, that zip file was not  to be included. Correct way is to run ./install.sh

Take your time. No hurry. I won't be posting any update for until next week

maniacx, 2023-10-08:
@MasinAD 
So I tried to run the extension on my VM ubuntu 22.04 and seems like the extension preferences errored.
It is because I used ADW.PasswordEntryRow which was introduced in libadwaita 1.2 while Ubuntu 22.04 still uses libadwaita 1.1 or 1.0 i think.

Anyways I am using GTK.PassworkEntry now and Extension Preferences shows up without error.
And Althought GTK.PasswordEntry doesn't have caps lock indicator and Enter button, still somehow looks better than the ADW.PassordEntryRow.

Anyways I push the latest commit replacing ADW.PassordEntryRow with GTK.PassworkEntry. Kindly test the updated branch.

maniacx, 2023-10-12:
Hello @MasinAD 

I have pushed the commit to the main branches with some minor changes, into respective main branches for release. If everything works well on your end, I release an update.

For Gnome 42-44 test this branch.
https://github.com/maniacx/Battery-Health-Charging/tree/GN44-main

If you have already moved to Gnome45 then this is the branch
https://github.com/maniacx/Battery-Health-Charging/tree/GN45-main

MasinAD, 2023-10-13:
Hello @maniacx,

it works well on my private Dell Latitude with Arch Linux. I hadn't the opportunity to try it on a Ubuntu Dell this week because of being sick at home but I'll try ASAP and report back. As long as there were no regressions nobody should complain, I think. And if there are any bugs left we can pin them down in the next few rounds.

Thanks again for taking the time!

maniacx, 2023-10-17:
@MasinAD 
Now available on gnome shell extension website.
I will close this issue.
Thank for info and testing.

closed 2023-10-17 by maniacx

alexanderadam, 2024-06-04:
Do I understand this correctly that BIOS password support is built-in?
I can't find anything in the settings. :face_with_peeking_eye:

maniacx, 2024-06-05:
Yes. the extension support bios password only when using Dell Command Center (cctk)

Let me explain.

**1. If only Libsmbios is installed (Dell command center not installed)**
You will not see the option for Bios password setting (as with libsmbios bios password doesnt work)

**2. If only Dell command configure is installed (libsmbios not installed)**
You will see the bios password settings appear.

[image]

**3. If both are installed libsmbios and dell command configure.**
Then you will see another settings appear, that will ask you, which library you want to use to change charging threshold and you have to select what is working for you.

[image]

If you have selected Dell command configure, then Bios password settings will appear.