maniacx/Battery-Health-Charging #55: Error on "non-thinkpad"

E34rrsff, 2023-08-05:
I have a Lenovo Yoga/Ideapad & there's this error that I don't know what to do with. Any help please?

[image]

maniacx, 2023-08-05:
Which distro are you using?

Also can you check if you can set and read charging threshold manually using terminal. Try these command

`echo 1 | sudo tee /sys/bus/platform/drivers/ideapad_acpi/VPC2004:00/conservation_mode`
`cat /sys/bus/platform/drivers/ideapad_acpi/VPC2004:00/conservation_mode`
`echo 0 | sudo tee /sys/bus/platform/drivers/ideapad_acpi/VPC2004:00/conservation_mode`
`cat /sys/bus/platform/drivers/ideapad_acpi/VPC2004:00/conservation_mode
`

Send me the output of this.

E34rrsff, 2023-08-05:
It's the same directories my computer has & I've made a script to change it for myself so I know it does work on mine

```
user@pc:~$ distro
Name: Ubuntu 23.04
Version: 23.04 (lunar)
Codename: lunar
```
```
user@pc:~$ ls /sys/bus/platform/drivers/ideapad_acpi/VPC2004\:00/conservation_mode
/sys/bus/platform/drivers/ideapad_acpi/VPC2004:00/conservation_mode
```

Now that I'm checking it again, the extension is working & it's changing the contents of the file to 1/0, but it's still is showing me the error notification. Maybe it detected the wrong device?

```
user@pc:~$ cat /sys/bus/platform/drivers/ideapad_acpi/VPC2004\:00/conservation_mode 
1
```
Then after toggling it again:
```
user@pc:~$ cat /sys/bus/platform/drivers/ideapad_acpi/VPC2004\:00/conservation_mode 
0
```

maniacx, 2023-08-05:
When do you get the error?
1. Is it when you change conservation_mode from extension?
2. Or does it show during logging in from lockkscreen, sleep, or power off??

E34rrsff, 2023-08-06:
Only when I'm toggling conservation mode with the extension

maniacx, 2023-08-06:
I have added debugging message to this trial version. Install it.
[debug.zip](https://github.com/maniacx/Battery-Health-Charging/files/12268684/debug.zip)

if installation fails with msgfmt error install gettext. (sudo apt install gettext)

After installing thre trial extension logout and relogin

Open new terminal and paste this command
`journalctl -f -o cat /usr/bin/gnome-shell`
You will see a live logging.
Toggle 4 time conservation mode using extension and there will be some output on the terminal.
Paste it here

E34rrsff, 2023-08-07:
```
user@pc:~$ journalctl -f -o cat /usr/bin/gnome-shell
GNOME Shell terminated with an error: Xwayland exited unexpectedly
DING: Detected async api for thumbnails
DING: DBus interface for Switcheroo control (net.hadess.SwitcherooControl) is now available.
DING: DBus interface for Nautilus (org.gnome.Nautilus.FileOperations2) is now available.
DING: DBus interface for Nautilus (org.freedesktop.FileManager1) is now available.
DING: DBus interface for Gvfs daemon (org.gtk.vfs.Metadata) is now available.
DING: ** Message: 16:08:34.590: Connecting to org.freedesktop.Tracker3.Miner.Files
DING: GNOME nautilus 44.2.1
Received notification for window. 0 notifications remaining.
DING: DBus interface for File-roller (org.gnome.ArchiveManager1) is now available.
argv = pkexec,/usr/local/bin/batteryhealthchargingctl-user,LENOVO,1
stdout = 
stderr = 
Battery Health Charging: xxxxxxxxxxxx status = 0
Battery Health Charging: xxxxxxxxxxxx readMode = 0
Battery Health Charging: xxxxxxxxxxxx condition not met
argv = pkexec,/usr/local/bin/batteryhealthchargingctl-user,LENOVO,0
stdout = 
stderr = 
Battery Health Charging: xxxxxxxxxxxx status = 0
Battery Health Charging: xxxxxxxxxxxx readMode = 1
Battery Health Charging: xxxxxxxxxxxx condition not met
argv = pkexec,/usr/local/bin/batteryhealthchargingctl-user,LENOVO,1
stdout = 
stderr = 
Battery Health Charging: xxxxxxxxxxxx status = 0
Battery Health Charging: xxxxxxxxxxxx readMode = 0
Battery Health Charging: xxxxxxxxxxxx condition not met
argv = pkexec,/usr/local/bin/batteryhealthchargingctl-user,LENOVO,0
stdout = 
stderr = 
Battery Health Charging: xxxxxxxxxxxx status = 0
Battery Health Charging: xxxxxxxxxxxx readMode = 1
Battery Health Charging: xxxxxxxxxxxx condition not met
```

maniacx, 2023-08-08:
I failed to add a debug line. Silly me.  Anyways, the extension as soon as it write threshold on /sys/...../conservation_mode it will read and verify if its written value has accepted. Probably it seems like you laptop takes time to update the value.

So kindly help me with another test and give me the output of these 2 command in terminal

`echo 1 | sudo tee /sys/bus/platform/drivers/ideapad_acpi/VPC2004:00/conservation_mode && cat /sys/bus/platform/drivers/ideapad_acpi/VPC2004:00/conservation_mode && cat /sys/bus/platform/drivers/ideapad_acpi/VPC2004:00/conservation_mode`

`echo 0 | sudo tee /sys/bus/platform/drivers/ideapad_acpi/VPC2004:00/conservation_mode && cat /sys/bus/platform/drivers/ideapad_acpi/VPC2004:00/conservation_mode && cat /sys/bus/platform/drivers/ideapad_acpi/VPC2004:00/conservation_mode`

Also I have created another debug extension, in this if verifying fails it will verify again after 200ms.
Kindly install and test and monitor/send me the logs using `journalctl -f -o cat /usr/bin/gnome-shell
[debug-2.zip](https://github.com/maniacx/Battery-Health-Charging/files/12286051/debug-2.zip)
`

E34rrsff, 2023-08-08:
```
user@pc:~$ journalctl -f -o cat /usr/bin/gnome-shell 
stderr = 
Battery Health Charging: xxxxxxxxxxxx status = 0
Battery Health Charging: xxxxxxxxxxxx readMode = 0
Battery Health Charging: xxxxxxxxxxxx condition not met
argv = pkexec,/usr/local/bin/batteryhealthchargingctl-user,LENOVO,0
stdout = 
stderr = 
Battery Health Charging: xxxxxxxxxxxx status = 0
Battery Health Charging: xxxxxxxxxxxx readMode = 1
Battery Health Charging: xxxxxxxxxxxx condition not met
```
```
user@pc:~$ file=/sys/bus/platform/drivers/ideapad_acpi/VPC2004\:00/conservation_mode 
user@pc:~$ cat $file
0
user@pc:~$ echo 1 | sudo tee $file && cat $file
1
0
user@pc:~$ cat $file
1
```

That seems odd to me, I think it might take longer than 200ms because it still gives me the notification. Do you know if it taking that long is normal?

E34rrsff, 2023-08-08:
Oh, I copied the fjust the files directly from the zip, not the ones in the `debug` folder. Copying the ones in the `debug` folder made it go away now.

maniacx, 2023-09-02:
@E34rrsff Hi
Can you Please test this new version for me. 
[test.zip](https://github.com/maniacx/Battery-Health-Charging/files/12504926/test.zip)

I have added 200ms reverification, but it is a litttle different than debug2.zip that I have sent previosly.
If everything works well, I will upload for review and publish  on Gnome-extension website

Thanks

E34rrsff, 2023-09-05:
I fully restarted my computer to make sure nothing is cached or whatever, but I'm getting the "Charging threshold not  updated" error again like before. Also I checked if there were any logs with `journalctl` just in case, but yeah there's nothing showing up. It is still toggling the `conservation_mode` file's contents with 1 & 0, though.

maniacx, 2023-09-07:
I made a mistake with test.zip
Try this one 
[test2.zip](https://github.com/maniacx/Battery-Health-Charging/files/12544304/test2.zip)
Hopefully no more mistake.

E34rrsff, 2023-09-07:
Yeah it seems to work now, although I don't know if the gschemas.compiled file is intentionally meant to be excluded in that zip 'cause I got an error that it was missing before

maniacx, 2023-09-07:
Thanks for testing. I will take check on the schema

E34rrsff, 2023-09-07:
Thanks for helping, the update is working perfectly

closed 2023-09-07 by maniacx