maniacx/Battery-Health-Charging #60: Add compatibility to Razer

pramodhrachuri, 2023-09-19:
You add compatibility to razer by using the following repo
https://github.com/Razer-Linux/razer-laptop-control-no-dkms

Example commands
```
pramodh@darkhold:~ ➤ razer-cli read bho
RES: GetBatteryHealthOptimizer { is_on: true, threshold: 70 }
Battery health optimization is on with a threshold of 70
pramodh@darkhold:~ ➤ razer-cli write bho on 70
RES: SetBatteryHealthOptimizer { result: true }
Battery health optimization is on with a threshold of 70
pramodh@darkhold:~ ➤ razer-cli write bho off
RES: SetBatteryHealthOptimizer { result: true }
Successfully turned off bho
pramodh@darkhold:~ ➤ razer-cli write bho on 70
RES: SetBatteryHealthOptimizer { result: true }
Battery health optimization is on with a threshold of 70

```

maniacx, 2023-09-20:
Ok. Looking into it.
I just to confirm.  Let me know if the statement i make below is correct.

1. There is no start threshold . Only end threshold

2.bho off =  no threshold =  threshold at 100% 

3. bho on = (what ever you set the threshold). 

4. value of threshold can be any value between 50 to 80. Can I set value to 51?

5. What is you path to the charger?
usually is it at   **/sys/class/power_supply/** could be example AC0 or AC1 or ADP0 or may be some custom name.
it should have a node/file called **online** inside which reads 1 when charger is plugged in or 0 when charger is disconnected. 

Also what version of Gnome are you using? I will send a test version when it is ready.
Let me know.

pramodhrachuri, 2023-09-20:
1. Correct. No start threshold
2. Correct.
3. Correct.
4. Correct. The threshold value must be a multiple of five between 50 and 80. Both included.
5. Correct. /sys/class/power_supply/AC0/online is the path

You got all of them correct, lol.

I use 42.9 on Ubuntu 22.04.3.

maniacx, 2023-09-20:
Also I want to know what does read command output when bho is off. Can you run the 3 command  below in that order/sequence and give me the output of the  command

razer-cli write bho on 70
razer-cli write bho off
razer-cli read bho

pramodhrachuri, 2023-09-20:
Ah. I forgot to include that in the first comment. Here you go

```
pramodh@darkhold:~➤ razer-cli write bho on 70
RES: SetBatteryHealthOptimizer { result: true }
Battery health optimization is on with a threshold of 70

pramodh@darkhold:~➤ razer-cli write bho off
RES: SetBatteryHealthOptimizer { result: true }
Successfully turned off bho

pramodh@darkhold:~➤ razer-cli read bho
RES: GetBatteryHealthOptimizer { is_on: false, threshold: 80 }
Battery health optimization is off
```

pramodhrachuri, 2023-09-20:
Works well! I have cross-checked if the limits are being applied using the read command, which works!

A few nitpicks
1. I cannot set "50%" in maximum lifespan mode. It looks like the limits are not set properly. I have changed the variable to the following.
```
        this.endBalancedRangeMax = 80;
        this.endBalancedRangeMin = 65;
        this.endMaxLifeSpanRangeMax = 80;
        this.endMaxLifeSpanRangeMin = 50;
```
2. It made me give it sudo permissions. I don't think it is required for razer-cli.

I will do the reboot, logout-login testing and update you.

pramodhrachuri, 2023-09-20:
It seems pretty stable even after three reboots and a couple of logout-logins. 
Thank you very much!
Let me know whenever you push the update to extensions.gnome.org. I will let others in the Razer-Linux community

pramodhrachuri, 2023-09-24:
Thanks for adding compatibility to Razer in the main branch. I just got the update from Gnome Extensions.
However, some parameters still need to be changed in [Razer.js](https://github.com/maniacx/Battery-Health-Charging/blob/main/devices/Razer.js).

Razer-cli doesn't need root access.
```
        this.deviceNeedRootPermission = true;
```

The supported range of percentages is 50 to 80, both included.
```
        this.endBalancedRangeMax = 80;
        this.endBalancedRangeMin = 65;
        this.endMaxLifeSpanRangeMax = 80;
        this.endMaxLifeSpanRangeMin = 45;
```

maniacx, 2023-09-24:
> Razer-cli doesn't need root access.

I will have look at this. And try to implement it at the next update.

>The supported range of percentages is 50 to 80, both included.

I didnt understand, can you explain it better?
**this.endMaxLifeSpanRangeMin**  is set already set  50 and not 45

https://github.com/maniacx/Battery-Health-Charging/blob/4a8fe952e4c7c5c04ad9d492fdc03a03ef9664d9/devices/Razer.js#L34

Do you also want me to change  **this.endBalancedRangeMin**  to 50?

pramodhrachuri, 2023-09-24:
> this.endMaxLifeSpanRangeMin is set already set 50 and not 45
Sorry my bad. Confused with the behaviour of the extension due to the old version. The current endMaxLifeSpanRangeMin is correct. But endBalancedRangeMax and endMaxLifeSpanRangeMax have to be changed to 80. 
My apolories for the confusion.

I can experiment with deviceNeedRootPermission and let you know if the extension works without it.

maniacx, 2023-09-24:
No. I wont work.

maniacx, 2023-09-24:
Lol. I am blind. I didnt realise it was 85.. and not 80

maniacx, 2023-09-24:
OK. Uninstall Polkit rules from extension settings since razer-cli doesnt need root.
Uninstall the extension.
Log off
Log in

Install [test-no-root.zip](https://github.com/maniacx/Battery-Health-Charging/files/12707925/test-no-root.zip)
Check if it works.
It should not show any notification regarding polkit rules
In extension settings, It should not display an option to install / remove polkit rules.
Let me know

pramodhrachuri, 2023-09-24:
Works!
There is no notification regarding polkit rules and no option in the settings.

maniacx, 2023-09-26:
I have uploaded new verion on gnome extension site.
If all works well, kindly close this issue.
Thanks for the information and testing

closed 2023-09-26 by pramodhrachuri