maniacx/Battery-Health-Charging #61: TypeError: headerbar.pack_start is not a function

luna-duclos, 2023-09-21:
When opening the settings menu for the extension, I get the following error:

```
TypeError: headerbar.pack_start is not a function

Stack trace:
  addMenuToHeader@/home/luna/.local/share/gnome-shell/extensions/Battery-Health-Charging@maniacx.github.com/preferences/menu.js:24:15
  addMenu@/home/luna/.local/share/gnome-shell/extensions/Battery-Health-Charging@maniacx.github.com/preferences/menu.js:13:20
  fillPreferencesWindow@/home/luna/.local/share/gnome-shell/extensions/Battery-Health-Charging@maniacx.github.com/prefs.js:26:12
  _init@resource:///org/gnome/Shell/Extensions/js/extensionPrefsDialog.js:27:29
  ExtensionPrefsDialog@resource:///org/gnome/Shell/Extensions/js/extensionPrefsDialog.js:10:4
  OpenExtensionPrefsAsync@resource:///org/gnome/Shell/Extensions/js/extensionsService.js:124:33
  async*_handleMethodCall@resource:///org/gnome/gjs/modules/core/overrides/Gio.js:373:35
  _wrapJSObject/<@resource:///org/gnome/gjs/modules/core/overrides/Gio.js:408:34
  run@resource:///org/gnome/Shell/Extensions/js/dbusService.js:186:20
  main@resource:///org/gnome/Shell/Extensions/js/main.js:22:13
  run@resource:///org/gnome/gjs/modules/script/package.js:206:19
  start@resource:///org/gnome/gjs/modules/script/package.js:190:8
  @/usr/share/gnome-shell/org.gnome.Shell.Extensions:1:17
  
```

maniacx, 2023-09-21:
Which version of gnome are you using? 45? Which OS?

maniacx, 2023-09-22:
Could test this test2.zip on posted here
https://github.com/maniacx/Battery-Health-Charging/issues/63#issuecomment-1730504992

if you run into msgfmt error, install **gettext**

luna-duclos, 2023-09-23:
Hello!

I tried the test2.zip from the other issue and that indeed works and the menu that was broken appears properly as well.
Here is my OS/distro/gnome info:

OS: Arch Linux
OS Build ID: rolling
GNOME Vrsion: 44.5
Windowing system: Wayland

maniacx, 2023-09-23:
Thank you for reporting the bug and testing.

closed 2023-09-26 by maniacx