maniacx/Battery-Health-Charging #63: Extension settings show an error on 2020 Asus Zephyrus G14

tolis626, 2023-09-21:
As the title says. I installed the extension, it notifies me about installing Polkit (Which is already installed) and then throws the following error.
`The settings of extension Battery-Health-Charging@maniacx.github.com had an error:
```
TypeError: headerbar.pack_start is not a function
```

Stack trace:
```
addMenuToHeader@/home/tolis/.local/share/gnome-shell/extensions/Battery-Health-Charging@maniacx.github.com/preferences/menu.js:24:15
addMenu@/home/tolis/.local/share/gnome-shell/extensions/Battery-Health-Charging@maniacx.github.com/preferences/menu.js:13:20
fillPreferencesWindow@/home/tolis/.local/share/gnome-shell/extensions/Battery-Health-Charging@maniacx.github.com/prefs.js:26:12
_init@resource:///org/gnome/Shell/Extensions/js/extensionPrefsDialog.js:27:29
ExtensionPrefsDialog@resource:///org/gnome/Shell/Extensions/js/extensionPrefsDialog.js:10:4
OpenExtensionPrefsAsync@resource:///org/gnome/Shell/Extensions/js/extensionsService.js:124:33
async*LaunchExtensionPrefsAsync@resource:///org/gnome/Shell/Extensions/js/extensionsService.js:110:14
_handleMethodCall@resource:///org/gnome/gjs/modules/core/overrides/Gio.js:373:35
_wrapJSObject/<@resource:///org/gnome/gjs/modules/core/overrides/Gio.js:408:34
run@resource:///org/gnome/Shell/Extensions/js/dbusService.js:186:20
main@resource:///org/gnome/Shell/Extensions/js/main.js:22:13
run@resource:///org/gnome/gjs/modules/script/package.js:206:19
start@resource:///org/gnome/gjs/modules/script/package.js:190:8
@/usr/share/gnome-shell/org.gnome.Shell.Extensions:1:17
```
`
I am running Garuda Linux with Gnome. Thanks!

maniacx, 2023-09-21:
Hmm. You are the second user who has report this. Which Gnome version are you using?

tolis626, 2023-09-21:
44.5. At first I thought it might be related to the g14 specific stuff I've added to, among others, control battery charge limits, but that would make sense if the extension didn't do what it was supposed to. Now it's not even showing the settings. I tried removing polkit but it breaks quite a few dependencies, so I decided to follow the instructions and report the issue here before I do anything rash.

Also thanks for your work!

maniacx, 2023-09-21:
Hey can you test this.
[test1.zip](https://github.com/maniacx/Battery-Health-Charging/files/12694028/test1.zip)

It is related to the menu that takes you to the project page/report bug/ etc.
I have disabled it for now in this test1.zip
Install it using but running ./install.sh from terminal
if it throw an error regarding msgfmt install gettext.
sudo apt install gettext or equivalent for Garuda Linux

tolis626, 2023-09-21:
Yup, that worked. Like, instantly, no problems. Ran the script, logged out, logged back in, it was working, prompted me to install the polkit thingy, logged out again, logged back in, all is fine. So whatever the problem was, you solved it (honestly, the details are a bit over my head).

It was the first time installing it. It's been a while since I've used Gnome and so far I was using the ROG Control Center to control charging thresholds, but it was rather cumbersome and problematic for me, so I searched for alternatives and here I am. Right now it works like a charm, it's discreet and it's built right into the shell, love it. Thank you!

PS : If there's anything else you'd want me to check I'd be happy to help. Otherwise I'm gonna mark this as resolved (I think that's what the "close with comment button does", no?)

maniacx, 2023-09-21:
I will investigate and let you know, if I need any further test / or info
As I am not experiencing this issue but I am still on Gnome 44.3 (Ubuntu 23.04). So still need to find out what is breaking it.

Could you also check what is your GTK version from terminal using

`gtk-launch --version`

tolis626, 2023-09-21:
Sure, it's 3.24.38.

maniacx, 2023-09-21:
[test2.zip](https://github.com/maniacx/Battery-Health-Charging/files/12694243/test2.zip)

Also test this version. If all goes well you should be able to see this menu.
[image]

tolis626, 2023-09-22:
Sorry I'm late, I had to go to bed last night. So, I installed the test 2 version on top of the old version (just ran install.sh which finished successfully, logged out, logged back in), no such menu exists. Should I maybe remove the old version and install this one?

maniacx, 2023-09-23:
As long as you have installed and logout - login it should work.

Is this icon not displays in your header bar for extension settings?
[image]

tolis626, 2023-09-23:
[image]
Yup, not there.

closed 2023-09-26 by maniacx