maniacx/Battery-Health-Charging #68: Install script error on NixOS

Radeox, 2023-11-23:
Hello!
I don't know if it's a NixOS related issue or something else but when I try to install the required scripts using the extension menu I get an "Unexpected error".

I've had a look at the install scripts in `tool` directory I tried to run it with `sudo bash installer.sh install` and it successfully creates 2 files:

* `/usr/local/bin/batteryhealthchargingctl-root`
* `/etc/polkit-1/rules.d/10-dem.batteryhealthcharging.setthreshold-root.rules`

Then it exits normally without errors but when I try to enable the extension or logging in again I get the same error that warns me to install the scripts.

I tried running it with the `--tool-user radeox` option but with the same results.

Maybe I'm missing something but my laptop should be supported and I verified that I can manually set conservative_mode from shell.

maniacx, 2023-11-24:
Hello Radeox. I was curious, so I fired up Virtual Machine and installed NixOS. This was my first time using NixOS, and wow. I was really struggling with.
Even simple task, like installing gettext, or getting logs from gnome-shell, I wasnt successfull.
Nothing is straight forward and after experimenting a bit, I tried to iron out issue with the extension one by one. here what I found.

First I found there was is no "/usr/local/bin/" folder. But seems like the installer.sh creates if it installer works.

Secondly I found that none of the script runs directly due to bash enviroment, hence the extension fails to install.
In other system, bash script can executed  directly, which this extension relies on
Example you run ```/usr/local/bin/batteryhealthchargingctl-radeox``` or ``` ./installer.sh --tool-user radeox install```. On Nixos it wont run and need to prefix command with bash or sh
Workaround was to edit shebang on the top of these to files to run is without bash prefix 
Need to change from ```#!/bin/bash``` to ```#!/usr/bin/env bash```. After doing this installation of Polkit from extension preference was possible.

And third Polkit rules are added differently it seems. https://nixos.wiki/wiki/Polkit
I did tried to add the rule to /etc/nixos/configuration.nix but it still wont apply.

Just to rule out the possiblilty of issue my extension script,  I experimented polkit with ```fdisk -l``` command, which requires root permission to get information on disk drives.

In my system **Fedora** in terminal typing ```pkexec /usr/bin/fdisk -l``` should ask for password and execute it.
I changed the contents of ```/usr/local/bin/batteryhealthchargingctl-radeox``` to this
```
polkit.addRule(function(action, subject) {
    if (action.id == "org.freedesktop.policykit.exec" &&
        action.lookup("program") == "/usr/bin/fdisk" &&
        subject.isInGroup("radeox"))
    {
     	return polkit.Result.YES;
    }
});
```
Now in terminal if I run ```pkexec /usr/bin/fdisk -l```, it will execute without the password with root previleges.

I tried the same as per **Nixos** documentation by adding the rule to /etc/nixos/configuration.nix but couldnt get pkexec fdisk command working without password

```
  security.polkit.enable = true
  security.polkit.extraConfig = ''
    polkit.addRule(function(action, subject) {
      if (action.id == "org.freedesktop.policykit.exec" &&
          action.lookup("program") == "/nix/store/yhd9h9d140z2ydh9j4h8s2lmxihy29g8-system-path/bin/fdisk" &&
          subject.isInGroup("radeox"))
      {
          return polkit.Result.YES;
      }
    })
  '';
```

Finally. Some hurdles preventing me to do any sort of debugging in NixOS as I could not figure out how to 
1. install gettext (Command in fedora ```sudo dnf install gettext```)
2. Read gnome-shell logs (Command in fedora ```journalctl -f -o cat /usr/bin/gnome-shell```
3. Copy files from Fedora to NixOS on Gnome Boxes VM (NixOS have even blocked copy/pasting and sending files from host to guest).
4. Offcourse how to correctly apply polkit rules.

Radeox, 2023-11-24:
1. All the packages are usually managed from your `configurations.nix` but you can install them the "classic" way using something like `nix-env -iA nixos.gettext` (You can search for packages [here](https://search.nixos.org/packages?channel=23.05) if needed)
2. Since all the packages are links here I think you should use the link to the current system gnome-shell, something like this `sudo journalctl -f -o cat /run/current-system/sw/bin/gnome-shell`
3. Never tried running NixOS in VM so can't help you with that
4. I'll try to look into the polkit issue but I think this should be done inside the Nix package instead of changing the `configuration.nix`; There is a package already for your extension [here](https://mynixos.com/nixpkgs/package/gnomeExtensions.battery-health-charging). Maybe you could contact the maintainer of that package

piegamesde, 2023-11-24:
Hi, as a rule of thumb, if an extension is not working properly on NixOS, this is almost always a packaging issue, especially when the extension interacts with the rest of the system beyond GNOME itself. So please open a downstream issue at the Nixpkgs repo instead.

The installer script cannot work on NixOS by design, as folders like `/etc` are intentionally not writable by applications. Instead, the declarative configuration management through the module system is used instead. I honestly don't know what the best way to handle this would be here, although I wouldn't consider it an unsolvable issue.

maniacx, 2023-11-25:
@Radeox @piegamesde 

I got extension working on my VM running NixOS.
Had to do some minor changes with the extension.
You can check and download on nixos branch
https://github.com/maniacx/Battery-Health-Charging/tree/nixos

Apart from that also need to add polkit rule to nixos configurator file. I have made a nix configuration file which you can import. nix file and readme with insrtuctions are available here: 
https://github.com/maniacx/Battery-Health-Charging/tree/nixos-configuration

Let me know if it works and if the instructions in readme are correct or can be improved.

maniacx, 2023-11-26:
Hello @Radeox 

Update. You do not need to add any additional coniguration to nixos. I have updated the extension brancg nixos and it should work out of the box without the need of any configuration file as mentioned in my previos post.

I turns out that the polkit rules installed by the extension were applied nixos (I was under the impression it didnt), but the rules where incorrect , well for nixos.
The previous polkit rules checked for a group name radeox(user) as in all other fedora will have a group named radeox(user), but in nixos this group is not added. 
So I changed the polkit rules to check for the user instead.

Try to install the updated version of extension for here
https://github.com/maniacx/Battery-Health-Charging/tree/nixos

If you have installed the "gnome-ext-battery-health-charging.nix" as mentioned in the previous post, please delete it and remove the imports from configuration.nix, rebuild the system and test.

Radeox, 2023-11-27:
Hi @maniacx

I've just removed the extension and reinstall it from the new `nixos` branch but I get the error `TThe extensioons in incompatible with the current GNOME version` and when I open the related settings menu I get 

```
SyntaxError: import declarations may only appear at top level of a module

Stack trace:
  _init@resource:///org/gnome/Shell/Extensions/js/extensionPrefsDialog.js:23:33
  ExtensionPrefsDialog@resource:///org/gnome/Shell/Extensions/js/extensionPrefsDialog.js:10:4
  OpenExtensionPrefsAsync@resource:///org/gnome/Shell/Extensions/js/extensionsService.js:124:33
  async*_handleMethodCall@resource:///org/gnome/gjs/modules/core/overrides/Gio.js:373:35
  _wrapJSObject/<@resource:///org/gnome/gjs/modules/core/overrides/Gio.js:408:34
  run@resource:///org/gnome/Shell/Extensions/js/dbusService.js:186:20
  main@resource:///org/gnome/Shell/Extensions/js/main.js:22:13
  run@resource:///org/gnome/gjs/modules/script/package.js:206:19
  start@resource:///org/gnome/gjs/modules/script/package.js:190:8
  @/nix/store/qp8wzqrlgk6b1a4gh6px8p3j08dvzbrd-gnome-shell-44.2/share/gnome-shell/.org.gnome.Shell.Extensions-wrapped:7:17
```

Do you have a solution for this? 🤔

maniacx, 2023-11-27:
Which version of gnome are you using?

Radeox, 2023-11-27:
Gnome 44.2

Radeox, 2023-11-27:
Okay my bad, I was on the wrong branch (Gnome-45), I don't know why I was thinking that when cloning it selects the branch you are in on github. Now I reinstalled the extension from the correct branch and it's working correctly! 👍🏻

maniacx, 2023-11-27:
> Okay my bad, I was on the wrong branch (Gnome-45), I don't know why I was thinking that when cloning it selects the branch you are in on github. Now I reinstalled the extension from the correct branch and it's working correctly! 👍🏻

Ok. Good to know.

 In few days I might release  the version nixos on gnome extension site. Just waiting on another finalizing another feature users request and will release them together.

Thanks for reporting and testing.

Radeox, 2023-11-27:
Thanks to you for the effort in the debugging and for this fix! 👏🏻 When you release it on gnome extensions the nix package should be updated too, I don't know if @piegamesde can update it as maintainer or else.

closed 2023-12-08 by maniacx