maniacx/Battery-Health-Charging #74: For ThinkPads with Power Bridge -- Manage discharge order

eriklautenschlaeger, 2024-01-20:
## Problem description
* Some ThinkPads are equipped with a Power Bridge. They have an internal battery and an external battery.
There is a known issue with some models where the internal battery discharges before the external--which counteracts the whole point of the Power Bridge.

## Conceived solution
* Manipulate the /sys/class/power_supply/BAT0/charge_behaviour parameter accordingly.
Test protocol as stated on [Reddit](https://www.reddit.com/r/thinkpad/comments/144qh70/thinkpad_dual_battery_linux_gnome/).

Test pending.

eriklautenschlaeger, 2024-01-20:
Working on this over [here](https://gist.github.com/eriklautenschlaeger/ce0707af5d4887e3a98206b48a91c813). 
I will have this finished up by tomorrow. If you see something dumb or need anything else just shoot a comment. :-)

eriklautenschlaeger, 2024-01-20:
I think this can be cut short: 
After testing I can say that the behavior of "force-discharge" for a given battery where this is applied is that the battery starts discharging with around 11000mW immediately until it reaches 0% -- regardless of the AC status. 

Let me know if I should test anything else.

maniacx, 2024-01-20:
Anyways I really apologize, I may have deleted the branch from my private repo as well. And the changed were huge. I dont think I will have to time work on this again.

```
# echo "force-discharge" | sudo tee /sys/class/power_supply/BAT1/charge_behaviour
force-discharge
tee: /sys/class/power_supply/BAT1/charge_behaviour: Input/output error
```
also with such errors, i may be harder to get this working.

I really apologize, I should have check first if I deleted it the branch , before asking you to test.

But atleast now you know that can control (although buggy)  which battery the laptop should use when unplugged.
Make bash script to make it easier to switch between batteries.

Oh I forgot, u have TLP, i think it has simpler command to trigger force discharge

eriklautenschlaeger, 2024-01-20:
No worries at all -- my conclusion was that this won't work anyway :-D

maniacx, 2024-01-20:
> the auto behavior can't be overwritten while AC is unplugged.

Oh I somehow missed that in your earlier post.  So it cannot be done.
I will close the issue.

closed 2024-01-20 by maniacx