maniacx/Battery-Health-Charging #76: General Question

Mtillmann, 2024-02-06:
I am wondering if my installation is running correctly. I've set a threshold of 85% which is reflected in `/sys/class/power_supply/BATT/charge_control_end_threshold`. However gnome's native battery charge percentage still shows 100%. Is this working as intended?

maniacx, 2024-02-06:
if you set threshold to 85% when you battery level is 100%, it will discharge the battery at a extremely slow rate. and may take several hours to discharge from 100% to 85%. One way of finding out if threshold actually work is to discharge the battery level below 90% by disconnecting the charger. once below 90% connect the charger and verify that the battery level doesn't increase.

Another way of checking is by running in terminal. 
`cat /sys/class/power_supply/BATT/status`
It should report as NotCharging.

Another way is to use upower command `upower -d`
Under Device: /org/freedesktop/UPower/devices/battery_BATT > state should `not` be `charging` . I could be in `pending-charge` state or `discharging` state. Athough if battery level is  99/100% it might report `fully-charged` instead of any other states `discharging` or  `pending-charge`

Mtillmann, 2024-02-07:
Thanks for the reply. As I suspected `charge_control_end_threshold` is completely ignored. I let the charge drop below 85, then plugged in the charger and it immediately went beyond 85% while the state is "charging". I think this is an issue with my asus laptop as everything is terrible and nothing works with it. Thanks for work and effort!

closed 2024-02-07 by Mtillmann