maniacx/Battery-Health-Charging #78: Apple Intel Not working

maniacx, 2024-02-19:
Hello gealum.

Can you post the output of these two commands in terminal / console

```
ls -l /sys/class/power_supply/BAT0/charge_control_end_threshold
ls -l  /sys/module/applesmc
```

kaveht, 2024-02-19:
Hello maniacx,
here are the outputs:
`$ ls -l  /sys/module/applesmc
total 0
-r--r--r-- 1 root root 4096 Feb 19 14:35 coresize
drwxr-xr-x 2 root root    0 Feb 19 12:17 drivers
drwxr-xr-x 2 root root    0 Feb 19 12:17 holders
-r--r--r-- 1 root root 4096 Feb 19 14:35 initsize
-r--r--r-- 1 root root 4096 Feb 18 18:37 initstate
drwxr-xr-x 2 root root    0 Feb 19 12:17 notes
-r--r--r-- 1 root root 4096 Feb 19 14:35 refcnt
drwxr-xr-x 2 root root    0 Feb 19 12:17 sections
-r--r--r-- 1 root root 4096 Feb 19 14:35 srcversion
-r--r--r-- 1 root root 4096 Feb 19 14:35 taint
--w------- 1 root root 4096 Feb 18 18:37 uevent
-r--r--r-- 1 root root 4096 Feb 19 14:35 version
`
`ls -l /sys/class/power_supply/BAT0/charge_control_end_threshold
-rw-r--r-- 1 root root 4096 Feb 18 18:37 /sys/class/power_supply/BAT0/charge_control_end_threshold
`

maniacx, 2024-02-19:
You seem to have the correct path. I need to get more details. I have added some logs to test.zip. Follow the instructions

1 Please remove/uninstall this extension. 

2 Run command the command to reset gsetting for the extension
```
gsettings --schemadir /home/$USER/.local/share/gnome-shell/extensions/Battery-Health-Charging@maniacx.github.com/schemas reset-recursively org.gnome.shell.extensions.Battery-Health-Charging
```

3. Install gettext.  (Require to install extension manually)
```
sudo apt install gettext
```

4. Install the test.zip [test.zip](https://github.com/maniacx/Battery-Health-Charging/files/14336749/test.zip)
Download. In the zip right-click on Install.sh and run as program

6. log out and relogin

7. Monitor and read log. Open terminal
```journalctl -f -o cat /usr/bin/gnome-shell```
Now this terminal window will display realtime logs

8. Enable the extension.

Check and post here here any logs that starts with or related to Battery Health Charging

Also give me the output of 
```cat /sys/devices/virtual/dmi/id/sys_vendor```
I need this to fix the issue of pointing to wrong readme file which displayed link for apple-m-series insteaa of intel

kaveht, 2024-02-21:
Here is the output from journalctl
`Battery Health Extension:  Checking apple intel 
Battery Health Extension:  -- found apple path check 1 -- 
Battery Health Extension:  -- found apple path check 2 -- 
`
`$ cat /sys/devices/virtual/dmi/id/sys_vendor
Apple Inc.
`

maniacx, 2024-02-21:
So what exactly is the issue? 
is it still giving notification saying missing dependencies?

kaveht, 2024-02-22:
This time it didn't.  I believe it is working and I am able to change the thresholds.  Thanks for your help.

maniacx, 2024-02-22:
Strange... I havent changed anything. Only added logs. 

Anyway thanks for point out about the wrong dependencies link. Seems like both m-series and intel uses same Vendor DMI `Apple Inc`. So I will just link to device-compatiblity page instead of specific device page.

kaveht, 2024-02-22:
Since I am new to the linux world, I had asked a friend to reinstall pop os and install the extension;  I wonder if that is why it is working.  In any case, it is great having the option to change the thresholds.  Thanks again.

maniacx, 2024-02-22:
No problem. I will close this issue. If any problem feel free to re-open this issue.

closed 2024-02-22 by maniacx