maniacx/Battery-Health-Charging #79: What to do when I'm getting `Encountered an unexpected error. (Dell)`

alexanderadam, 2024-02-25:
I'm getting the error [`Encountered an unexpected error. (Dell)`](https://github.com/search?q=repo%3Amaniacx%2FBattery-Health-Charging%20Encountered%20an%20unexpected%20error.&type=code) on current Fedora on an XPS 13 9300.

This message popped up after a recent upgrade and now I'd love to know what I can do to tackle this.

maniacx, 2024-02-25:
This is how it works. You can test everything with command line.

## Check for executable installed
First it will check if this file exist. if it exist it, knows it is a dell laptop. But I think it has detected it as the Dell is mentioned in the error, but no harm checking.

`ls -l /sys/devices/platform/dell-laptop`

For dell you need to install either one of the package. libsmbios or Dell command center. Some laptop libsmbios works but Dell command center does not, but on others laptop Dell command center works but libsmbios does not.
So next step the extension will check if the executable exist.

extensionis check if program is available in enviroment. If it doesnt find the path it will check for the default installation location.

You can try test this by running the command
Program in enviroment. (This will ouptut charging mode information)
```
sudo smbios-battery-ctl --get-charging-cfg`
```
Default installation location
```
ls -l /usr/sbin/smbios-battery-ctl
```

It does the same for Dell command center.
Program in enviroment. (This will ouptut charging mode information)
```
sudo cctk  --PrimaryBattChargeCfg
```
Default installation location
```
ls -l /opt/dell/dcc/cctk
```

It willl use the package that is installed. If both are install, you will get an option to choose in extension preferences between Dell command center and libsmbios.

## Test if charging mode changes using command line.
1. Check which package is installed. 
2. Try changing threshold using command line. You can find command line details here 
https://maniacx.github.io/Battery-Health-Charging/device-compatibility/dell#testing-charging-threshold-using-command-line

3. I read a that some xps doesnt work will libsmbios, but works well with Dell command center.
If you are already using Dell Command Center, there could a possibility that Fedora has automatically installed libsmbios, as libsmbios is also needed as dependencies for firmware updates. and the extension may have choosen to use libsmbios (smbios-battery-ctl) as a way to change threshold, but since xps doesnt work with libsmbios it fails. 
So check if by any chance both package are install, libsmbios and dell command center. If both are installed you will find an option in extension settings to change the settings.

[image]

maniacx, 2024-02-26:
Yes everything seem fine from your end. No there are no logs. I added few logs, that will show me what command are execute and what are the response. Kindly install this [test.zip](https://github.com/maniacx/Battery-Health-Charging/files/14412459/test.zip)  and check the log `journalctl -f -o cat /usr/bin/gnome-shell`

How did you manage to get time of the command execution in terminal? never new that trick.

Also I saw pkexec commands took 3 second. Could use sudo and run those commands again. Just to check how long it takes, so that entering password time is not accounted.

alexanderadam, 2024-04-21:
> Kindly install this [test.zip](https://github.com/maniacx/Battery-Health-Charging/files/14412459/test.zip)

I'd love to but how do I do that?

> How did you manage to get time taken to execution the command in terminal? never new that trick.

Actually I'm not sure where this is coming from. I'm using [Starship](https://starship.rs/) for my shell and I think that [it is responsible for that output](https://github.com/starship/starship/blob/cce9d335095a10e7f53cb5c6e54d27d3ce8ac1e1/src/modules/cmd_duration.rs#L132).

> Also I saw pkexec commands took 3 second. Could use sudo and run those commands again. Just to check how long it takes, so that entering password time is not accounted.

The time was because of the password entering :wink:

maniacx, 2024-04-22:
Extract the test.zip. open the folder. Either right click on file install.sh and run as program. Or you can open terminal and type  ./install.sh

alexanderadam, 2024-05-30:
Hello @maniacx, I'm very sorry for the late response.

I think now it stopped working completely.
This happens after activating it:

```
Battery Health Charging: ---- argv = pkexec,/usr/local/bin/batteryhealthchargingctl-user,CHECKINSTALLATION,/home/user/.local/share/gnome-shell/extensions/Battery-Health-Charging@maniacx.github.com/resources,user ----
Battery Health Charging: ---- stdout = Battery Health Charging: checking rules !
Battery Health Charging: checking ctl !
Battery Health Charging: installation is up to date 
 ----
Battery Health Charging: ---- status = 0 ----
Battery Health Charging: ---- argv = pkexec,/usr/local/bin/batteryhealthchargingctl-user,DELL_RELATIVE_SMBIOS_BAT_READ ----
Battery Health Charging: ---- stdout = Charging mode: custom
Charging interval: (55, 60)
 ----
Battery Health Charging: ---- status = 0 ----
Battery Health Charging: ---- argv = pkexec,/usr/local/bin/batteryhealthchargingctl-user,DELL_RELATIVE_SMBIOS_BAT_WRITE,100,95 ----
Battery Health Charging: ---- stdout = Charging mode has been set to: custom
Custom charge interval has been set to (95, 100)
 ----
Battery Health Charging: ---- status = 0 ----
Battery Health Charging: ---- argv = pkexec,/usr/local/bin/batteryhealthchargingctl-user,DELL_RELATIVE_SMBIOS_BAT_READ ----
Battery Health Charging: ---- stdout = Charging mode: custom
Charging interval: (55, 60)
 ----
Battery Health Charging: ---- status = 0 ----
Unhandled promise rejection. To suppress this warning, add an error handler to your promise chain with .catch() or a try-catch block around your await expression. Stack trace of the failed promise:
_checkCompatibility@file:///home/user/.local/share/gnome-shell/extensions/Battery-Health-Charging@maniacx.github.com/lib/driver.js:108:30
IntializeDriver@file:///home/user/.local/share/gnome-shell/extensions/Battery-Health-Charging@maniacx.github.com/lib/driver.js:19:14
enable@file:///home/user/.local/share/gnome-shell/extensions/Battery-Health-Charging@maniacx.github.com/extension.js:30:32
_callExtensionEnable@resource:///org/gnome/shell/ui/extensionSystem.js:266:38
_onEnabledExtensionsChanged@resource:///org/gnome/shell/ui/extensionSystem.js:626:24
_loadExtensions/<@resource:///org/gnome/shell/ui/extensionSystem.js:731:18
createCheckedMethod/<@resource:///org/gnome/gjs/modules/core/overrides/Gio.js:730:46
enableExtension@resource:///org/gnome/shell/ui/extensionSystem.js:289:29
EnableExtension@resource:///org/gnome/shell/ui/shellDBus.js:453:38
_handleMethodCall@resource:///org/gnome/gjs/modules/core/overrides/Gio.js:329:38
_wrapJSObject/<@resource:///org/gnome/gjs/modules/core/overrides/Gio.js:408:34
@resource:///org/gnome/shell/ui/init.js:21:20
```

maniacx, 2024-06-01:
Install this and send me the logs again.
[test2.zip](https://github.com/user-attachments/files/15522989/test2.zip)

alexanderadam, 2024-06-02:
here you are:

```
Battery Health Charging: ---- argv = pkexec,/usr/local/bin/batteryhealthchargingctl-user,CHECKINSTALLATION,/home/user/.local/share/gnome-shell/extensions/Battery-Health-Charging@maniacx.github.com/resources,user ----
Battery Health Charging: ---- stdout = Battery Health Charging: checking rules !
Battery Health Charging: checking ctl !
Battery Health Charging: installation is up to date 
 ----
Battery Health Charging: ---- status = 0 ----
Battery Health Charging: ---- argv = pkexec,/usr/local/bin/batteryhealthchargingctl-user,DELL_RELATIVE_SMBIOS_BAT_READ ----
Battery Health Charging: ---- stdout = Charging mode: custom
Charging interval: (55, 60)
 ----
Battery Health Charging: ---- status = 0 ----
Battery Health Charging: ---- output = Charging mode: custom
Charging interval: (55, 60)

Battery Health Charging: ---- filteredOutput = Charging mode custom
Charging interval 55 60
Battery Health Charging: ---- splitOutput = Charging mode custom,Charging interval 55 60
Battery Health Charging: ---- firstLine = Charging,mode,custom
Battery Health Charging: ---- modeRead = custom
Battery Health Charging: ---- secondLine = Charging,interval,55,60
Battery Health Charging: ---- parseInt(secondLine[2]) = 55
Battery Health Charging: ---- parseInt(secondLine[3]) = 60
Battery Health Charging: ---- argv = pkexec,/usr/local/bin/batteryhealthchargingctl-user,DELL_RELATIVE_SMBIOS_BAT_WRITE,100,95 ----
Battery Health Charging: ---- stdout = Charging mode has been set to: custom
Custom charge interval has been set to (95, 100)
 ----
Battery Health Charging: ---- status = 0 ----
Battery Health Charging: ---- argv = pkexec,/usr/local/bin/batteryhealthchargingctl-user,DELL_RELATIVE_SMBIOS_BAT_READ ----
Battery Health Charging: ---- stdout = Charging mode: custom
Charging interval: (55, 60)
 ----
Battery Health Charging: ---- status = 0 ----
Battery Health Charging: ---- output = Charging mode: custom
Charging interval: (55, 60)

Battery Health Charging: ---- filteredOutput = Charging mode custom
Charging interval 55 60
Battery Health Charging: ---- splitOutput = Charging mode custom,Charging interval 55 60
Battery Health Charging: ---- firstLine = Charging,mode,custom
Battery Health Charging: ---- modeRead = custom
Battery Health Charging: ---- secondLine = Charging,interval,55,60
Battery Health Charging: ---- parseInt(secondLine[2]) = 55
Battery Health Charging: ---- parseInt(secondLine[3]) = 60
Unhandled promise rejection. To suppress this warning, add an error handler to your promise chain with .catch() or a try-catch block around your await expression. Stack trace of the failed promise:
_checkCompatibility@file:///home/user/.local/share/gnome-shell/extensions/Battery-Health-Charging@maniacx.github.com/lib/driver.js:108:30
IntializeDriver@file:///home/user/.local/share/gnome-shell/extensions/Battery-Health-Charging@maniacx.github.com/lib/driver.js:19:14
enable@file:///home/user/.local/share/gnome-shell/extensions/Battery-Health-Charging@maniacx.github.com/extension.js:30:32
_callExtensionEnable@resource:///org/gnome/shell/ui/extensionSystem.js:266:38
_onEnabledExtensionsChanged@resource:///org/gnome/shell/ui/extensionSystem.js:626:24
_loadExtensions/<@resource:///org/gnome/shell/ui/extensionSystem.js:731:18
createCheckedMethod/<@resource:///org/gnome/gjs/modules/core/overrides/Gio.js:730:46
enableExtension@resource:///org/gnome/shell/ui/extensionSystem.js:289:29
EnableExtension@resource:///org/gnome/shell/ui/shellDBus.js:453:38
_handleMethodCall@resource:///org/gnome/gjs/modules/core/overrides/Gio.js:329:38
_wrapJSObject/<@resource:///org/gnome/gjs/modules/core/overrides/Gio.js:408:34
@resource:///org/gnome/shell/ui/init.js:21:20
```

thank you so much for your help! :pleading_face:

maniacx, 2024-06-03:
There are two issues, that I see.

1. Is your charging threshold when applied it , the extension reads them to verify, when read it seems to be not updated.
2. After verification fails, (which happens in your case) it should give a notificaion.
```
Encountered an unexpected error. (Dell)
```
But the extension errors out. 

1. Did you see this notification when you enable extension? or do you still see this.

2. Can you disable the extension and copy  all these command, and paste in terminal. And send me the output.
Note: Do not copy one command at a time. Copy the whole list of command and paste in terminal

```
echo "---Read 1"
sudo smbios-battery-ctl --get-charging-cfg
echo "---Write 1 55 60"
sudo smbios-battery-ctl --set-charging-mode='custom'
sudo smbios-battery-ctl --set-custom-charge-interval=55 60
echo "---Read 2"
sudo smbios-battery-ctl --get-charging-cfg
echo "---Write 2 95 100"
sudo smbios-battery-ctl --set-charging-mode='custom'
sudo smbios-battery-ctl --set-custom-charge-interval=95 100
echo "---Read 3"
sudo smbios-battery-ctl --get-charging-cfg
echo "---Write 3 75 80"
sudo smbios-battery-ctl --set-charging-mode='custom'
sudo smbios-battery-ctl --set-custom-charge-interval=75 80
echo "---Read 4"
sudo smbios-battery-ctl --get-charging-cfg
echo "---Write 4 55 60"
sudo smbios-battery-ctl --set-charging-mode='custom'
sudo smbios-battery-ctl --set-custom-charge-interval=55 60
echo "---Read 5"
sudo smbios-battery-ctl --get-charging-cfg
```

maniacx, 2024-06-03:
Also I think you may have already move to GNOME46. And the test2 I gave you was still an older version (Sorry I was lazy)
Anyways test this extension. It may not fix the threshold issue, but it might fix the error in extension that prevents it from sending notification and start the UI.
[test3.zip](https://github.com/user-attachments/files/15527823/test3.zip)
Enable it, send me the logs. Let me know if you see the notification this time

alexanderadam, 2024-06-03:
>     Can you disable the extension and copy  all these command, and paste in terminal. And send me the output.
>        Note: Do not copy one command at a time. Copy the whole list of command and paste in terminal

![---Read 1 Charging mode: custom Charging interval: (55, 60) ---Write 1 55 60 Charging mode has been set to: custom Custom charge interval has been set to (55, 60) ---Read 2 Charging mode: custom Charging interval: (55, 60) ---Write 2 95 100 Charging mode has been set to: custom Custom charge interval has been set to (95, 100) ---Read 3 Charging mode: custom Charging interval: (55, 60) ---Write 3 75 80 Charging mode has been set to: custom Custom charge interval has been set to (75, 80) ---Read 4 Charging mode: custom Charging interval: (55, 60) ---Write 4 55 60 Charging mode has been set to: custom Custom charge interval has been set to (55, 60) ---Read 5 Charging mode: custom Charging interval: (55, 60)](https://github.com/maniacx/Battery-Health-Charging/assets/372620/93899e98-773a-49b2-999a-3cbe9953fd8c)

The other info comes later

maniacx, 2024-06-03:
For some reason your libsmbios doesn't update your threshold.
As you can see, changing threshold to doesn't work even using commandline it still stays on 55-60

When I was adding support for Dell, I read that most older laptop works well with libsmbios, buymt newer laptop have issues with it, and this users use Dell command configure.

You could try installing Dell command configure (dcc)

And try changing threshold using Dell command configure. If changing threshold works using dcc, you can change setting in extension preference > device to use Dell command configure.

https://macs18max.github.io//Battery-Health-Charging/device-compatibility/dell

maniacx, 2024-06-03:
Since I have fedora 40 too, i tried to install dcc even though my laptop is not dell.

First time for me installing rpm package locally. (I could be wrong or there might be a better way)
I downloaded the dcc package 4.11.0 for **RHEL 9.0 (64-bit)** from this website.
extract the package.

To install I used.

```
sudo rpm -i /home/$USER/Downloads/command-configure-4.11.0-3.el9.x86_64/srvadmin-hapi-9.5.0-4063.16816.el8.x86_64.rpm
```
and then
```
sudo rpm -i /home/$USER/Downloads/command-configure-4.11.0-3.el9.x86_64/command-configure-4.11.0-3.el9.x86_64.rpm
```
But for me runnning cctk doesnt do anything (May be because my laptop is not dell)

To Read threshold
```
sudo /opt/dell/dcc/cctk --PrimaryBattChargeCfg
```

To apply threshold
```
sudo /opt/dell/dcc/cctk --PrimaryBattChargeCfg=Custom:95-100
```
```
sudo /opt/dell/dcc/cctk --PrimaryBattChargeCfg=Custom:75-80
```
```
sudo /opt/dell/dcc/cctk --PrimaryBattChargeCfg=Custom:55-60
```

You can try to apply threshold and read threshold again and check if it reads back the applied threshold.

Incase. You want to uninstall dcc you can do it by
```
sudo rpm -ev srvadmin-hapi-9.5.0-4063.16816.el8.x86_64
```
```
sudo rpm -ev command-configure-4.11.0-3.el9.x86_64
```

alexanderadam, 2024-06-03:
Okay, so for a short moment I thought that I found the culprit:
[Dell also allows to set charging thresholds in the BIOS](https://www.dell.com/support/manuals/en-uk/dcpm2.1/userguide_dell-v1/battery-settings?guid=guid-0fbbbeff-4928-4def-89af-3d28d0a231ce&lang=en-us&lwp=rt)!
And apparently I did that and I forgot about it.
Now I set it back to "standard".

Setting the thresholds via terminal works now:

![sudo smbios-battery-ctl --get-charging-cfg echo "---Write 1 55 60" sudo smbios-battery-ctl --set-charging-mode='custom' sudo smbios-battery-ctl --set-custom-charge-interval=55 60 echo "---Read 2" sudo smbios-battery-ctl --get-charging-cfg echo "---Write 2 95 100" sudo smbios-battery-ctl --set-charging-mode='custom' sudo smbios-battery-ctl --set-custom-charge-interval=95 100 echo "---Read 3" sudo smbios-battery-ctl --get-charging-cfg echo "---Write 3 75 80" sudo smbios-battery-ctl --set-charging-mode='custom' sudo smbios-battery-ctl --set-custom-charge-interval=75 80 echo "---Read 4" sudo smbios-battery-ctl --get-charging-cfg echo "---Write 4 55 60" sudo smbios-battery-ctl --set-charging-mode='custom' sudo smbios-battery-ctl --set-custom-charge-interval=55 60 echo "---Read 5" sudo smbios-battery-ctl --get-charging-cfg ---Read 1 Charging mode: standard ---Write 1 55 60 Charging mode has been set to: custom Custom charge interval has been set to (55, 60) ---Read 2 Charging mode: standard ---Write 2 95 100 Charging mode has been set to: custom Custom charge interval has been set to (95, 100) ---Read 3 Charging mode: standard ---Write 3 75 80 Charging mode has been set to: custom Custom charge interval has been set to (75, 80) ---Read 4 Charging mode: standard ---Write 4 55 60 Charging mode has been set to: custom Custom charge interval has been set to (55, 60) ---Read 5 Charging mode: standard](https://github.com/maniacx/Battery-Health-Charging/assets/372620/774a21ae-9813-4f8e-8fb7-39a7f8c6cd92)

However, the Battery Health Charging extension still shows the very same notification with the error.

This is how its log looks like:

```
Battery Health Charging: ---- argv = pkexec,/usr/local/bin/batteryhealthchargingctl-user,CHECKINSTALLATION,/home/user/.local/share/gnome-shell/extensions/Battery-Health-Charging@maniacx.github.com/resources,user ----
Battery Health Charging: ---- stdout = Battery Health Charging: checking rules !
Battery Health Charging: checking ctl !
Battery Health Charging: installation is up to date 
 ----
Battery Health Charging: ---- status = 0 ----
Battery Health Charging: ---- argv = pkexec,/usr/local/bin/batteryhealthchargingctl-user,DELL_RELATIVE_SMBIOS_BAT_READ ----
Battery Health Charging: ---- stdout = Charging mode: standard
 ----
Battery Health Charging: ---- status = 0 ----
Battery Health Charging: ---- output = Charging mode: standard

Battery Health Charging: ---- filteredOutput = Charging mode standard
Battery Health Charging: ---- splitOutput = Charging mode standard
Battery Health Charging: ---- firstLine = Charging,mode,standard
Battery Health Charging: ---- modeRead = standard
Battery Health Charging: ---- argv = pkexec,/usr/local/bin/batteryhealthchargingctl-user,DELL_RELATIVE_SMBIOS_BAT_WRITE,100,95 ----
Battery Health Charging: ---- stdout = Charging mode has been set to: custom
Custom charge interval has been set to (95, 100)
 ----
Battery Health Charging: ---- status = 0 ----
Battery Health Charging: ---- argv = pkexec,/usr/local/bin/batteryhealthchargingctl-user,DELL_RELATIVE_SMBIOS_BAT_READ ----
Battery Health Charging: ---- stdout = Charging mode: standard
 ----
Battery Health Charging: ---- status = 0 ----
Battery Health Charging: ---- output = Charging mode: standard

Battery Health Charging: ---- filteredOutput = Charging mode standard
Battery Health Charging: ---- splitOutput = Charging mode standard
Battery Health Charging: ---- firstLine = Charging,mode,standard
Battery Health Charging: ---- modeRead = standard
```

The charging mode is now in `standard`:
[image]

A little code change for the extension could be that if the charging mode stays at `custom` (compare the charging mode of the first snippets to the one in this comment), and can't get over a certain threshold, then no error could be shown if the threshold is the same anyway (i.e. when it is set to 60 either way).
And if it can't get higher than a certain threshold a notification could be shown like "_The Battery Health Extension can't set higher thresholds. It's appears that there's a custom threshold set in BIOS._"

However, it doesn't explain why the extension still doesn't work although setting it via CLI works now.

maniacx, 2024-06-03:
So when we change the threshold in libsmbios (or dell command center) it makes changes in the bios, that is how the laptop limits the battery threshold even though the laptop is powered off and not runnning linux/windows os. (Unless dell does it differently, I do not know)

And as you can see, libsmbios is not working.
Once you change threshold to **custom**, it should switch from **standard** to **custom**

Look at this post to see, how it behave on for other dell laptops with libsmbios working.
https://github.com/maniacx/Battery-Health-Charging/issues/7#issuecomment-1458708793
You can go through that thread(issue) to understand how it works for Dell laptops. That was request from dell user to support dell laptops.

alexanderadam, 2024-06-03:
> as you can see, libsmbios is not working on your laptop. Once you apply **custom** mode, it should switch from **standard** to **custom**, but in your case it doesnt change to **custom** and still reads _standard_*

Oh, I see. That was my fault. :face_with_peeking_eye: 
But I finally found the culprit now: it's [libsmbios issue 103](https://github.com/dell/libsmbios/issues/103).
`libsmbios` doesn't work when an admin password is set. :roll_eyes: 

So this means that this is an :bug: `libsmbios` bug.
The Battery Health extension could only give users a hint (i.e. "_The Battery Health Extension can't set higher thresholds. Please check that you don't have an admin password set in BIOS and check https://github.com/dell/libsmbios/issues/103 for more information._")

I'll close this issue now since I believe that other nothing else than a helpful message can be done here.

Thank you for your help! :pray:

closed 2024-06-03 by alexanderadam

maniacx, 2024-06-03:
There is no-one fault here, we are all learning something new. It is difficult to know why libsmbios is not working (there could be several factors) and the sending a notification that admin passwords is set would be incorrect unless unless extension could confirm it.

Also dell command configure support changing threshold with admin password,  and so does this extension.
Another users requested it, and I was able to implement it and user confirmed it is working.

Check the discussion here.
https://github.com/maniacx/Battery-Health-Charging/issues/54

Feel free to open this if you have any other issue.