maniacx/Battery-Health-Charging #84: Thinkpad L13 Yoga battery threshold quirks

cameronaw13, 2024-04-04:
Thank you for your work and dedication on this application. I really appreciate it. I wanted to inform you on this weird issue I've been having on my Thinkpad L13 Yoga Gen 2a laptop running Fedora 39, Gnome 45, Battery-Health-Charging v60.

By default, setting the threshold gives an error saying "Charging threshold not updated. (Thinkpad BAT0)". Despite this, it was still keeping my laptop at 79% while charging demonstrating that it worked despite this error.

Eventually I got tired to seeing this notification every time I logged in so I investigated the issue. At first I thought I had the same issue as #55 despite it already being resolved but later I found that it is different.

I ran the cat commands [listed here](https://maniacx.github.io/Battery-Health-Charging/device-compatibility/thinkpad#testing-charging-threshold-using-command-line) to check my thresholds and interestingly both of them always equal the same value but are randomly assigned to either the chosen starting or ending threshold.

For example, by default, Balanced Mode will set both `charge_control_start_threshold` and `charge_control_end_threshold` to either 80 or 75. This is the same with the Full Capacity and Max Lifespan Modes.

After setting the Balanced and Max Lifespan Modes to have the same starting and ending thresholds, I stopped receiving the errors. I couldn't do this with the Full Capacity mode as it always requires the starting threshold to be 2% lower than the ending threshold.

Interestingly, this conflicts directly with the notion [listed here](https://maniacx.github.io/Battery-Health-Charging/device-compatibility/thinkpad#testing-charging-threshold-using-command-line) that the `charge_control_end_threshold` > `charge_control_start_threshold`. I'm wondering if my laptop is some sort of edge case where it doesn't apply this rule and sets a single threshold value.

maniacx, 2024-04-04:
If start and end threshold shows same value, I cannot much about it in the extension.

Here are some things you can try.

1. Resetting bios.
* If you have made any chances to default settings of bios, make a note of what you have changed, and  reset bios to defaults.

2. Also you can try hard reset bios to reset embedded control values.

From TLP documentation.

>Reset the EC: shut down the system, then press the emergency reset hole (button) on the bottom of the ThinkPad with a paper clip. On older models, the EC is reset by shutting down, removing all batteries, disconnecting the power supply and pressing the power button for 30 seconds.

Also check if you have any bios updates pending.
Sometimes new update won't show in Linux, but will show up if you running Windows .

3. Does the threshold for start and end are same all the time or only when you wake up from sleep or login?

cameronaw13, 2024-04-05:
The start and end thresholds are the same all the time no matter if I restart, log in, or wake from sleep except for when the default threshold is set (end threshold is 100 and the start threshold is 0)

I've tried steps 1-3 except for the windows part and nothing changes. The core functionality of setting thresholds still works but the interface the extension provides isn't able to account for this weird behavior.

cameronaw13, 2024-04-24:
I think my issue is described in the FAQ from [TLP's battery care documentation](https://linrunner.de/tlp/faq/battery.html#charge-thresholds-shown-by-tlp-stat-b-do-not-correspond-to-the-configured-ones) where one of the two thresholds are always displayed incorrectly. Seems that its a firmware issue and that, despite it doing this, it still applies the thresholds correctly.

The "Battery Health charging error" notification is a little annoying but the core issue cant be fixed on your end so I'll be closing this.

maniacx, 2024-05-23:
Hello @cameronaw13 

Did you find solution for your issue?

cameronaw13, 2024-05-23:
The behavior shown in the firmware's threshold values it to be expected for the thinkpad l-series laptops, but I do still face issues with the extension complaining about this behavior.
If the extension allowed me to set the start/stop thresholds to any value, provided that start_threshold <= stop_threshold, it would likely solve the issue on my end. But I'm uncertain if this should be implemented since it could cause issues with other threshold implementations.

maniacx, 2024-05-23:
Could you try copy pasting thr command in terminal and post the results

```
echo "---- Disabling extension ----"
gnome-extensions disable Battery-Health-Charging@maniacx.github.com

echo "---- Read 1 ----"
cat  /sys/class/power_supply/BAT0/charge_control_start_threshold
cat /sys/class/power_supply/BAT0/charge_control_end_threshold

echo "---- Write 1 Change Threshold 55/60 ----"
echo '55' | sudo tee /sys/class/power_supply/BAT0/charge_control_start_threshold
echo '60' | sudo tee /sys/class/power_supply/BAT0/charge_control_end_threshold

echo "---- Read 2----"
cat  /sys/class/power_supply/BAT0/charge_control_start_threshold
cat  /sys/class/power_supply/BAT0/charge_control_end_threshold

echo "---- Write 2 Change Threshold 75/80 ----"
echo '80' | sudo tee /sys/class/power_supply/BAT0/charge_control_end_threshold
echo '75' | sudo tee /sys/class/power_supply/BAT0/charge_control_start_threshold

echo "---- Read 3----"
cat  /sys/class/power_supply/BAT0/charge_control_start_threshold
cat  /sys/class/power_supply/BAT0/charge_control_end_threshold

echo "Write 3 Change Threshold 75/80 second attempt ----"
echo '80' | sudo tee /sys/class/power_supply/BAT0/charge_control_end_threshold
echo '75' | sudo tee /sys/class/power_supply/BAT0/charge_control_start_threshold

echo "---- Read 4----"
cat  /sys/class/power_supply/BAT0/charge_control_start_threshold
cat  /sys/class/power_supply/BAT0/charge_control_end_threshold

echo "Write 4 Change Threshold 55/60 ----"
echo '55' | sudo tee /sys/class/power_supply/BAT0/charge_control_start_threshold
echo '60' | sudo tee /sys/class/power_supply/BAT0/charge_control_end_threshold

echo "---- Read 5----"
cat  /sys/class/power_supply/BAT0/charge_control_start_threshold
cat  /sys/class/power_supply/BAT0/charge_control_end_threshold

echo "---- Write 5 Change Threshold 75/80 ----"
echo '80' | sudo tee /sys/class/power_supply/BAT0/charge_control_end_threshold
echo '75' | sudo tee /sys/class/power_supply/BAT0/charge_control_start_threshold

echo "---- Read 6----"
cat  /sys/class/power_supply/BAT0/charge_control_start_threshold
cat  /sys/class/power_supply/BAT0/charge_control_end_threshold

echo "Write 6 Change Threshold end only 80 second attempt ----"
echo '80' | sudo tee /sys/class/power_supply/BAT0/charge_control_end_threshold

echo "---- Read 7----"
cat  /sys/class/power_supply/BAT0/charge_control_start_threshold
cat  /sys/class/power_supply/BAT0/charge_control_end_threshold

gnome-extensions enable Battery-Health-Charging@maniacx.github.com
```

cameronaw13, 2024-05-23:
Sure, here's the output:
```
---- Disabling extension ----
---- Read 1 ----
60
60
---- Write 1 Change Threshold 55/60 ----
Place your finger on the fingerprint reader
55
60
---- Read 2----
60
60
---- Write 2 Change Threshold 75/80 ----
80
75
---- Read 3----
75
75
Write 3 Change Threshold 75/80 second attempt ----
80
75
---- Read 4----
75
75
Write 4 Change Threshold 55/60 ----
55
60
---- Read 5----
60
60
---- Write 5 Change Threshold 75/80 ----
80
75
---- Read 6----
75
75
Write 6 Change Threshold end only 80 second attempt ----
80
---- Read 7----
80
80
```

maniacx, 2024-05-23:
Ok. I was another issue here with ThinkPad. https://github.com/maniacx/Battery-Health-Charging/issues/91 And this case the issue happens only when raising threshold it needs to apply threshold again.

But your case seems completely different. In you case the threshold reads the same no matter what.

But how about actual charging and discharge,  does it follow the start_threshold and end_threshold?

 For example if you set start_threshold = 60 and end_threshold = 75.
 I assume reading threshold both start_thrrshold and end_threshold will read 60.
But Does actual baftery charging cycle from 75 - 60, or does the actual battery value stays on 60 and never charges to 75?

cameronaw13, 2024-05-24:
Decided to reset the thresholds to the same values they were when I performed a bios reset:
```
cameron@fedora-thinkpad:~$ echo '0' | sudo tee /sys/class/power_supply/BAT0/charge_control_start_threshold
0
cameron@fedora-thinkpad:~$ echo '100' | sudo tee /sys/class/power_supply/BAT0/charge_control_end_threshold
100
cameron@fedora-thinkpad:~$ cat /sys/class/power_supply/BAT0/charge_control_start_threshold
0
cameron@fedora-thinkpad:~$ cat /sys/class/power_supply/BAT0/charge_control_end_threshold
100
```
I then set it to your suggested thresholds:
```
cameron@fedora-thinkpad:~$ echo '60' | sudo tee /sys/class/power_supply/BAT0/charge_control_start_threshold
60
cameron@fedora-thinkpad:~$ echo '75' | sudo tee /sys/class/power_supply/BAT0/charge_control_end_threshold
75
cameron@fedora-thinkpad:~$ cat /sys/class/power_supply/BAT0/charge_control_start_threshold
75
cameron@fedora-thinkpad:~$ cat /sys/class/power_supply/BAT0/charge_control_end_threshold
75
```
In this case charging stops at 74% (threshold ± 1 I assume) and stays at that percentage without cycling.

I also tried setting the end threshold first before setting the start threshold:
```
cameron@fedora-thinkpad:~$ echo '0' | sudo tee /sys/class/power_supply/BAT0/charge_control_start_threshold
0
cameron@fedora-thinkpad:~$ echo '100' | sudo tee /sys/class/power_supply/BAT0/charge_control_end_threshold
100
cameron@fedora-thinkpad:~$ cat /sys/class/power_supply/BAT0/charge_control_start_threshold
0
cameron@fedora-thinkpad:~$ cat /sys/class/power_supply/BAT0/charge_control_end_threshold
100
cameron@fedora-thinkpad:~$ echo '75' | sudo tee /sys/class/power_supply/BAT0/charge_control_end_threshold
75
cameron@fedora-thinkpad:~$ echo '60' | sudo tee /sys/class/power_supply/BAT0/charge_control_end_threshold
60
cameron@fedora-thinkpad:~$ cat /sys/class/power_supply/BAT0/charge_control_end_threshold
60
cameron@fedora-thinkpad:~$ cat /sys/class/power_supply/BAT0/charge_control_start_threshold
60
```
In this case charging stops at 59% and it again stays at that percentage without cycling.

maniacx, 2024-05-24:
I think it is charging and discharging 75 - 60

Usually charging is quite fast. But discharging is quite slow may take more than few hours to reach 60. I suggest keep threshold for testing 75 - 60. One it reaches 75. Probably in few hours it will drop 5- 10%. 
So longer hours testing may be needed. Use it for a day or two and let me know your observation.  No hurry

maniacx, 2024-05-26:
I have modified the extension and uploaded the new branch
https://github.com/maniacx/Battery-Health-Charging/tree/thinkpad_quirky_firmware

In extension prefs > Thinkpad > Enable "Skip Threshold Verification" and that should skip reading and verifying the threshold. In this case the extension will just apply the threshold, but wont verify and notify if threshold is applied.

Please install by running ./install.sh and test it.

I have Asus laptop, but can usually simulate the different devices such as thinkpads, huawei by just changing the threshold paths to fake once.
But this kind of quirky behaviour is difficult to simulate and I cannot test it.
So please test it and if there are any issue/bugs, please check the logs.

To check logs, disable extension

Open terminal and run
```
journalctl -f -o cat /usr/bin/gnome-shell
```
keep terminal window open and enable extension, check the logs in the terminal window.

Similiarly if you cannot open prefs/settings for this extension.
Open another terminal window and check gjs logs

```
journalctl -f -o cat /usr/bin/gjs
```
and post them here

cameronaw13, 2024-05-28:
I'm not sure of this is intentional or not due to the wip state of this branch, but one glaring issue I'm seeing on my end is that the UI within the quick settings isn't updating at all. Visually its stuck on the mode it starts with and changing modes doesn't give any visual response other than within the extension's settings and in gnome-settings. The charge thresholds are also still set to NaN / NaN %.

But, more importantly, it seems that the actual thresholds are correctly applied to my laptop without having an error appear on the screen. I'll test it more throughout the week.

cameronaw13, 2024-06-05:
I've tested it out for a week now and everything seems to work smoothly. The thresholds are properly set and the UI within the quick settings menu also updates properly.

closed 2024-07-10 by maniacx