maniacx/Battery-Health-Charging #87: Feature request: "Override" battery status in GNOME?

Waffleophagus, 2024-04-28:
Would it be at all possible to attempt to override the default battery percentage to have the OS think in terms of 0-100% regardless of if I set it to max charge at 80%? It just makes life easier to think in terms of "full" at 100% instead of at 80% and so the ability to adjust it to think that 80% is "full" would be awesome. If its not possible, totally understandable but figure I'd ask!

[image]

Cheers,
Waffleophagus

(P.S. this is awesome! thanks for the work so far!)

maniacx, 2024-04-29:
If you mean 
Charging threshold = 80%
Actual battery level = 80%
Display battery level as = 100%
(Similar to what is done in tuxedo laptops i think)

Then Yes. It is possible but will be a little complicated. It can be modified here
https://github.com/maniacx/Battery-Health-Charging/blob/c318f3409be1977c191c0c476f3193ba639e09dd/lib/powerIcon.js#L106 if the "Change charging behavior is changed" was applied.

The problem would be how to handle the situation when 
charging threshold =80%, 
the actual battery =90%.
Then what should the level be displayed? 110%?

Also another problem would be this extension, supports multiple device which support different level of threshold. Such as 60%, 80%, 85%. That would be very confusing, when they set to threshold to 80% then this is the new 100% and then if they switch to 60% then this will be the new 100.

Waffleophagus, 2024-04-29:
Yea, Your scenario fits exactly what I am requesting! I totally get your concerns though. Personally, the "if threshold is 80, and battery is at 90%, show 110" is honestly fine by me for this. I do totally understand the implications of all the devices. I have worked in mobile gaming and man, nothing is worse than some obscure device acting up. If it helps, I am on the Framework 16. My thought process on your second problem was that with your icon, if elected, it would be mostly straight forward: 
[image]

If it showed 100% in the far right, but 60 on your indicator, I would know that my max right now is 60. If I change it, it would then tell me the new max and I'd be at 100% of that charge limit. I appreciate you humoring my ideas here, and understand if its not something you want to dive into.

Cheers,
Waffleophagus

maniacx, 2024-04-30:
I will have a look at this when I have free time, if I succeed will give you a test version. But could take me a few days,  as I am very busy these day.. keep an eye on this post.

maniacx, 2024-05-07:
I didnt have much time to spend on this. But try this, 
https://github.com/maniacx/Battery-Health-Charging/tree/tray_compensator-GNOME45

It is not perfect, but should work.
Not perfect because there are several laptops that donot have threshold values, such as Lenovo Ideapad when threshold is enabled it could be 60 or 80% depending on the firmware of the laptop, and also dell have adaptive and express mode. and I have not taken into account for those device. For rest laptops it should work. let me know

Waffleophagus, 2024-05-08:
I've been using it all night so far on my framework and it works *wonderfully* thank you!