maniacx/Battery-Health-Charging #88: Compatibility with Fujitsu Siemens S936

Yattatux, 2024-04-30:
Any chance of adding the s936 to the list of compatible laptops?

Many thanks for your work.

maniacx, 2024-05-01:
I can add fujistsu to the extension with some information/help from you. I just checked the linux kernel and seems like a dev added support for fujistsu 3 months ago 

https://github.com/torvalds/linux/blob/master/drivers/platform/x86/fujitsu-laptop.c

https://github.com/torvalds/linux/commit/14c8a1451810cf616c959574bee4fad246f3a0cb

Can you control the charging threshold using command line?

You should be able to set charging threshold using command

```
echo '60' | sudo tee /sys/class/power_supply/BAT0/charge_control_end_threshold
```

Note that BAT0 may be BAT1 in some laptops

Also provide me a list of charging devices 

```
ls -l /sys/class/power_supply/
```

Also need to know location of fujistsu module, so that the extension can set properties for fujistsu. 

Post the result of this.

```
ls -l /sys/devices/platform/framework_laptop
```

Yattatux, 2024-05-01:
Hello, thanks for reply.

my system:
```
$ uname -a
Linux fedora 6.8.7-200.fc39.x86_64 #1 SMP PREEMPT_DYNAMIC Wed Apr 17 19:35:11 UTC 2024 x86_64 GNU/Linux
```

Output in my laptop:
```
$ echo '60' | sudo tee /sys/class/power_supply/BAT0/charge_control_end_threshold
[sudo] password: 
tee: /sys/class/power_supply/BAT0/charge_control_end_threshold: File or directory does not exist
```

```
$ ls -l /sys/class/power_supply/
total 0
lrwxrwxrwx. 1 root root 0  1 mag  2024 AC -> ../../devices/platform/ACPI0003:00/power_supply/AC
lrwxrwxrwx. 1 root root 0  1 mag  2024 CMB1 -> ../../devices/LNXSYSTM:00/LNXSYBUS:00/PNP0C0A:00/power_supply/CMB1
```

```
$ ls -l /sys/devices/platform/framework_laptop
ls: impossibile accedere a '/sys/devices/platform/framework_laptop': File or directory does not exist
```

```
$ ls -l /sys/class/power_supply/CMB1/
total 0
-rw-r--r--. 1 root root 4096  1 mag 21.29 alarm
-r--r--r--. 1 root root 4096  1 mag 21.26 capacity
-r--r--r--. 1 root root 4096  1 mag 21.29 capacity_level
-r--r--r--. 1 root root 4096  1 mag 21.26 charge_full
-r--r--r--. 1 root root 4096  1 mag 21.26 charge_full_design
-r--r--r--. 1 root root 4096  1 mag 21.26 charge_now
-r--r--r--. 1 root root 4096  1 mag 21.26 current_now
-r--r--r--. 1 root root 4096  1 mag 21.26 cycle_count
lrwxrwxrwx. 1 root root    0  1 mag 21.26 device -> ../../../PNP0C0A:00
drwxr-xr-x. 3 root root    0  1 mag  2024 hwmon1
-r--r--r--. 1 root root 4096  1 mag 21.26 manufacturer
-r--r--r--. 1 root root 4096  1 mag 21.26 model_name
drwxr-xr-x. 2 root root    0  1 mag 21.29 power
-r--r--r--. 1 root root 4096  1 mag 21.26 present
-r--r--r--. 1 root root 4096  1 mag 21.26 serial_number
-r--r--r--. 1 root root 4096  1 mag 21.26 status
lrwxrwxrwx. 1 root root    0  1 mag  2024 subsystem -> ../../../../../../class/power_supply
-r--r--r--. 1 root root 4096  1 mag 21.26 technology
-r--r--r--. 1 root root 4096  1 mag 21.26 type
-rw-r--r--. 1 root root 4096  1 mag  2024 uevent
-r--r--r--. 1 root root 4096  1 mag 21.26 voltage_min_design
-r--r--r--. 1 root root 4096  1 mag 21.26 voltage_now
```

maniacx, 2024-05-02:
My mistake the command was frameworks laptops
Try this.
```
 ls -l /sys/devices/platform/fujistsu-laptop
```
If still not found, 
Do 
```
ls -l /sys/devices/platform/
``` 
And see if there is any module registered under fujitsu such as `fujistu-laptop` or `fujitsu_laptop`

Regarding charging threshold seems like the patch is not yet merged or accepted in the main line kernel. That why you are not seeing `charge_control_end_threshold` under your battery.  If you know how to
1. apply the patch to the kernel and compile and install it or compile it as a module
2. How to revert back to the original kernel incase the kernel doesn't boot.

 you can apply the kernel patch and test it.

I have never done it, or I cannot guide you. 

Or else wait for it to be merged in the main line

Yattatux, 2024-05-02:
Thanks for reply ;)

Output:

```
$  ls -l /sys/devices/platform/fujitsu-laptop/
totale 0
-r--r--r--. 1 root root 4096  2 mag 21.11 dock
lrwxrwxrwx. 1 root root    0  2 mag 20.31 driver -> ../../../bus/platform/drivers/fujitsu-laptop
-rw-r--r--. 1 root root 4096  2 mag 21.11 driver_override
-r--r--r--. 1 root root 4096  2 mag 21.11 lid
-r--r--r--. 1 root root 4096  2 mag 21.11 modalias
drwxr-xr-x. 2 root root    0  2 mag 21.11 power
-r--r--r--. 1 root root 4096  2 mag 21.11 radios
lrwxrwxrwx. 1 root root    0  2 mag 20.32 subsystem -> ../../../bus/platform
-rw-r--r--. 1 root root 4096  2 mag 20.32 uevent
```

I too, have never patched the kernel.
In the next days I might try to do that, maybe I can succeed.

Thanks again ^_

maniacx, 2024-05-23:
Seems like the patch for fujistsu is merged in kernel 6.9.1

Yattatux, 2024-05-23:
> Seems like the patch for fujistsu is merged in kernel 6.9.1

Hello @maniacx and thanks for info.
I'm on Fedora 41 now, and kernel 6.9.0-64 works ^__^ 
```
$ uname -a
Linux fedora 6.9.0-64.fc41.x86_64 #1 SMP PREEMPT_DYNAMIC Mon May 13 11:58:46 UTC 2024 x86_64 GNU/Linux
```
```
$ echo '60' | sudo tee /sys/class/power_supply/CMB1/charge_control_end_threshold
[sudo] password: 
60
```
```
$ ls -l /sys/devices/platform/fujitsu-laptop
total 0
-r--r--r--. 1 root root 4096 May 23 16:25 dock
lrwxrwxrwx. 1 root root    0 May 23 16:13 driver -> ../../../bus/platform/drivers/fujitsu-laptop
-rw-r--r--. 1 root root 4096 May 23 16:25 driver_override
-r--r--r--. 1 root root 4096 May 23 16:25 lid
-r--r--r--. 1 root root 4096 May 23 16:25 modalias
drwxr-xr-x. 2 root root    0 May 23 16:25 power
-r--r--r--. 1 root root 4096 May 23 16:25 radios
lrwxrwxrwx. 1 root root    0 May 23 16:14 subsystem -> ../../../bus/platform
-rw-r--r--. 1 root root 4096 May 23 16:14 uevent
```

maniacx, 2024-05-23:
How about reading threshold? Does that work too? change threshold with different values and  try reading using cat command

```
echo "---- write ----"
echo '80' | sudo tee /sys/class/power_supply/CMB1/charge_control_end_threshold

echo "---- read ----"
cat /sys/class/power_supply/CMB1/charge_control_end_threshold
```

I have uploaded a test branch. https://github.com/maniacx/Battery-Health-Charging/tree/test_fujistsu
If reading works, this branch should work. Install extension by running ./install.sh, and restart gnome-shell.
Enable extension and test if it works.

If there is no compatible extension app or extension manager app for Fedora 41, you can enable the extension using command-line

```
gnome-extensions enable Battery-Health-Charging@maniacx.github.com
```

Note:  I do not know which gnome-shell version is Fedora 41 is using. it could Gnome 46 which is compatible or may be the latest main branch, which may or may not be compatible, if there are major changes in the latest gnome-shell.

Let me know how it goes.

Yattatux, 2024-05-23:
I download new branch, installed it by ./install.sh
Log out/ Log in
and then:
`gnome-extensions enable Battery-Health-Charging@maniacx.github.com`

There is a error message when I enable it:
```
Battery Healt Charging Error
Please install polkit from extension settings under installation.
```
[image]

maniacx, 2024-05-23:
Did you install polkit rules from the extension preferences.
Click on Settings in that notification, and it will take you to extension prefs.
In extension prefs > General > Installation > Click install.

another way to open extension preferences  window is using extension app or extension manager app. or you can use command line 
```
gnome-extensions prefs Battery-Health-Charging@maniacx.github.com

```

[image]

closed 2024-05-26 by maniacx