maniacx/Battery-Health-Charging #89: BAT1 not visible in quick settings or extension preferences (Pop_OS!, Thinkpad T480)

martin-scheele, 2024-05-07:
Using Pop_OS! (Gnome version 42.5) on a Thinkpad T480. 

Extension works fine with the internal battery, but I just got a used genuine power bridge battery and it doesn't show up anywhere in the extension GUI. I tried resetting gsettings and uninstalling/reinstalling to no avail. The thresholds are able to be set manually via command line so I assume its just an issue with the extension.

Let me know if I can provide any other information, and thanks for the extension!

maniacx, 2024-05-07:
Try this commands these 3 command. it will disable the extension before resetting gsettings and enable the extension.

```
gnome-extensions disable Battery-Health-Charging@maniacx.github.com

gsettings --schemadir /home/$USER/.local/share/gnome-shell/extensions/Battery-Health-Charging@maniacx.github.com/schemas reset-recursively org.gnome.shell.extensions.Battery-Health-Charging

gnome-extensions enable Battery-Health-Charging@maniacx.github.com
```

If it still doesnt work.
Can you send a screen shot of your quick settings panel 
[image]

Also post the command you used to change charging threshold for  power bridge battery and internal battery.

martin-scheele, 2024-05-07:
(I misclicked close issue, sorry about that!)

Unfortunately disabling and resetting gsettings did not resolve the issue.

Here's a screenshot of my quick settings panel:

[image]

The commands I used to manually set the charging thresholds are the ones that are specified in the Device Compatibility - Thinkpad (Dual) page:

`echo '55' | pkexec tee /sys/class/power_supply/BAT1/charge_control_start_threshold`
`echo '60' | pkexec tee /sys/class/power_supply/BAT1/charge_control_end_threshold`

I checked that they went into effect with the `cat` commands and they persisted after reboot.

martin-scheele, 2024-05-08:
Running into another issue that is probably unrelated to the extension, but I figured I'd relay the information just in case.

The manual thresholds that I set for the secondary battery do not seem to be operating off of the current battery level of the secondary battery - I can't quite figure out where it is getting its values from, but it _seems_ like its related to the total current capacity of both batteries.

For example, my internal battery is close to 100% (I just finished trying to manually calibrate it), and the secondary battery is around 50%. The total current capacity read as 67%. With the start and end thresholds for both batteries set to 55 and 60 respectively, the secondary battery stopped charging before it hit 60%, somewhere around 57%. 

Strangely, setting the BAT1 end threshold to 100 did not resume charging. I had to lower the BAT1 start threshold to some lower value (I only tested 0) for charging to resume. With the end threshold set to 60 and start still at 0, charging continued until 59%. 

I haven't tested enough (and don't have enough of an understanding of how the OS handles charging in the first place) to understand what is happening, so I'm not sure if its just a quirk with all dual battery laptops, or an issue with the T480 firmware, or is somehow related to the extension.

martin-scheele, 2024-05-08:
Disregard the last comment lol, I just misunderstood how the battery thresholds work - sorry if I wasted your time making you read it !

maniacx, 2024-05-09:
No problem.

I VM for ubuntu 22,04 was working so wasnt able to test the extension. Just got it running today and it seem to work.

Can you post the output of this command?

```
echo "---- BAT0" && ls  /sys/class/power_supply/BAT0 && echo "---- BAT1" && ls /sys/class/power_supply/BAT1
```

Also the output of this command

```
gsettings --schemadir /home/$USER/.local/share/gnome-shell/extensions/Battery-Health-Charging@maniacx.github.com/schemas set org.gnome.shell.extensions.Battery-Health-Charging device-type
```

martin-scheele, 2024-05-09:
First command output: 

```
---- BAT0
alarm		  charge_control_end_threshold	  cycle_count	      energy_now    power	   status      uevent
capacity	  charge_control_start_threshold  device	      hwmon2	    power_now	   subsystem   voltage_min_design
capacity_level	  charge_start_threshold	  energy_full	      manufacturer  present	   technology  voltage_now
charge_behaviour  charge_stop_threshold		  energy_full_design  model_name    serial_number  type
---- BAT1
alarm		  charge_control_end_threshold	  cycle_count	      energy_now    power	   status      uevent
capacity	  charge_control_start_threshold  device	      hwmon3	    power_now	   subsystem   voltage_min_design
capacity_level	  charge_start_threshold	  energy_full	      manufacturer  present	   technology  voltage_now
charge_behaviour  charge_stop_threshold		  energy_full_design  model_name    serial_number  type
```

Second command output: 

```
Usage:
  gsettings [--schemadir SCHEMADIR] set SCHEMA[:PATH] KEY VALUE

Set the value of KEY to VALUE

Arguments:
  SCHEMADIR A directory to search for additional schemas
  SCHEMA    The name of the schema
  PATH      The path, for relocatable schemas
  KEY       The key within the schema
  VALUE     The value to set
```

I assume something's wrong considering the second command doesn't work.

maniacx, 2024-05-10:
The first command your output seems good as expected. This is strange, that extension doesn't detect both path. I need to check and dig deeper about whats going on. 

Second command is my mistake again. It should be get and not set. Copy paste error.

First retry this command. And post the output here.

```
gsettings --schemadir /home/$USER/.local/share/gnome-shell/extensions/Battery-Health-Charging@maniacx.github.com/schemas get org.gnome.shell.extensions.Battery-Health-Charging device-type
```
You might get an result as 20 or 21. Post whatever result you get.

Second let try to disable extension, change, device-type and enable extension and let see what happens.
Copy paste this command.

```
gnome-extensions disable Battery-Health-Charging@maniacx.github.com

gsettings --schemadir /home/$USER/.local/share/gnome-shell/extensions/Battery-Health-Charging@maniacx.github.com/schemas set org.gnome.shell.extensions.Battery-Health-Charging device-type 19

gnome-extensions enable Battery-Health-Charging@maniacx.github.com

```
let me know if it changes anything or of you are able to see settings for both batteries

martin-scheele, 2024-05-10:
The device-type outputs as '11' which I see corresponds to system76 machines - I couldn't tell how exactly the device type gets initialized, but could there be some incorrect logic in the initialization when using pop_os? 

Setting the device-type manually worked correctly - thanks for walking me through this! 

Is there anything else that I can do to help troubleshoot this?

maniacx, 2024-05-10:
Oh OK.

So this extension supports different laptop which have different way of setting threshold.

And to distinguish the laptop, I use kernel modules to detect if laptop is ThinkPad, Asus, or samsung etc etc.
 And for system76 laptops the extension checks for this following file path.

/sys/module/system76_acpi

And if it finds it, it will load system76 properties. 

But since pop os is developed by system76 i think the same module is also loaded on non system76 laptops.

Could you check if your laptop has this path?

```
ls /sys/module/system76_acpi
```

martin-scheele, 2024-05-10:
I just checked and it does have the system76_acpi module, but it also has the thinkpad_acpi module. I assume you could just check for the existence of both paths and default to the thinkpad properties if both are found.

maniacx, 2024-05-10:
One last thing. So that I can make changes to the extension. The output of 

```
cat /sys/devices/virtual/dmi/id/sys_vendor
```

martin-scheele, 2024-05-11:
Sure thing, the output is `LENOVO`.

maniacx, 2024-05-11:
thank you. I'll add a dmi match for system76 laptops so that it doesn't effect ThinkPad or any other device in the next version.

martin-scheele, 2024-05-11:
Awesome, glad I could help bring this to your attention - thanks for your time!

maniacx, 2024-05-15:
@martin-scheele 
I have updated the branch
Could you test this for me

https://github.com/maniacx/Battery-Health-Charging/tree/GNOME42-44

run install.sh

restart gnome-shell

disable extension, reset gsettings and enable it. and see if it auto detects thinkpad correctly

```
gnome-extensions disable Battery-Health-Charging@maniacx.github.com

gsettings --schemadir /home/$USER/.local/share/gnome-shell/extensions/Battery-Health-Charging@maniacx.github.com/schemas reset-recursively org.gnome.shell.extensions.Battery-Health-Charging

gnome-extensions enable Battery-Health-Charging@maniacx.github.com
```

martin-scheele, 2024-05-17:
Using the updated branch works as expected. The device-type reads as `19` and both batteries are detected.

closed 2024-05-26 by maniacx