maniacx/Battery-Health-Charging #90: Battery status icon always as charging.

danroxha, 2024-05-07:
[image]

The extension does not change the icon to symbolize the end of battery power given the conservative limit enabled.

Extension disabled
[image]

closed 2024-05-07 by danroxha

MaximalCats, 2024-09-10:
Have you found a fix? It's purely cosmetical (battery is in fact, not charging), but still annoying.

MaximalCats, 2024-09-11:
> @MaximalCats
> 
> In extension prefs, **Change Battery indication icon behavior** enable it. Let me, me know if it doesnot work for you.

Did not work, but now I'm thinking the entire extension doesn't work (battery level stays at 99%)

maniacx, 2024-09-11:
@MaximalCats
Lets discuss this here
https://github.com/maniacx/Battery-Health-Charging/issues/105