maniacx/Battery-Health-Charging #91: threshold not updated error raised even though the change was successful

SCV-ready, 2024-05-20:
## bug:
## expected behaviour: 
changing the profiles using the widget should change the thresholds and update both icon and panel successfully.
## actual behaviour:
icon and BAT0 state successfully changed, but the panel raises an error and doesnt display accurate values (NaN/NaN%).
moreover, the panel doesnt update to the current selection.
## steps to reproduce: 
1. install directly from gnome extensions
2. add permissions to polkit
3. switch states using the panel widget.

## relevant system details:
```
System: 
 Kernel: 6.8.9-200.fc39.x86_64 arch: x86_64
    bits: 64
  Desktop: GNOME v: 45.6 Distro: Fedora Linux 39 (Workstation Edition)
Machine:
  Type: Laptop System: LENOVO product: 20H50070IV v: ThinkPad E570
```

maniacx, 2024-05-22:
Sorry for the late reply.

Could you post the output of the following in console/terminal

```
echo "---- Disabling extension ----"
gnome-extensions disable Battery-Health-Charging@maniacx.github.com

echo "---- Get Device Type ----"
gsettings --schemadir /home/$USER/.local/share/gnome-shell/extensions/Battery-Health-Charging@maniacx.github.com/schemas get org.gnome.shell.extensions.Battery-Health-Charging device-type

echo "---- Read 1 ----"
cat  /sys/class/power_supply/BAT0/charge_control_start_threshold
cat /sys/class/power_supply/BAT0/charge_control_end_threshold

echo "---- Change Threshold 55/60 ----"
echo '55' | sudo tee /sys/class/power_supply/BAT0/charge_control_start_threshold
echo '60' | sudo tee /sys/class/power_supply/BAT0/charge_control_end_threshold

echo "---- Read 2----"
cat  /sys/class/power_supply/BAT0/charge_control_start_threshold
cat  /sys/class/power_supply/BAT0/charge_control_end_threshold

echo "---- Change Threshold 75/80 ----"
echo '80' | sudo tee /sys/class/power_supply/BAT0/charge_control_end_threshold
echo '75' | sudo tee /sys/class/power_supply/BAT0/charge_control_start_threshold

echo "---- Read 3----"
cat  /sys/class/power_supply/BAT0/charge_control_start_threshold
cat  /sys/class/power_supply/BAT0/charge_control_end_threshold

gnome-extensions enable Battery-Health-Charging@maniacx.github.com

```
Copy the whole list of command and paste in terminal
Brief description of the command. It will disable the extension, get the device type form gsettings, read threshold, set threshold to 55/60, read again and change threshold to 75/80 and read again. 
Post the output in your reply.

SCV-ready, 2024-05-22:
```
---- Disabling extension ----
---- Get Device Type ----
20
---- Read 1 ----
80
75
---- Change Threshold 55/60 ----
55
60
---- Read 2----
60
55
---- Change Threshold 75/80 ----
80
75
---- Read 3----
75
75

```

maniacx, 2024-05-22:
Strange. I never owned a thinkpad device, but from what I researched that in most thinkpad firmwares require a condition to be met which is start_threshold < end_threshold when we apply threshold. But in your a case it may be opposite,

In the commands I posted earlier. 

May be in your case start_threshold needs to be applied before end_threshold.

Lets try somethings again.

```
echo "---- Disabling extension ----"
gnome-extensions disable Battery-Health-Charging@maniacx.github.com

echo "---- Read 1 ----"
cat  /sys/class/power_supply/BAT0/charge_control_start_threshold
cat /sys/class/power_supply/BAT0/charge_control_end_threshold

echo "---- Write 1 Change Threshold 55/60 ----"
echo '55' | sudo tee /sys/class/power_supply/BAT0/charge_control_start_threshold
echo '60' | sudo tee /sys/class/power_supply/BAT0/charge_control_end_threshold

echo "---- Read 2----"
cat  /sys/class/power_supply/BAT0/charge_control_start_threshold
cat  /sys/class/power_supply/BAT0/charge_control_end_threshold

echo "---- Write 2 Change Threshold 75/80 ----"
echo '80' | sudo tee /sys/class/power_supply/BAT0/charge_control_end_threshold
echo '75' | sudo tee /sys/class/power_supply/BAT0/charge_control_start_threshold

echo "---- Read 3----"
cat  /sys/class/power_supply/BAT0/charge_control_start_threshold
cat  /sys/class/power_supply/BAT0/charge_control_end_threshold

echo "Write 3 Change Threshold 55/60 ----"
echo '55' | sudo tee /sys/class/power_supply/BAT0/charge_control_start_threshold
echo '60' | sudo tee /sys/class/power_supply/BAT0/charge_control_end_threshold

echo "---- Read 4----"
cat  /sys/class/power_supply/BAT0/charge_control_start_threshold
cat  /sys/class/power_supply/BAT0/charge_control_end_threshold

echo "Write 4 Change Threshold 55/60 ----"
echo '75' | sudo tee /sys/class/power_supply/BAT0/charge_control_start_threshold
echo '80' | sudo tee /sys/class/power_supply/BAT0/charge_control_end_threshold

echo "---- Read 5----"
cat  /sys/class/power_supply/BAT0/charge_control_start_threshold
cat  /sys/class/power_supply/BAT0/charge_control_end_threshold

gnome-extensions enable Battery-Health-Charging@maniacx.github.com

```

SCV-ready, 2024-05-22:
```
---- Disabling extension ----
---- Read 1 ----
75
75
---- Write 1 Change Threshold 55/60 ----
55
60
---- Read 2----
60
55
---- Write 2 Change Threshold 75/80 ----
80
75
---- Read 3----
75
75
Write 3 Change Threshold 55/60 ----
55
60
---- Read 4----
60
55
Write 4 Change Threshold 55/60 ----
75
tee: /sys/class/power_supply/BAT0/charge_control_start_threshold: Invalid argument
80
---- Read 5----
80
55

```

maniacx, 2024-05-23:
Hmm. Seems like it rejects threshold if start_threshold > end_threshold.

Not sure why the end_threshold wont update when raising threshold with start_threshold > end_threshold conditions met. Guess some issues with firmware.

Since you have mentioned that applying end_threshold again changes value to 80, lets try these command again.

```
echo "---- Disabling extension ----"
gnome-extensions disable Battery-Health-Charging@maniacx.github.com

echo "---- Read 1 ----"
cat  /sys/class/power_supply/BAT0/charge_control_start_threshold
cat /sys/class/power_supply/BAT0/charge_control_end_threshold

echo "---- Write 1 Change Threshold 55/60 ----"
echo '55' | sudo tee /sys/class/power_supply/BAT0/charge_control_start_threshold
echo '60' | sudo tee /sys/class/power_supply/BAT0/charge_control_end_threshold

echo "---- Read 2----"
cat  /sys/class/power_supply/BAT0/charge_control_start_threshold
cat  /sys/class/power_supply/BAT0/charge_control_end_threshold

echo "---- Write 2 Change Threshold 75/80 ----"
echo '80' | sudo tee /sys/class/power_supply/BAT0/charge_control_end_threshold
echo '75' | sudo tee /sys/class/power_supply/BAT0/charge_control_start_threshold

echo "---- Read 3----"
cat  /sys/class/power_supply/BAT0/charge_control_start_threshold
cat  /sys/class/power_supply/BAT0/charge_control_end_threshold

echo "Write 3 Change Threshold 75/80 second attempt ----"
echo '80' | sudo tee /sys/class/power_supply/BAT0/charge_control_end_threshold
echo '75' | sudo tee /sys/class/power_supply/BAT0/charge_control_start_threshold

echo "---- Read 4----"
cat  /sys/class/power_supply/BAT0/charge_control_start_threshold
cat  /sys/class/power_supply/BAT0/charge_control_end_threshold

echo "Write 4 Change Threshold 55/60 ----"
echo '55' | sudo tee /sys/class/power_supply/BAT0/charge_control_start_threshold
echo '60' | sudo tee /sys/class/power_supply/BAT0/charge_control_end_threshold

echo "---- Read 5----"
cat  /sys/class/power_supply/BAT0/charge_control_start_threshold
cat  /sys/class/power_supply/BAT0/charge_control_end_threshold

echo "---- Write 5 Change Threshold 75/80 ----"
echo '80' | sudo tee /sys/class/power_supply/BAT0/charge_control_end_threshold
echo '75' | sudo tee /sys/class/power_supply/BAT0/charge_control_start_threshold

echo "---- Read 6----"
cat  /sys/class/power_supply/BAT0/charge_control_start_threshold
cat  /sys/class/power_supply/BAT0/charge_control_end_threshold

echo "Write 6 Change Threshold end only 80 second attempt ----"
echo '80' | sudo tee /sys/class/power_supply/BAT0/charge_control_end_threshold

echo "---- Read 7----"
cat  /sys/class/power_supply/BAT0/charge_control_start_threshold
cat  /sys/class/power_supply/BAT0/charge_control_end_threshold

gnome-extensions enable Battery-Health-Charging@maniacx.github.com
```

SCV-ready, 2024-05-23:
```
---- Disabling extension ----
---- Read 1 ----
75
75
---- Write 1 Change Threshold 55/60 ----
55
60
---- Read 2----
60
55
---- Write 2 Change Threshold 75/80 ----
80
75
---- Read 3----
75
75
Write 3 Change Threshold 75/80 second attempt ----
80
75
---- Read 4----
75
75
Write 4 Change Threshold 55/60 ----
55
60
---- Read 5----
60
55
---- Write 5 Change Threshold 75/80 ----
80
75
---- Read 6----
75
75
Write 6 Change Threshold end only 80 second attempt ----
80
---- Read 7----
80
75

```
i think i saw on the source code on the devices that the firmware takes a while to update, so an await command is used somewhere between the two in the gnome extension (or is supposed to). 
i also tested applying the command and pasting the cat commands after each one. it seems that the first time only registers the lower bound and applies it to both limits, and the second try places the top threshold correctly.

```
gnome-extensions enablegnome-extensions disable Battery-Health-Charging@maniacx.github.com

echo '55' | sudo tee /sys/class/power_supply/BAT0/charge_control_start_threshold
echo '60' | sudo tee /sys/class/power_supply/BAT0/charge_control_end_threshold

echo 'post 60/55'

cat  /sys/class/power_supply/BAT0/charge_control_start_threshold
cat /sys/class/power_supply/BAT0/charge_control_end_threshold

echo 'switch'

echo '80' | sudo tee /sys/class/power_supply/BAT0/charge_control_end_threshold
echo '75' | sudo tee /sys/class/power_supply/BAT0/charge_control_start_threshold
echo '80' | sudo tee /sys/class/power_supply/BAT0/charge_control_end_threshold

echo 'post 80/75/80'

cat  /sys/class/power_supply/BAT0/charge_control_start_threshold
cat /sys/class/power_supply/BAT0/charge_control_end_threshold

gnome-extensions enablegnome-extensions enable Battery-Health-Charging@maniacx.github.com
```
running this yielded this output:
```
55
60
post 60/55
60
55
switch
80
75
80
post 80/75/80
80
75
```
so when raising thresholds applying upper -> lower -> upper happens to work.

maniacx, 2024-05-23:
In this case I can check each threshold and apply threshold again to the one that is not updated. 

I will send you a test version later, once I have some free time.

maniacx, 2024-05-26:
I have modified the extension and uploaded the new branch
https://github.com/maniacx/Battery-Health-Charging/tree/thinkpad_quirky_firmware

Please install by running ./install.sh and test it.

I have Asus laptop, but can usually simulate the different devices such as thinkpads, huawei by just changing the threshold paths to fake once.
But this kind of quirky behaviour is difficult to simulate, so please test it and if there are any issue/bugs, please check the logs.

To check logs, disable extension

Open terminal and run
```
journalctl -f -o cat /usr/bin/gnome-shell
```
keep terminal window open and enable extension, check the logs in the terminal window.

Similiarly if you cannot open prefs/settings for this extension.
Open another terminal window and check gjs logs

```
journalctl -f -o cat /usr/bin/gjs
```

SCV-ready, 2024-05-28:
testing the new branch now. i enabled the thinkpad option on the device section. looks good from initial tests, will update if it behaves weirdly.

maniacx, 2024-05-28:
Oh. For your device, I have added it is a different fix.
You do not need to Enable "Skip threshold verification" in extension preference.
Kindly disable it and test and let me know. If the fix works, I will merge it in the main branch. If it doesn't work, we will see what is the issue and if it gets to complex, I might just discard the fix and you can use it by enabling "Skip threshold verification".

Just help we with the feedback by testing it with  "Skip threshold verification" disabled.

SCV-ready, 2024-05-29:
these are all the logs.

```
> journalctf -f -o cat /usr/bin/gnome-shell

JS ERROR: TypeError: this.actor is null
_syncEnabled@resource:///org/gnome/shell/ui/windowManager.js:145:25
onStopped@resource:///org/gnome/shell/ui/windowManager.js:157:35
_makeEaseCallback/<@resource:///org/gnome/shell/ui/environment.js:84:22
_easeActorProperty/<@resource:///org/gnome/shell/ui/environment.js:250:60
_destroyWindowDone@resource:///org/gnome/shell/ui/windowManager.js:1538:21
onStopped@resource:///org/gnome/shell/ui/windowManager.js:1506:39
_makeEaseCallback/<@resource:///org/gnome/shell/ui/environment.js:84:22
_easeActor/<@resource:///org/gnome/shell/ui/environment.js:173:64
@resource:///org/gnome/shell/ui/init.js:21:20

libinput error: event4  - SynPS/2 Synaptics TouchPad: kernel bug: Touch jump detected and discarded.
See https://wayland.freedesktop.org/libinput/doc/1.25.0/touchpad-jumping-cursors.html for details
libinput error: event4  - SynPS/2 Synaptics TouchPad: kernel bug: Touch jump detected and discarded.
See https://wayland.freedesktop.org/libinput/doc/1.25.0/touchpad-jumping-cursors.html for details
libinput error: event4  - SynPS/2 Synaptics TouchPad: kernel bug: Touch jump detected and discarded.
See https://wayland.freedesktop.org/libinput/doc/1.25.0/touchpad-jumping-cursors.html for details
libinput error: event4  - SynPS/2 Synaptics TouchPad: kernel bug: Touch jump detected and discarded.
See https://wayland.freedesktop.org/libinput/doc/1.25.0/touchpad-jumping-cursors.html for details
libinput error: event4  - SynPS/2 Synaptics TouchPad: WARNING: log rate limit exceeded (5 msgs per 24h). Discarding future messages.

> journalctl -f -o cat /usr/bin/gjs 

(empty)
```

keeping both journalctl terminals open didnt show new messages after using the extension. however the behaviour is the same as from before - end threshold doesnt get updated on a single switch, panel cant get current state + percentages , and an error is raised as a notification.

maniacx, 2024-05-30:
Since **skip threshold verification** wont work for your case in future test leave is DISABLED

I have added some logs to get a better idea of what going on.

Install the updated branch and start logging using

```
journalctf -f -o cat /usr/bin/gnome-shell
```
Enable the extension.

Make changes to threshold and post result.
I made changes using github editor, so there might be some syntax error (No eslint and no time to boot my linux), hopefully not silly syntax error. try it and let me know

SCV-ready, 2024-05-30:
just installed the new branch, rebooted my machine, disabled the extension, ran this and re-enabled it and ran a bunch of changes to the limits.
```
$ journalctl -f -o cat /usr/bin/gnome-shell 
---- Battery Health Charging : Pre-write this._startValue = 75
---- Battery Health Charging : Pre-write this._oldEndValue = 80
---- Battery Health Charging : Pre-write this._oldStartValue = 80
---- Battery Health Charging : Post-write this._status = 0
---- Battery Health Charging : Post-write this._endValue = 80
---- Battery Health Charging : Post-write this._startValue = 75
---- Battery Health Charging : Post-write this._oldEndValue = 80
---- Battery Health Charging : Post-write this._oldStartValue = 80
---- Battery Health Charging : Updated START threshold with this._status = 0
---- Battery Health Charging : _reVerifyThreshold with this._status = 0
---- Battery Health Charging : ----- Start set threshold
---- Battery Health Charging : Pre-write this._endValue = 80
---- Battery Health Charging : Pre-write this._startValue = 75
---- Battery Health Charging : Pre-write this._oldEndValue = 75
---- Battery Health Charging : Pre-write this._oldStartValue = 80
---- Battery Health Charging : Post-write this._status = 0
---- Battery Health Charging : Post-write this._endValue = 80
---- Battery Health Charging : Post-write this._startValue = 75
---- Battery Health Charging : Post-write this._oldEndValue = 75
---- Battery Health Charging : Post-write this._oldStartValue = 80
---- Battery Health Charging : _reVerifyThreshold with this._status = 0
---- Battery Health Charging : ----- Start set threshold
---- Battery Health Charging : Pre-write this._endValue = 100
---- Battery Health Charging : Pre-write this._startValue = 95
---- Battery Health Charging : Pre-write this._oldEndValue = 75
---- Battery Health Charging : Pre-write this._oldStartValue = 80
---- Battery Health Charging : Post-write this._status = 0
---- Battery Health Charging : Post-write this._endValue = 100
---- Battery Health Charging : Post-write this._startValue = 95
---- Battery Health Charging : Post-write this._oldEndValue = 95
---- Battery Health Charging : Post-write this._oldStartValue = 0
---- Battery Health Charging : _reVerifyThreshold with this._status = 0
---- Battery Health Charging : ----- Start set threshold
---- Battery Health Charging : Pre-write this._endValue = 60
---- Battery Health Charging : Pre-write this._startValue = 55
---- Battery Health Charging : Pre-write this._oldEndValue = 95
---- Battery Health Charging : Pre-write this._oldStartValue = 0
---- Battery Health Charging : Post-write this._status = 0
---- Battery Health Charging : Post-write this._endValue = 60
---- Battery Health Charging : Post-write this._startValue = 55
---- Battery Health Charging : Post-write this._oldEndValue = 60
---- Battery Health Charging : Post-write this._oldStartValue = 60
---- Battery Health Charging : Updated START threshold with this._status = 0
---- Battery Health Charging : _reVerifyThreshold with this._status = 0
---- Battery Health Charging : ----- Start set threshold
---- Battery Health Charging : Pre-write this._endValue = 80
---- Battery Health Charging : Pre-write this._startValue = 75
---- Battery Health Charging : Pre-write this._oldEndValue = 55
---- Battery Health Charging : Pre-write this._oldStartValue = 60
---- Battery Health Charging : Post-write this._status = 0
---- Battery Health Charging : Post-write this._endValue = 80
---- Battery Health Charging : Post-write this._startValue = 75
---- Battery Health Charging : Post-write this._oldEndValue = 75
---- Battery Health Charging : Post-write this._oldStartValue = 80
---- Battery Health Charging : _reVerifyThreshold with this._status = 0
---- Battery Health Charging : ----- Start set threshold
---- Battery Health Charging : Pre-write this._endValue = 60
---- Battery Health Charging : Pre-write this._startValue = 55
---- Battery Health Charging : Pre-write this._oldEndValue = 75
---- Battery Health Charging : Pre-write this._oldStartValue = 80
---- Battery Health Charging : Post-write this._status = 0
---- Battery Health Charging : Post-write this._endValue = 60
---- Battery Health Charging : Post-write this._startValue = 55
---- Battery Health Charging : Post-write this._oldEndValue = 60
---- Battery Health Charging : Post-write this._oldStartValue = 60
---- Battery Health Charging : Updated START threshold with this._status = 0
---- Battery Health Charging : _reVerifyThreshold with this._status = 0
[... 33 lines cut ...]
---- Battery Health Charging : _reVerifyThreshold with this._status = 0
---- Battery Health Charging : ----- Start set threshold
---- Battery Health Charging : Pre-write this._endValue = 60
---- Battery Health Charging : Pre-write this._startValue = 55
---- Battery Health Charging : Pre-write this._oldEndValue = 95
---- Battery Health Charging : Pre-write this._oldStartValue = 0
---- Battery Health Charging : Post-write this._status = 0
---- Battery Health Charging : Post-write this._endValue = 60
---- Battery Health Charging : Post-write this._startValue = 55
---- Battery Health Charging : Post-write this._oldEndValue = 60
---- Battery Health Charging : Post-write this._oldStartValue = 60
---- Battery Health Charging : Updated START threshold with this._status = 0
---- Battery Health Charging : _reVerifyThreshold with this._status = 0
---- Battery Health Charging : ----- Start set threshold
---- Battery Health Charging : Pre-write this._endValue = 100
[... cut ...]
---- Battery Health Charging : Pre-write this._oldEndValue = 55
---- Battery Health Charging : Pre-write this._oldStartValue = 60
---- Battery Health Charging : Post-write this._status = 0
---- Battery Health Charging : Post-write this._endValue = 100
---- Battery Health Charging : Post-write this._startValue = 95
---- Battery Health Charging : Post-write this._oldEndValue = 95
---- Battery Health Charging : Post-write this._oldStartValue = 0
---- Battery Health Charging : _reVerifyThreshold with this._status = 0
```

maniacx, 2024-05-30:
That is quite strange.

It is so random,
Sometimes applied start threshold is read on both charge_control_end_threshold and charge_control_start_threshold.
Sometime applied end threshold value is read  on both charge_control_end_threshold and charge_control_start_threshold,.
 on both charge_control_end_threshold and charge_control_start_threshold, 
Sometimes applied start threshold is read in charge_control_end_threshold and end threshold is read in charge_control_start_threshold.

I cannot find any pattern here on how the threshold values are applied and read. So i do not know what logic to apply, to fix this issue.

This is something, I should have asked first before trying to fix the extension, but it forgot to ask.
Did you try to reset the Embedded controller? You can do this by
1) Resettting Bios settings to factory defaults (Note/Save all information regarding booting order or any changes you have made to the bios so that you can reapply them after doing resetting your BIOS to factory defaults.
2) Some thinkpads have a reset hold, where you can use a pin to push the button for 5 to 10 seconds to clear the embedded controller when laptop is of
3) Some models, (that do not have reset hold)  you need to press power and hold button for 10+ seconds to reset emebedded controller when laptop is off/

or 4) Also are you using the latest BIOS firmware, Although you are on your own if you decide to update BIOS firmware, as crash or power loss during BIOS update can completely break down making laptop unrepairable if BIOS firmware update get interrupted, and I do not want to be blamed for this.

Point 1, 2 and 3 have fixed these kinda off issue on some thinkpads whilst on other point 4 (BIOS update) has seems to have fixed these kinda off issue.
 I would recommend point 1, 2 or 3. . 

For bios updates would be last resort  and do it if,  only if it is not using the latest vesion and if you are comfortable updating firmware.

SCV-ready, 2024-05-30:
i am hesitant on resetting the bios settings due to remembering that some of them i changed on purpose (mostly due to stability with nvidia drivers, which is an unrelated issue), but i don't quite remember all the changes, and i would rather not tinker with it too much for stability reasons.  however, the battery/power section is default already as far as i remember.

i did an EC reset, no change. bios is latest for this model.

i know this is a wild shot, but can you insert a few calls of reading the value in `/sys/class/power_supply/BAT0/charge_control_end_threshold` and `/sys/class/power_supply/BAT0/charge_control_start_threshold` after applying the values but before verifying them? about 4 calls on each should be enough since most of the changes applied have less changes than this. (or potentially just one or two after each value set). 

this is based on the quirk i found with the updates applying only after reading for a few times, which was obvious when i ran the watch command with different intervals as it basically ran the same sequence of values but slower. i might try and tinker with the extension on my machine as well since its easier to debug locally(especially with random behaviour like this), maybe it will narrow it down a bit.

maniacx, 2024-05-30:
Yeah I think it is best if you can try and debug a little more.

Thinkpad.js is sufficient, unless just for reading and writing threshold.
I try to explain a little bit.

In thinkpad.js
Your section would be Line 10 to 18 that is related to BAT0

And the class ThinkpadSingleBatteryBAT0 Line 205 to 425, although the force discharge part is not related to you so Line 205 to 343.

---
**Writing threshold**
`runCommandCtl` is what is used to write threshold. example
```
[this._status] = await runCommandCtl(this._ctlPath, 'BAT0_END_START', `${this._endValue}`, `${this._startValue}`, null)
```
`BAT0_END_START` Can be found here in  batteryhealthchargingctl. it is bash script uses command like echo)  https://github.com/maniacx/Battery-Health-Charging/blob/834eaef54c45f41f08a95820674211fc8d3b3899/resources/batteryhealthchargingctl#L63-L67 where it will apply end threshold first before start threshold later

---
**Reading threshold**
`this._oldEndValue = readFileInt(BAT0_END_PATH)` Basically reads threshold  from `'/sys/class/power_supply/BAT0/charge_control_end_threshold'` where BAT0_END_PATH is defined here
https://github.com/maniacx/Battery-Health-Charging/blob/834eaef54c45f41f08a95820674211fc8d3b3899/devices/Thinkpad.js#L11

---
**Verify threshold** 
Reads the threshold, if read value (this._oldEndValue) is equal to applied threshold value by extension (this._endValue) (aslo same ccheck for  start threshold). then it updates UI, by emitting a signal.
 `this.emit('threshold-applied', 'success');` 
when signal is emitted, the UI part of extension will read `this.endLimitValue` and `this.startLimitValue` and this value will be display in QS Panel and Notifcation.

If signal emitted is `this.emit('threshold-applied', 'failed');` the UI will give a notification that error has occurred.

https://github.com/maniacx/Battery-Health-Charging/blob/834eaef54c45f41f08a95820674211fc8d3b3899/devices/Thinkpad.js#L318-L328

---

In summary
1.
https://github.com/maniacx/Battery-Health-Charging/blob/834eaef54c45f41f08a95820674211fc8d3b3899/devices/Thinkpad.js#L270-271
first the extension verifies, if read threshold is equal to threshold applied, this is to prevent the extension for writing threshold again if the new threshold to be applied is already applied, so no point of writing.  when extension in enabled by user,  resuming from sleep, or hibernation or loggin in. If equal it updates UI and exits class

2.
 if not equal it will write new threshold.
https://github.com/maniacx/Battery-Health-Charging/blob/834eaef54c45f41f08a95820674211fc8d3b3899/devices/Thinkpad.js#L276-279

3. Verifies again, if correctly applied updates UI and exit class. If incorrectly update continue with 4

4.
This is what I added as a fix for your model.
https://github.com/maniacx/Battery-Health-Charging/blob/834eaef54c45f41f08a95820674211fc8d3b3899/devices/Thinkpad.js#L298-304
As i thought with earlier testing that end threshold doesnt get updated and need to be applied again.  but seems like this isnt the case. The original extension doesnt contain step 4

5.
https://github.com/maniacx/Battery-Health-Charging/blob/834eaef54c45f41f08a95820674211fc8d3b3899/devices/Thinkpad.js#L330-342

After verification fails at step3 create a delay, wait for 200ms and call `reverifyThreshold()` which calls verifythreshold() again to check if threshold are applied correctly, it is still fails to update after 200ms, it emit signal informing threshold-applied has failed

Happy debugging

closed 2024-10-16 by maniacx