maniacx/Battery-Health-Charging #94: Another Samsung laptops

albersonmiranda, 2024-05-30:
Extension isn't compatible with my Samsung X30 (2020):
```
# dmidecode 3.3
Getting SMBIOS data from sysfs.
SMBIOS 3.2.0 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
	Manufacturer: SAMSUNG ELECTRONICS CO., LTD.
	Product Name: 550XCJ/550XCR
	Version: P14RFH
	Family: Notebook 5 Series
```
Nonetheless, I do have /sys/class/power_supply/BAT1/capacity currently set to 85 (legacy from previous Windows installation). Also sys/class/power_supply/BAT1/uevent with all settings:
```
POWER_SUPPLY_NAME=BAT1
POWER_SUPPLY_TYPE=Battery
POWER_SUPPLY_STATUS=Charging
POWER_SUPPLY_PRESENT=1
POWER_SUPPLY_TECHNOLOGY=Li-ion
POWER_SUPPLY_CYCLE_COUNT=45
POWER_SUPPLY_VOLTAGE_MIN_DESIGN=11400000
POWER_SUPPLY_VOLTAGE_NOW=12230000
POWER_SUPPLY_CURRENT_NOW=0
POWER_SUPPLY_CHARGE_FULL_DESIGN=3780000
POWER_SUPPLY_CHARGE_FULL=3500000
POWER_SUPPLY_CHARGE_NOW=2975000
POWER_SUPPLY_CAPACITY=85
POWER_SUPPLY_CAPACITY_LEVEL=Normal
POWER_SUPPLY_MODEL_NAME=SR Real Battery
POWER_SUPPLY_MANUFACTURER=SAMSUNG Electronics
POWER_SUPPLY_SERIAL_NUMBER=123456789
```

I can't change it back to 100 to test it but it seems to be a way.

maniacx, 2024-05-30:
If you laptop X30 notebook has the feature to change charging threshold, then most likely you can linux kernel will support it and so will this extension.
The charging threshold is not a software  feature but is a hardware feature that manufcaturer provide on a BIOS level, so the charging threshold are applied even when the laptop is powered OFF.
So here are some question.
1. Does you laptop, have the feature to turn on/off "Battery life extender" in Windows or in BIOS. Are you able to use this feature? If no, then most likely you wont find support in Linux as well
2.  If yes, than most likely  samsungs WMI for linux kernel will detect it (unless the way to read Embeeded controller is different)
If linux kernel has detected your model is compatible, these path will be exposed in sysfs. you can check by using ls command in terminal

```
ls /sys/devices/platform/samsung/
```
You should see `battery_life_extender` sysfs path in this output.

If the path is present you can enable charging threshold by writing 1 to it or disable charging threshold by writing 0

```
echo '1' | sudo tee /sys/devices/platform/samsung/battery_life_extender

```
If this works, then the extension is compatible with your model.

Post your output for `ls /sys/devices/platform/samsung/` and you kernel version and your linux distribution your are using (ex Fedora 39, Ubuntu 22.04)

albersonmiranda, 2024-06-01:
@maniacx yep, I used to let it turned on in my previous installation (Windows) via Samsung settings. Then I made a Ubuntu fresh install (no dual boot) and now I can't turn it back to 100%.

Unfortunately, `/sys/devices/platform/samsung/` doesn't exists:
`ls: não foi possível acessar '/sys/devices/platform/samsung/': Arquivo ou diretório inexistente` (pt-BR 🇧🇷)

I'm on
OS: Ubuntu 22.04.4 LTS x86_64 
Host: 550XCJ/550XCR P14RFH 
Kernel: 6.5.0-35-generic 
DE: GNOME 42.9 
WM: Mutter

maniacx, 2024-06-01:
post the output of these command

```
echo '--- sys_vendor ---'
cat /sys/devices/virtual/dmi/id/sys_vendor

echo '--- board_name ---'
cat /sys/devices/virtual/dmi/id/board_name

echo '--- chassis_type ---'
cat /sys/devices/virtual/dmi/id/chassis_type

echo '--- product_name ---'
cat /sys/devices/virtual/dmi/id/product_name

echo '--- ls-bat ---'
ls -l /sys/class/power_supply/BAT1/

echo '--- ls-platform ---'
ls -l /sys/devices/platform/

echo '--- ls-platform ---'
ls -l /sys/devices/platform/

echo '--- ls-module ---'
ls -l /sys/module/
```

albersonmiranda, 2024-06-01:
```
--- sys_vendor ---
SAMSUNG ELECTRONICS CO., LTD.
--- board_name ---
NP550XCJ-XS1BR
--- chassis_type ---
10
--- product_name ---
550XCJ/550XCR
--- ls-bat ---
total 0
-rw-r--r-- 1 root root 4096 jun  1 16:36 alarm
-r--r--r-- 1 root root 4096 jun  1 08:52 capacity
-r--r--r-- 1 root root 4096 jun  1 16:36 capacity_level
-r--r--r-- 1 root root 4096 jun  1 15:19 charge_full
-r--r--r-- 1 root root 4096 jun  1 16:36 charge_full_design
-r--r--r-- 1 root root 4096 jun  1 15:19 charge_now
-r--r--r-- 1 root root 4096 jun  1 15:19 current_now
-r--r--r-- 1 root root 4096 jun  1 08:52 cycle_count
lrwxrwxrwx 1 root root    0 jun  1 16:36 device -> ../../../PNP0C0A:00
drwxr-xr-x 3 root root    0 jun  1 08:52 hwmon2
-r--r--r-- 1 root root 4096 jun  1 16:36 manufacturer
-r--r--r-- 1 root root 4096 jun  1 16:36 model_name
drwxr-xr-x 2 root root    0 jun  1 16:36 power
-r--r--r-- 1 root root 4096 jun  1 15:19 present
-r--r--r-- 1 root root 4096 jun  1 16:36 serial_number
-r--r--r-- 1 root root 4096 jun  1 15:19 status
lrwxrwxrwx 1 root root    0 jun  1 14:34 subsystem -> ../../../../../../../../../class/power_supply
-r--r--r-- 1 root root 4096 jun  1 16:36 technology
-r--r--r-- 1 root root 4096 jun  1 08:52 type
-rw-r--r-- 1 root root 4096 jun  1 14:34 uevent
-r--r--r-- 1 root root 4096 jun  1 15:19 voltage_min_design
-r--r--r-- 1 root root 4096 jun  1 15:19 voltage_now
--- ls-platform ---
total 0
drwxr-xr-x 3 root root    0 jun  1 16:36  ACPI0003:00
drwxr-xr-x 3 root root    0 jun  1 16:36  ACPI000C:00
drwxr-xr-x 3 root root    0 jun  1 16:36  acpi-cpufreq
drwxr-xr-x 4 root root    0 jun  1 08:52  coretemp.0
drwxr-xr-x 3 root root    0 jun  1 16:36  efivars.0
drwxr-xr-x 3 root root    0 jun  1 16:36  eisa.0
drwxr-xr-x 4 root root    0 jun  1 16:36 'Fixed MDIO bus.0'
drwxr-xr-x 4 root root    0 jun  1 08:52  i8042
drwxr-xr-x 3 root root    0 jun  1 16:36  INT33A1:00
drwxr-xr-x 4 root root    0 jun  1 08:52  INT3400:00
drwxr-xr-x 3 root root    0 jun  1 08:52  intel_rapl_msr.0
drwxr-xr-x 3 root root    0 jun  1 16:36  kgdboc
drwxr-xr-x 3 root root    0 jun  1 16:36  microcode
drwxr-xr-x 3 root root    0 jun  1 16:36  MSFT0101:00
drwxr-xr-x 3 root root    0 jun  1 16:36  pcspkr
drwxr-xr-x 3 root root    0 jun  1 16:36  PNP0C0C:00
drwxr-xr-x 3 root root    0 jun  1 16:36  PNP0C0D:00
drwxr-xr-x 4 root root    0 jun  1 08:52  PNP0C14:03
drwxr-xr-x 4 root root    0 jun  1 08:52  PNP0C14:04
drwxr-xr-x 4 root root    0 jun  1 08:52  PNP0C14:05
drwxr-xr-x 2 root root    0 jun  1 16:36  power
drwxr-xr-x 4 root root    0 jun  1 08:52  reg-dummy
drwxr-xr-x 3 root root    0 jun  1 08:52  regulatory.0
drwxr-xr-x 3 root root    0 jun  1 16:36  rtc-efi.0
drwxr-xr-x 3 root root    0 jun  1 16:36  SAM0427:00
drwxr-xr-x 5 root root    0 jun  1 08:52  serial8250
drwxr-xr-x 3 root root    0 jun  1 08:52  snd-soc-dummy
-rw-r--r-- 1 root root 4096 jun  1 11:25  uevent
drwxr-xr-x 6 root root    0 jun  1 08:52  USBC000:00
--- ls-platform ---
total 0
drwxr-xr-x 3 root root    0 jun  1 16:36  ACPI0003:00
drwxr-xr-x 3 root root    0 jun  1 16:36  ACPI000C:00
drwxr-xr-x 3 root root    0 jun  1 16:36  acpi-cpufreq
drwxr-xr-x 4 root root    0 jun  1 08:52  coretemp.0
[... 280 lines cut ...]
drwxr-xr-x 5 root root    0 jun  1 08:52 videodev
drwxr-xr-x 3 root root    0 jun  1 16:36 virtio_blk
drwxr-xr-x 3 root root    0 jun  1 16:36 virtio_mmio
drwxr-xr-x 3 root root    0 jun  1 16:36 virtio_net
drwxr-xr-x 3 root root    0 jun  1 16:36 virtio_pci
drwxr-xr-x 2 root root    0 jun  1 16:36 virtio_pci_legacy_dev
drwxr-xr-x 2 root root    0 jun  1 16:36 virtio_pci_modern_dev
drwxr-xr-x 3 root root    0 jun  1 16:36 virtual_root
drwxr-xr-x 3 root root    0 jun  1 16:36 vt
drwxr-xr-x 3 root root    0 jun  1 16:36 watchdog
drwxr-xr-x 7 root root    0 jun  1 08:52 wmi
drwxr-xr-x 6 root root    0 jun  1 08:52 wmi_bmof
drwxr-xr-x 3 root root    0 jun  1 16:36 workqueue
drwxr-xr-x 6 root root    0 jun  1 08:52 x86_pkg_temp_thermal
drwxr-xr-x 3 root root    0 jun  1 16:36 xen
drwxr-xr-x 3 root root    0 jun  1 16:36 xen_acpi_processor
drwxr-xr-x 3 root root    0 jun  1 16:36 xen_blkfront
drwxr-xr-x 3 root root    0 jun  1 16:36 xen_netfront
drwxr-xr-x 3 root root    0 jun  1 16:36 xhci_hcd
drwxr-xr-x 6 root root    0 jun  1 08:52 xhci_pci
drwxr-xr-x 5 root root    0 jun  1 08:52 xhci_pci_renesas
drwxr-xr-x 5 root root    0 jun  1 08:52 x_tables
drwxr-xr-x 2 root root    0 jun  1 16:36 xz_dec
drwxr-xr-x 3 root root    0 jun  1 16:36 zswap
```

maniacx, 2024-06-01:
There are not charging threshold exposed in your sysfs. Also samsungs wmi is not loaded.
your systems dmi  matches the one in kernel, so i thing some other criteria isnt matched that is preventing loading this module

https://github.com/torvalds/linux/blob/ec9eeb89e60d86fcc0243f47c2383399ce0de8f8/drivers/platform/x86/samsung-laptop.c#L1510-L1515

Is this a stock kernel? Is everythings else working in your laptop such as keybaoard backlight, screen turning off when closing laptop lid, are you able to control screen brightness?

albersonmiranda, 2024-06-01:
I guess this laptop doesn't have backlight (Fn + F9 turns screen off), but screen does turns off when closing lid and I can control brightness and volume with F keys. Yes, it's stock Ubuntu kernel.

maniacx, 2024-06-02:
Unfortunately I cannot support, if the charging threshold is not supported by the kernel or library. Usually other laptops are able to change charging threshold using command-line and this extension just provides a GUI and execute those command. But in your case, it is not, so I apologize, there is nothing I can do.

albersonmiranda, 2024-06-02:
No worries, thanks for taking a look!

closed 2024-06-02 by albersonmiranda