maniacx/Battery-Health-Charging #95: Doubts about how this works

datalot-369, 2024-06-11:
- After OneUI 6.1, Samsung introduced a battery limit of 80%.
- However, it is well-known that phones need a battery to work so, hypothetically, removing the phone's battery while charging would turn off the phone.
- This is not true for laptops. You can use your laptop without any connected battery, as long as it is connected to a power supply.
- Now, I'm on Fedora 40, with GNOME 46, and I just noticed how functional and useful is the integrated 'Power Modes', which makes me wonder why aren't there any 'Battery Protection' features built-in.
- I discovered this extension through the Extensions Manager after searching for something that could provide that in Linux/Fedora.
- I installed the dependencies through `sudo dnf install  libsmbios-2.4.3-13.fc40.x86_64`, but the extension didn't work, so I executed `smbios-battery-ctl --help` and Fedora automatically suggested me to install `smbios-utils-python.x86_64`, so I installed it and the extension worked like a charm. Thanks, Fedora.
- I followed the installation instructions (which are actually easy and intuitive, good job), set the **Balanced Battery Mode**, and waited for the laptop to charge.
- It charged from 19% to 79% and the charging stopped. The laptop's LED turned off.  The laptop is running at 0.0W from the battery according to **Vitals GNOME Extension**, but I'm confused because the battery is not charging or discharging. It is holding that 79% (I've not disconnected the power yet). So:
- Is the battery in use while I'm still connected to the power?
- Is my laptop running directly from the AC Charger?
- Or is the battery charging and discharging 1% constantly to hold that 79%? (that's how smartphones work)
- Thanks in advance.

maniacx, 2024-06-11:
I am assume most laptop behaves the same but could be different, it all depend on how manufacturer implements it.
But for most laptop it works exactly the same how you described it. Once it reaches 80%, the battery will discharge at an extremely slow rate. (On my laptop to go from 80% to 79% takes few hours). It kind off like battery is neither gets charged by the charger nor powers the laptop. 

It is definitely not  cycling from 80% to 79%. (AFAIK) For dell laptop minimum threshold difference is 5%, for example if you set 80/75 threshold it will except is. But if you set 80/78 it wont except the threshold values. If you set it the end threshold to 80 and start threshold to 75, it should definitely cycle form 80 to 75, but charging from 75 to 80 is at normal charging speed, but discharging from 80 to 75 will take a couple of hours. I would observe the behavior for one or 2 days.

datalot-369, 2024-06-12:
Main: My concern is whether or not is better for battery health to work with the power connected or connect and disconnect the power to cycle battery use normally. I have now set it to _Max lifespan_ while I work with power connected.

Off-topic: Why do you hate Samsung 24+ !???  That's exactly the phone I want to buy by the end of this year 😭 I love Samsung's OneUI, so I wouldn't change that plan just for the Android Skin.

maniacx, 2024-06-12:
For better battery health, max lifespan is best, however most manufacturer (asus, lenovo) which were offering 60% limit have moved to 80% as there is marginal difference in battery health when it come to 60% vs 80% limit.

It is better for battery health if you connect power and use threshold like 80% or 60% rather than manually connecting and disconnecting. The more you cycle the battery (like charging 80% and discharging to below 20%) , the degradation is more. 

Off-topic:  Well, since you want to buy S24 , I list what I do not like about it.
1. The screen look dull, no matter which color mode, or temperature I use. Comparing my oneplus 7 pro and my work phone iphone 13, I prefer oneplus 7 pro and iphone screen. (although this screen is much sharper, I always use the phone in max resolution). Later I found out, that s23 user also complained on reditt that S23 screen was better S24. I think it is due to the new technology they implemented is the screen to reduce reflections. But I still see myself when watching some dark/ night scene in movie on netflix. What kind off reflection issue they solved, go figure.

2. The screen scratches easily, it has been only 2 months and my screen is already cracked and the amount scratches on it is a lot, and they are increasing with time. My usage is the same as oneplus pro nothing has changed, and on oneplus 7 pro device with almost 4 years of usage, countless drops, I do not have a single crack, and only 2 visible scratches.

3. I cannot use configure the power button to trigger samsung pay (There is option only for google pay) . But google pay can be payed without unlocking which is a security risk.

4.  everything is double. 2 app stores, 2 payment apps, 2 browsers, I hate those bloats and cannot even unistall them.

5.  I miss the alert slider is oneplus, I could put it in silent, without having to unlock the phone.

6.  The gallery app do not have button google lens only bixby and bixby is dumb. One oneplus the google lens button, which made it much easier to copy paste content or translate, but I found way, I use share option and then use google lens, 3 click more to do this so it is not so bad, atleast it is doable.

7. When I hold the phone , due to sharp edges the feeling is annoying (I thought I will get use to it but still didnt) 

8. Some SMS do not have delete button in notifications. now I have to open the messaging app everytime I have to delete unwanted messages.

oxygen os was well designed to made things much easier. I regret so much not going for the oneplus again which only 700 USD, and paid 1000 USD (300 more) for this device for nothing. The only reason I steered toward samsung was because they promised 5 years updates.

closed 2024-06-12 by datalot-369

datalot-369, 2024-06-13:
I'm amazed at how this continues to work when the laptop is on, idle, suspended, and even turned off.
Wow.

datalot-369, 2024-06-13:
1. And as for Samsung, you said you have the S24+, not the S24 Ultra. Only the S24 Ultra has that display technology, not base or plus models. Maybe you are just used to regular displays? I always use 'Natural' mode. It makes skin tones look so accurate in photos.
2. About scratches, I don't know. I've never had a phone (Samsung or not) without a respective screen protector and case.
3. Here in Venezuela we don't have such payment options available, no matter the company. Maybe you can set a Routine that launches when Bixy starts?
4. It's true that Samsung has made some bad decisions about the integration of two ecosystems, but it's actually difficult to avoid that situation. I just put the apps I don't use in a folder and uninstall/disable the ones that can be uninstalled/disabled. I use everything from Samsung instead of their Google alternatives: Calendar, Contacts, Messages, Clock, etc.
5. I think you can do that with the routines widget in lockscreen, or shaking the phone, but that's a matter of looking for that in the routines.
6. I have no idea how you could need the Google Lens button since you have Circle to Search available in the whole system. Samsung's Text button is good and well-integrated too.
7. I'm scared of using a naked phone.
8. Hm... Interesting that you delete your messages instantly. I have a 3 years old history in all my communication platforms e.e'

maniacx, 2024-06-14:
> I'm amazed at how this continues to work when the laptop is on, idle, suspended, and even turned off. Wow.

Yes, those changes are applied to embedded controller of the laptop, so i works even when it is powered off and no operating system running.

Regarding S24.
Haha. I am the opposite, I never use screen protector. Case, yes when oneplus was new for about 6 months, but later I never used one. And yes I delete unimportant message instantly if they are ads / offers and other junk stuff.
Thanks though, I didn't see or realize that circle text button was there, It does work, although I still prefer Lens. I assumed that eye icon on the top right corner was suppose to do that, and all it showed me was discover and wine no idea what that does.

I guess, i just have to get used to the new eco system, but if you are buying this model, you might check on the screen, coz it could be a deal breaker.