maniacx/Battery-Health-Charging #98: MSI Compatability

Gene-W, 2024-07-18:
Hello,
I see in the doc that MSI is supported and I have installed [msi-ec](https://github.com/BeardOverflow/msi-ec) but when I switch Battery Health Charging on I receive an incompatibility error message.

Mine is MSI Delta A5EFK. Is this model not compatible or have I missed something?

maniacx, 2024-07-19:
Charging threshold is a hardware feature provided by the manufacturers.
If your laptop has options to control charging threshold in Windows or in BIOS, chance are it will work with linux.

Extension will work if it finds a sys_fs path to change charging threshold.
In terminal try this commnad.
`
```
uname -r
ls -l /sys/devices/platform/msi-ec/
ls -l /sys/class/power_supply/
ls -l /sys/class/power_supply/BAT0/charge_control_end_threshold
ls -l /sys/class/power_supply/BAT1/charge_control_end_threshold
```
Post the output here.
if `ls -l /sys/class/power_supply/` will show either BAT0 or BAT1

if it contains BAT0 try this command
```
echo '60' | sudo tee /sys/class/power_supply/BAT0/charge_control_end_threshold
```

if it contains BAT1 try this command
```
echo '60' | sudo tee /sys/class/power_supply/BAT1/charge_control_end_threshold
```
Post the output here. and check if your battery stops charging at 60.

radical4ever, 2024-09-24:
> Charging threshold is a hardware feature provided by the manufacturers. If your laptop has options to control charging threshold in Windows or in BIOS, chances are it will work with linux.
> 
> Extension will work if it finds a sys_fs path to change charging threshold. In terminal try this commnad. `
> 
> ```
> uname -r
> ls -l /sys/devices/platform/msi-ec/
> ls -l /sys/class/power_supply/
> ls -l /sys/class/power_supply/BAT0/charge_control_end_threshold
> ls -l /sys/class/power_supply/BAT1/charge_control_end_threshold
> ```
> 
> Post the output here.
> 
> `ls -l /sys/class/power_supply/` will show either BAT0 or BAT1
> 
> if it contains BAT0 try this command
> 
> ```
> echo '60' | sudo tee /sys/class/power_supply/BAT0/charge_control_end_threshold
> ```
> 
> if it contains BAT1 try this command
> 
> ```
> echo '60' | sudo tee /sys/class/power_supply/BAT1/charge_control_end_threshold
> ```
> 
> Post the output here. and check if your battery stops charging at 60.

Hi, so i've tried this but when im in the part to set the charging threshold it pops me out "permission denied " even if it uses sudo

maniacx, 2024-09-25:
@radical4ever 
You havent provided the output of those command I posted so it is difficult to me to check, what is going on

I think "permission denied" usually shows when the **charge_control_end_threshold** is not available in sysfs, although it could be other issues as well.

First check if charge_control_end_threshold is available,  `ls -l /sys/class/power_supply/BAT0/charge_control_end_threshold`

If charge_control_end_threshold is  available try `pkexec` instead of `sudo`

If it is absent check for file `ls -l /sys/devices/platform/msi-ec/` which confirms that the module is loaded.
Also check for the kernel version `uname -r`
I think kernel older than 6.4 does not support charging_threshold.

https://maniacx.github.io/Battery-Health-Charging/device-compatibility/msi

radical4ever, 2024-09-25:
> @radical4ever You havent provided the output of those command I posted so it is difficult to me to check, what is going on
> 
> I think "permission denied" usually shows when the **charge_control_end_threshold** is not available in sysfs, although it could be other issues as well.
> 
> First check if charge_control_end_threshold is available, `ls -l /sys/class/power_supply/BAT0/charge_control_end_threshold`
> 
> If charge_control_end_threshold is available try `pkexec` instead of `sudo`
> 
> If it is absent check for node `ls -l /sys/devices/platform/msi-ec/` which confirms that the module is loaded. Also check for the kernel version `uname -r` I think kernel older than 6.4 does not support charging_threshold.
> 
> if it is older than 6.4 you have to install the kernel module seperately
> 
> https://github.com/BeardOverflow/msi-ec
> 
> FYI https://maniacx.github.io/Battery-Health-Charging/device-compatibility/msi

Hi, srry for not sending the output, in this case i tried both of the checks returning that the files doesnt exists
`ls -l /sys/devices/platform/msi-ec/
ls -l /sys/class/power_supply/
ls -l /sys/class/power_supply/BAT0/charge_control_end_threshold
ls -l /sys/class/power_supply/BAT1/charge_control_end_threshold
6.11.0-63.fc41.x86_64
ls: no se puede acceder a '/sys/devices/platform/msi-ec/': No existe el fichero o el directorio
total 0
lrwxrwxrwx. 1 root root 0 sep 25 17:08 ADP1 -> ../../devices/pci0000:00/0000:00:1f.0/PNP0C09:00/ACPI0003:00/power_supply/ADP1
lrwxrwxrwx. 1 root root 0 sep 25 17:08 BAT1 -> ../../devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:50/PNP0C09:00/PNP0C0A:00/power_supply/BAT1
ls: no se puede acceder a '/sys/class/power_supply/BAT0/charge_control_end_threshold': No existe el fichero o el directorio
ls: no se puede acceder a '/sys/class/power_supply/BAT1/charge_control_end_threshold': No existe el fichero o el directorio`

and when i tried `ls -l /sys/devices/platform/msi-ec/` puts again that it doesnt exists, i already installed msi-ec module but i still got the same messages

maniacx, 2024-09-25:
But for some reason msi-ec doesnot load.
I think not all msi models are supported.
First I would check you vendor id

```
cat '/sys/devices/virtual/dmi/id/sys_vendor'
```
It should be either `MICRO-STAR INT` or `Micro-Star International`

I think it best create an github issue at msi-ec kernel module.

https://github.com/BeardOverflow/msi-ec/issues

Provide them with all details 

Kernel version:
 ```
uname -r
```
Laptop Vendor name: 
```
cat /sys/devices/virtual/dmi/id/sys_vendor`
```
Laptop Model no:
```
cat /sys/devices/virtual/dmi/id/product_name`
```
EC firmware version:
```
cat /sys/devices/virtual/dmi/id/ec_firmware_release
```
Laptop Board name:
```
cat /sys/devices/virtual/dmi/id/board_name`
```

And explain that msi-ec is not loaded. May be your laptop is not yet supported. They might ask you for few information and help and add your make model.

maniacx, 2024-09-26:
@radical4ever 
Here are also the list of devices supported

https://github.com/glpnk/msi-ec/blob/main/docs/supported_devices.md

radical4ever, 2024-09-26:
@maniacx So after doing all this i realized i didnt check the compatibility very well and now that im putting all the commands that you told me to use to make the issue i realized there's no compatibilty for my laptop 🤣, so thanks anyways for helping me out with this, i'm going to open the request on their repo for compatibility with my laptop model (GF63 THIN 10SC)

closed 2024-10-16 by maniacx