maniacx/Battery-Health-Charging #99: feature - hide the button from the quick settings

baronleonardo, 2024-08-08:
Could we have a setting to hide the quick settings button, for me at least I set this and never change it again, so the visible button is kinda useless at this situation.
I would really appreciate that :) (I'm still appreciating you work even if this feature doesn't get implemented :D)

[image]

maniacx, 2024-08-10:
If I have give a feature to hide it, I have to find another way of changing the charging limit/threshold.

Hiding is easy part, but finding another way to change is the hard part.

I am not dismissing the idea yet, but I can take month to implement it (If I figure out to UI design part) as I very busy currently.
For now, I will only address bugs ,port to new Gnome version and supporting newer devices.

As a workaround (but not a straight forward way), you can use this extension.

https://github.com/qwreey/quick-settings-tweaks

The only problem is,  _Quick-settings-tweaks_ does not list _Battery Health Charging_ Quick Toggles (**ChargeLimit**) . **ChargeLimit** toggle only adds to the quicksettings menu, after it verifies that laptop supports threshold and verifies the threshold is applied correctly. It could be that _quick-settings-tweaks_ extensions gets the list before **ChargeLImit** toggle is added.

So you will have to disable and renable Quick-settings-tweaks for **ChargeLImit** to show in the list and hide/unhide the **ChargeLImit** quick toggle.

[image]

baronleonardo, 2024-08-10:
I mean we could configure it then hide it, and if we need to reconfigure it, we could go to the extension settings and make it visible again .. this will only need an option in the configuration window (this could be first stage if we could improve it more as adding the charges limit in the configuration window itself)

thanks for the workaround

closed 2026-04-24 by maniacx