maniacx/Battery-Health-Charging #100: Consider using framework_tool for Framework laptops?

paulvt, 2024-08-11:
Framework has a repository for the official tool for configuring the embedded controller: <https://github.com/FrameworkComputer/framework-system>. Would you consider using the official `framework_tool` as an alternative for the currently supported unofficial kernel module? Kernel modules are also always kind of an hassle, especially with signed kernels. 

Viewing/changing the threshold is done like this:

```shell
$ pkexec pkexec framework_tool --driver portio --charge-limit 80
Minimum 0%, Maximum 80%
```

Maybe it is possible support two back-ends, depending on the presence of `framework_tool` in the path or the presence of `/sys/class/power_supply/BAT1/charge_control_end_threshold` (i.e. the kernel module is loaded)?

maniacx, 2024-08-12:
What is the default location of framework_tool  when installed?

paulvt, 2024-08-12:
I myself build it manually because it is not packages for Debian, so I put it in `/usr/local/bin`. If distributed, it is probably just in `/usr/bin`. I think we can assume it is somewhere in the `$PATH`?

maniacx, 2024-08-12:
1. What happens if I add option --driver to kernel 6.10? Would it throw an error?

2. Can you run the tool in command line and post the exact command and the output returned by the tool.

Post the output of all commands related to charging threshold. Is there a read threshold command? Or only write threshold command?

paulvt, 2024-08-12:
> 1. What happens if I add option --driver to kernel 6.10? Would it throw an error?
> 

No it's using the portable I/O driver, which also works.

>     2. Can you run the tool in command line and post the exact command and the output returned by the tool.
> 
> Post the output of all commands related to charging threshold. Is there a read threshold command? Or only write threshold command?

It has both. When not passing an argument it is read:

Reading the charge limit:
```shell
$ pkexec framework_tool --driver portio --charge-limit
Minimum 0%, Maximum 80%
```

Writing 80% as charge limit:
```shell
$ pkexec framework_tool --driver portio --charge-limit 80
Minimum 0%, Maximum 80%
```

maniacx, 2024-08-12:
Also does everyone use portio driver or would there be users using cross_ec as driver on Linux?

paulvt, 2024-08-12:
No problem at all.

It uses the cros_ec driver by default, but this requires Linux 6.10.  This requirement only holds for the AMD version of the Framework laptops, the Intel ones already have cros_ec present. I believe there may be issues if the kernel is locked down when using portio, but I have Secure Boot disabled at the moment so I cannot test.

paulvt, 2024-08-12:
I guess a generic test could be for to check the existence of `/dev/cros_ec` and fall back to the portio driver otherwise?

maniacx, 2024-08-12:
Ok.
So my plan was to do this.

`if` kernel is >=  6.10
use `pkexec framework_tool --charge-limit 80`
`else`
use `pkexec framework_tool --driver portio  --charge-limit 80`

But I think the below would be a better way as you suggested.

`if` file `/dev/cros_ec` exist
use `pkexec framework_tool --driver cros_ec  --charge-limit 80`
`else`
use `pkexec framework_tool --driver portio  --charge-limit 80`

I wont do a kernel check as it would apply across all kernel versions. What do you think about this?

paulvt, 2024-08-13:
Seems good! There are still cases when it won't work, but I guess this also holds for the other kmod approach.

JanaCoulomb, 2024-08-15:
Actually had the same Idea of using framework_tool. Went down the rabbit hole of learning how gnome extensions work  😅 when i saw this thread i decided to clean my code a tiny bit and make this pull request: https://github.com/maniacx/Battery-Health-Charging/pull/101

maniacx, 2024-08-16:
@JanaCoulomb @paulvt
I have push a new branch just for reference

https://github.com/maniacx/Battery-Health-Charging/tree/framework_tool

Little explaination.

If only Kmod or framework_tool is installed, it will use the it.
If both Kmod or framework_tool and installed, it will provide the option in extensions preference > device allowing user to choose sysfs node or framework tool.

JanaCoulomb 
Will like your help , kindly submit some pathes.

1. For checking `/dev/ec_tool` , I have done it only once, during extension initialization. 
But I think it is better to do like you have done in your patch,  as I have seen in you added checks, at every threshold  update, instead of just once during extension initiliazation.  This will help incase user decides to remove ec_tool at runtime, falling back to portio.

2. You can remove `match` and use your logic it `parseInt(this._stdout.split(" ")[3].slice(0,-1));` I will leave this upto you.

3. Also add your code `readFile('/sys/devices/virtual/dmi/id/sys_vendor').includes("Framework"))` to check if laptop is frameworks. If you add this there is no need to for `if (!fileExists(VENDOR_FRAMEWORK))` so we can remove that.

4. keep threshold function sysfs and framework_tool seperate (I have already done it here in this patch). Helps during debugging (framework_tool or sysfs) , I can concetrate on that function without worry about breaking the other and retestinng everything.

I think it is better to give user a prefs to choose sysfs and framework_tool, helps during debugging, and will help incase framework_tool doesn't work
But I will leave final decision upto @JanaCoulomb and @paulvt.

Apologize, I will very busy, following weeks so expect delayed response from me.

JanaCoulomb. Thanks for learning and submitting those patches. Keep them coming. Let me know when it is finalized. I will merge all of it together.

paulvt, 2024-08-16:
I tried this extension by running `install.sh` from that branch. However, I get the notification:

> **Battery Health Charging Error**
> Missing dependencies

When I click on "Show details" in the notification, I do not get the details, but I see in the journal:
```
Aug 16 20:27:56 frame gnome-shell[92743]: JS WARNING: [file:///home/paul/.local/share/gnome-shell/extensions/Battery-Health-Charging@maniacx.github.com/lib/notifier.js 169]: Too many arguments to function Gio.app_info_launch_default_for_uri: expected 2, got 4
```

I am using GNOME Shell 46.4.

maniacx, 2024-08-16:
Need to replace

`if (!fileExists(VENDOR_FRAMEWORK))`

with

`if(!readFile('/sys/devices/virtual/dmi/id/sys_vendor').includes("Framework")))`

and add  readFile

`const {fileExists, readFileInt, readFile, runCommandCtl}`

paulvt, 2024-08-16:
Same error still, unfortunately. (I checked that it installed with the changes.)

maniacx, 2024-08-16:
> Aug 16 20:27:56 frame gnome-shell[92743]: JS WARNING: [file:///home/paul/.local/share/gnome-shell/extensions/Battery-Health-Charging@maniacx.github.com/lib/notifier.js 169]: Too many arguments to function Gio.app_info_launch_default_for_uri: expected 2, got 4

Thanks for bringing this up. I will have to check later what changed with this.

> Need to replace
> 
> if (!fileExists(VENDOR_FRAMEWORK))
> 
> with
> 
> if(!readFile('/sys/devices/virtual/dmi/id/sys_vendor').includes("Framework")))

I accidently added another close round bracket which would cause syntax error
Make sure your text is exactly like this.
```
    isAvailable() {
        if(!readFile('/sys/devices/virtual/dmi/id/sys_vendor').includes("Framework"))
            return false;

        this._hasSysfsNode = fileExists(BAT1_END_PATH);
```

After installation need to restart you gnome-shell. That is logout and relogin (wayland) or ALT+F2 and r for xorg.

Verify you dmi by running cat command

```
cat /sys/devices/virtual/dmi/id/sys_vendor
```

maniacx, 2024-08-16:
Also lin 125
Replace
```
        output = await runCommandCtl(this._ctlPath, 'FRAMEWORK_TOOL_THRESHOLD_WRITE', this._frameworkToolDriver, this._endValue, null);
```
with
```
        output = await runCommandCtl(this._ctlPath, 'FRAMEWORK_TOOL_THRESHOLD_WRITE', this._frameworkToolDriver, `${this._endValue}`, null);

```

paulvt, 2024-08-17:
Yes, now it loads without complaints. It asked me to install the extension with root permissions, did so.
Logged in anew again and now the extension is loaded, but there is no system indicator, no item in the quick settings panel. It is as if it isn't loaded.

```shell
$ cat /sys/devices/virtual/dmi/id/sys_vendor
Framework
```

paulvt, 2024-08-17:
Oh, after a few minutes, even though there is no UI, the GNOME Shell threw an error that things are wrong.
The journal reveals:

```
Aug 17 08:06:27 frame pkexec[119352]: pam_unix(polkit-1:session): session opened for user root(uid=0) by paul(uid=1000)
Aug 17 08:06:27 frame pkexec[119352]: paul: Executing command [USER=root] [TTY=unknown] [CWD=/home/paul] [COMMAND=/usr/local/bin/batteryhealthchargingctl-paul FRAMEWORK_TOOL_THRESHOLD_WRITE cros_ec 100]
```

For the `framework_tool`, the Cros EC parameter is `cros-ec` (dash, not underscore), so I updated it. Diff:

```diff
@@ -54,7 +54,7 @@ export const FrameworkSingleBatteryBAT1 = GObject.registerClass({
         this._settings.set_boolean('detected-framework-tool', this._hasFrameworkTool);
 
         if (this._hasFrameworkTool)
-            this._frameworkToolDriver = fileExists(CROS_EC_PATH) ? 'cros_ec' : 'portio';
+            this._frameworkToolDriver = fileExists(CROS_EC_PATH) ? 'cros-ec' : 'portio';
 
         return true;
     }
```

But then, if I run the script manually:
```shell
$ pkexec /usr/local/bin/batteryhealthchargingctl-paul FRAMEWORK_TOOL_THRESHOLD_WRITE cros-ec 100
/usr/local/bin/batteryhealthchargingctl-paul: line 264: framework_tool: command not found
```

It cleans the `$PATH`, or maybe running it from the terminal is different?

maniacx, 2024-08-17:
Does the framework_tool work in terminal?

`framework_tool --driver portio --charge-limit 80
Minimum 0%, Maximum 80%`

Try cross-ec too

`framework_tool --driver cros-ec --charge-limit 80
Minimum 0%, Maximum 100%`

Then try running batteryhealthchargingctl-paul with both portio as well as cross-ec ?

Does seem like something with path?
Batteryhealthchargingctl is using ,`#!/usr/bin/env bash` but I don't think that should matter

paulvt, 2024-08-17:
It does, it does not contain `/usr/local`. I put `echo $PATH` at the start of `batteryhealthchargingctl-paul` and it outputs:
```
/usr/sbin:/usr/bin:/sbin:/bin:/root/bin
```
It seems to even skip `/etc/profile` which sets the path correctly....

paulvt, 2024-08-17:
(I see by the way that the Dell SMBIOS tool also has it's path hardcoded to `/usr/sbin/smbios-battery-ctl` which also could be in another location.)

JanaCoulomb, 2024-08-17:
Just tested the framework_tool branch. Turns out "/sys/devices/platform/framework_laptop" doesn't exist on all distros on frameworks. The result: The isAvailable returns false and that creates an error telling me that there are missing dependencies.

For me I found the workaround to just check if the vendor is "Framework" using 
`readFile('/sys/devices/virtual/dmi/id/sys_vendor').includes("Framework")` in the Framework.js file

paulvt, 2024-08-17:
> Just tested the framework_tool branch. Turns out "/sys/devices/platform/framework_laptop" doesn't exist on all distros 
> For me I found the workaround to just check if the vendor is "Framework" using `readFile('/sys/devices/virtual/dmi/id/sys_vendor').includes("Framework")` in the Framework.js file

But this is already in this branch?

maniacx, 2024-08-17:
> (I see by the way that the Dell SMBIOS tool also has it's path hardcoded to /usr/sbin/smbios-battery-ctl which also could be in another location.)

Yes. My initial method for dell was using hardcoded path. But that didnt work for system like NixOS. 
So I had to switch to program in Path and use hardcoded path as fallback.

if > `GLib.find_program_in_path('smbios-battery-ctl')`  then `smbios-battery-ctl'" --set-charging-mode='adaptive'`

else if > `fileExist(/usr/sbin/smbios-battery-ctl') ` then `/usr/sbin/smbios-battery-ctl'" --set-charging-mode='adaptive'`

---

But by default, when framework tool is packaged as aur for Arch Linux it is in /usr/bin/
https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=framework-system-git

> If I call the script with sudo, it shows the correct paths, so I'm guessing that pkexec does something different.

I am curious about this though?

May be add this to batteryhealthchargingctl?
```
source /etc/profile
```
and also
```
source ~/.bashrc
```

maniacx, 2024-08-17:
```
source ~/.bashrc
```
May not work as user might be root, so path to bashrc may be a problem. need more testing.

paulvt, 2024-08-17:
> ```
> source ~/.bashrc
> ```
> 
> May not work as user might be root, so path to bashrc may be a problem. need more testing.

My root user has `/root/.profile` which reads `/root/.bashrc`. But sourcing `/etc/profile` already fixes the issue for me.

maniacx, 2024-08-17:
Ok. I already managed to push my commit to main branch GNOME45 ( no laptop, using phone browser). May be one of you can push all the fixes for framework tools as discussed above.

paulvt, 2024-08-18:
> Ok. I already managed to push my commit to main branch GNOME45 ( no laptop, using phone browser). May be one of you can push all the fixes for framework tools as discussed above.

 I just created PR #103 based on the `GNOME45` branch that includes the suggested changes.
 
 I have tested it using `install.sh` and it works. The only issue is the one that I mentioned earlier, I see no UI changes whatsoever, so no the extension is constantly putting the charge limit to 100% and I want it set to 80%.

maniacx, 2024-08-18:
Oh. I assumed everything was working... My misunderstanding..

If GUI it is not changing seems like it is reading but not writing.
check logs
```
journalctl -f -o cat /usr/bin/gnome-shell
```

Check if you can read or write directly using batteryhealthchargingctl command.
```
pkexec /usr/local/bin/batteryhealthchargingctl-paul FRAMEWORK_TOOL_THRESHOLD_WRITE cros-ec 100
```

There must be something that I have missed.

paulvt, 2024-08-18:
Yes, I see the plugin executing commands in the journal:
```
Aug 18 12:12:06 frame pkexec[209706]: pam_unix(polkit-1:session): session opened for user root(uid=0) by paul(uid=1000)
Aug 18 12:12:06 frame pkexec[209706]: paul: Executing command [USER=root] [TTY=unknown] [CWD=/home/paul] [COMMAND=/usr/local/bin/batteryhealthchargingctl-paul FRAMEWORK_TOOL_THRESHOLD_WRITE cros-ec 100]
Aug 18 12:14:22 frame pkexec[210465]: pam_unix(polkit-1:session): session opened for user root(uid=0) by paul(uid=1000)
Aug 18 12:14:22 frame pkexec[210465]: paul: Executing command [USER=root] [TTY=unknown] [CWD=/home/paul] [COMMAND=/usr/local/bin/batteryhealthchargingctl-paul CHECKINSTALLATION /home/paul/.local/share/gnome-shell/extensions/Battery-Health-Charging@maniacx.github.com/resources paul]
```

I can execute this commands successfully also in the terminal. (I have to because I have to set it back to 80.)

There is nothing in the quick settings panel, nor notifications, nor a preference button, nor a system indicator.
I had a system indicator before when the command was still failing, but that's gone now that everything is OK.

maniacx, 2024-08-19:
Hey @paulvt 
Can you test the updated version on GNOME45 branch?

paulvt, 2024-08-19:
I removed the extension, updated the branch, ran `./install.sh`, logged out. Used the Extensions tool to enable the extension, logged in again. I see it run the `CHECKINSTALLTION` command via the helper, but no UI.

paulvt, 2024-08-19:
It is passing the tool as `ARG1`, this should be the driver...

maniacx, 2024-08-20:
Yes honestly. There was only one mistake in first commit that I pushed. I accidently deleted the case terminator before FRAMWORK_TOOL_THRESHOLD_READ . That cause all the problem.

I still have to do some changes/improvement, with batteryhealthchargingctl as currently, it is unsafe if we think security wise.

> The UI seems working now (it takes a while!) 

How long does it take? Do you think delay is from the framework_tool or the extension?

paulvt, 2024-08-20:
It is the extension, it takes about 4-6 minutes. When I open and the resume the laptop at 7:46, at 7:50 it does `CHECKINSTALLATION` and then at 7:52 it does the `READ` and show the UI (and the notification that I have set it it the setting I've set it at previously already). Otherwise it works fine.

Also when I select a setting from the quick panel, it is not written (or I am to inpatient). If I run the `WRITE` command with the helper, it picks it up and updates the UI (and shows the notification).

maniacx, 2024-08-20:
You are not impatient lol. Max expected delay would be 1 sec not 4 minutes. 

I just pushed a branch with logs added.
https://github.com/maniacx/Battery-Health-Charging/tree/debug-framework

Let see what is going on. Lets reset and uninstall everything.

uninstall this extension.
Download the branch

Reset all gsettings related to this extension
```
gsettings --schemadir /home/$USER/.local/share/gnome-shell/extensions/Battery-Health-Charging@maniacx.github.com/schemas reset-recursively org.gnome.shell.extensions.Battery-Health-Charging
```

In tool folder there is and installer.sh script. Run using root privileges to remove polkit rules

```
./installer.sh --tool-user paul uninstall
```

Install the debug-framework branch

Restart gnome-shell

open new terminal/console
Run below command that print more details including time stamps. I have piped it with awk to remove unneed stuff, you may have to play around with this command a bit
```
journalctl -f -o short-precise /usr/bin/gnome-shell --priority=debug | awk '{$4=""; $5=""; print $0}'
```

Leave this terminal window open.

Disable the extension and enable it. And see if you find something and/or post it here.
 You should see something like

```
Aug 20 09:43:07.672903   Battery Health Charging: StdErr: Error accessing /usr/local/bin/batteryhealthchargingctl-maniacx: No such file or directory
```

paulvt, 2024-08-20:
I get the notification:

> *Batter Health Charging Error*
> Please install polkit from extension settings under installation

and I see an indicator icon with a red marker on top.
The logs contain (but takes two minutes too appear):
```
Aug 20 21:21:36.630220   cogl_framebuffer_set_viewport: assertion 'width > 0 && height > 0' failed
Aug 20 21:21:36.651019   cogl_framebuffer_set_viewport: assertion 'width > 0 && height > 0' failed
Aug 20 21:21:36.700447   cogl_framebuffer_set_viewport: assertion 'width > 0 && height > 0' failed
Aug 20 21:21:37.275201   cogl_framebuffer_set_viewport: assertion 'width > 0 && height > 0' failed
Aug 20 21:21:38.633415   cogl_framebuffer_set_viewport: assertion 'width > 0 && height > 0' failed
Aug 20 21:21:38.668866   cogl_framebuffer_set_viewport: assertion 'width > 0 && height > 0' failed
Aug 20 21:23:47.163948   JS ERROR: TypeError: this.actor is null
_syncEnabled@resource:///org/gnome/shell/ui/windowManager.js:145:25    
onStopped@resource:///org/gnome/shell/ui/windowManager.js:157:35    
_makeEaseCallback/<@resource:///org/gnome/shell/ui/environment.js:65:22    
_easeActorProperty/<@resource:///org/gnome/shell/ui/environment.js:232:60    
_destroyWindowDone@resource:///org/gnome/shell/ui/windowManager.js:1607:21    
onStopped@resource:///org/gnome/shell/ui/windowManager.js:1575:39    
_makeEaseCallback/<@resource:///org/gnome/shell/ui/environment.js:65:22    
_easeActor/<@resource:///org/gnome/shell/ui/environment.js:154:64    
@resource:///org/gnome/shell/ui/init.js:21:20    
```

paulvt, 2024-08-20:
Ok. I've done that and enabled the extension (this time using the right branch :sweat_smile:).
I get the notification that I should install the polkit stuff via the settings and see the indicator icon with a red marker in the top right corner.

The logs say:
```
Aug 20 21:34:27.776492   Battery Health Charging: ---- Extension Enabled ----
Aug 20 21:34:27.785611   Battery Health Charging: ---- _getCurrentDevice ----
Aug 20 21:34:27.786098   Battery Health Charging: ---- pre find_program_in_path ----
Aug 20 21:34:27.786146   Battery Health Charging: ---- post find_program_in_path ----
Aug 20 21:34:27.786226   Battery Health Charging: ---- Pre _checkInstallation ----
```
and that's unfortunately it! (I waited until 21:45.)
With the `GNOME45` branch I do see that "error accessing" error, but not with this branch.

maniacx, 2024-08-20:
Yes. Install polkit rules from extension settings.

maniacx, 2024-08-20:
Lets use only the debug-framework branch currently for debugging. 
GNOME45 branch wont sshow any logs unless there is a syntax error in js files, or the batterychargingctl report an error.

After installing extension from debug-framework branch, did you install polkit rules using extension settings?

If you install it using extension settings, it will run check_compability function again, and proceed with checkInstallation.
If you install the polkit rules using installer.sh , you will have to enable and disable the extension once.

* For now install polkit rules from extension.
* Wait for 15 second, if it doesnt show anything in log and doesnt show a quicksettings panel.
* Disable and enable the extension again.
No need to wait for 5 minutes. it things are not moving forward within 5 seconds something is wrong.
Thanks for all the troubleshooting.

paulvt, 2024-08-21:
I installed the pollkit rules (and helper) from the settings, because nothing happened I disabled and enabled the setting, logs:
```
Aug 21 07:45:24.650701   Battery Health Charging: ---- _getCurrentDevice ----
Aug 21 07:45:24.651191   Battery Health Charging: ---- pre find_program_in_path ----
Aug 21 07:45:24.651254   Battery Health Charging: ---- post find_program_in_path ----
Aug 21 07:45:24.651346   Battery Health Charging: ---- Pre _checkInstallation ----
Aug 21 07:45:24.651537   Battery Health Charging: ---- argv = pkexec,/usr/local/bin/batteryhealthchargingctl-paul,CHECKINSTALLATION,/home/paul/.local/share/gnome-shell/extensions/Battery-Health-Charging@maniacx.github.com/resources,paul ----

Aug 21 07:46:04.474160   Battery Health Charging: ---- Extension Disable ----

Aug 21 07:46:07.457506   Battery Health Charging: ---- Extension Enabled ----
Aug 21 07:46:07.465453   Battery Health Charging: ---- _getCurrentDevice ----
Aug 21 07:46:07.465887   Battery Health Charging: ---- pre find_program_in_path ----
Aug 21 07:46:07.465975   Battery Health Charging: ---- post find_program_in_path ----
Aug 21 07:46:07.466045   Battery Health Charging: ---- Pre _checkInstallation ----
Aug 21 07:46:07.466150   Battery Health Charging: ---- argv = pkexec,/usr/local/bin/batteryhealthchargingctl-paul,CHECKINSTALLATION,/home/paul/.local/share/gnome-shell/extensions/Battery-Health-Charging@maniacx.github.com/resources,paul ----
```

maniacx, 2024-08-21:
Ok. Now we are getting somewhere.

Seem like CHECKINSTALLATION is taking long time for some reason.

---

* disable the extension.
* Run command
```
pkexec /usr/local/bin/batteryhealthchargingctl-paul CHECKINSTALLATION /home/paul/.local/share/gnome-shell/extensions/Battery-Health-Charging@maniacx.github.com/resources paul && echo $?
```
1. Check how much time it takes.
2.  Keep in mind, it should not as for password at this point. It should run pkexec without password. 

---

Next lets check the exit status as well by combining `echo $?` Run the command.
```
pkexec /usr/local/bin/batteryhealthchargingctl-paul CHECKINSTALLATION /home/paul/.local/share/gnome-shell/extensions/Battery-Health-Charging@maniacx.github.com/resources paul && echo $?
```
Also note the time here. For me I get the recieve results instantly.
Expected results.
```
Battery Health Charging: checking rules !
Battery Health Charging: checking ctl !
Battery Health Charging: installation is up to date 
0
```

---
What is your kernel, debian version, and polkit version.

```
echo "-- Kernel --"
uname -r
echo "-- Distro --"
lsb_release -a
echo "-- polkit  --"
pkexec --version
```

paulvt, 2024-08-21:
So, yes, the command returns instantly. I had already established this because for #102  I have been running commands with `pkexec` a lot.

```
Battery Health Charging: checking rules !
Battery Health Charging: checking ctl !
Battery Health Charging: installation is up to date 
0
```

The system information:
```
-- Kernel --
6.10.3-amd64
-- Distro --
No LSB modules are available.
Distributor ID:	Debian
Description:	Debian GNU/Linux trixie/sid
Release:	n/a
Codename:	trixie
-- polkit  --
pkexec version 125
```

Note that the delay is exactly 120s, so curious :thinking:

maniacx, 2024-08-21:
Oh didn't realize polkitt 125 was released. I wonder if they have changed something.

I will install fedora rawhide today and test. This will give me an opportunity to also test the gnome 47 as well. Hopefully fedora rawhide use polkitt 125.

lmigfernandes, 2024-08-22:
Off-topic. I use Linux on Chromebooks. We flash Coreboot to run Linux. I use the "ectool for the Framework laptop" on Arch Linux. Seems the same, in my case "sudo ectool chargecontrol normal 75 80". Can we cover this or open a new issue?

maniacx, 2024-08-22:
@lmigfernandes 
Yes. Please open a new issue
Add some information.

exact command to read and write charging threshold and the output those command.

Also explanation of `normal` in the command

output of 
```
cat /sys/devices/virtual/dmi/id/sys_vendor
```

Location of where ectool is installed.

maniacx, 2024-08-22:
@paulvt 

I did run Fedora rawhide.
The extension runs well. No delays. It is using Gnome47 and polkit 125

I use a batch script to mimic the output of framework tool.
There is no delay
```
Battery Health Charging: ---- Extension Enabled ----
Battery Health Charging: ---- _getCurrentDevice ----
Battery Health Charging: ---- pre find_program_in_path ----
Battery Health Charging: ---- post find_program_in_path ----
Battery Health Charging: ---- Pre _checkInstallation ----
Battery Health Charging: ---- argv = pkexec,/usr/local/bin/batteryhealthchargingctl-maniacx,CHECKINSTALLATION,/home/maniacx/.local/share/gnome-shell/extensions/Battery-Health-Charging@maniacx.github.com/resources,maniacx ----
Battery Health Charging: ---- status = 0 ----
Battery Health Charging: ---- stdout = Battery Health Charging: checking rules !
Battery Health Charging: checking ctl !
Battery Health Charging: installation is up to date 
 ----
Battery Health Charging: ---- stderr =  ----
Battery Health Charging: ---- Post _checkInstallation ----
Battery Health Charging: ---- setThresholdLimit ----
Battery Health Charging: ---- _setThresholdFrameworkTool ----
Battery Health Charging: ---- argv = pkexec,/usr/local/bin/batteryhealthchargingctl-maniacx,FRAMEWORK_TOOL_THRESHOLD_READ,/usr/bin/framework_tool,portio ----
Battery Health Charging: ---- status = 0 ----
Battery Health Charging: ---- stdout = Minimum 0%, Maximum 100%
 ----
Battery Health Charging: ---- stderr =  ----
Battery Health Charging: ---- Start threshold panel ----
Battery Health Charging: ---- setThresholdLimit ----
Battery Health Charging: ---- _setThresholdFrameworkTool ----
Battery Health Charging: ---- argv = pkexec,/usr/local/bin/batteryhealthchargingctl-maniacx,FRAMEWORK_TOOL_THRESHOLD_READ,/usr/bin/framework_tool,portio ----
Battery Health Charging: ---- status = 0 ----
Battery Health Charging: ---- stdout = Minimum 0%, Maximum 100%
 ----
Battery Health Charging: ---- stderr =  ----
Battery Health Charging: ---- argv = pkexec,/usr/local/bin/batteryhealthchargingctl-maniacx,FRAMEWORK_TOOL_THRESHOLD_WRITE,/usr/bin/framework_tool,portio,80 ----
Battery Health Charging: ---- status = 0 ----
Battery Health Charging: ---- stdout = Minimum 0%, Maximum 80%
 ----
Battery Health Charging: ---- stderr =  ----
```

Makes me wonder what is the issue.

Let try yo run this in gjs-console and see what happens.
[test.zip](https://github.com/user-attachments/files/16714731/test.zip)

I have add a test.zip. exttract and run `gjs-console test.js`
See if you get any output or if there is any delay.

paulvt, 2024-08-22:
I still am in GNOME Shell 46.4.

The test program runs fine and instantaneous:
```
Gjs-Message: 21:56:24.702: JS LOG: arg = pkexec,/usr/local/bin/batteryhealthchargingctl-paul,CHECKINSTALLATION,/home/paul/.local/share/gnome-shell/extensions/Battery-Health-Charging@maniacx.github.com/resources,paul
Gjs-Message: 21:56:24.704: JS LOG: proc.init successful
Gjs-Message: 21:56:24.730: JS LOG: pre communicate_utf8_finish
Gjs-Message: 21:56:24.730: JS LOG: post communicate_utf8_finish
Gjs-Message: 21:56:24.730: JS LOG: post get_exit_status
Gjs-Message: 21:56:24.730: JS LOG: Promise stdout = Battery Health Charging: checking rules !
Battery Health Charging: checking ctl !
Battery Health Charging: installation is up to date 

Gjs-Message: 21:56:24.730: JS LOG: Promise stderr = 
Gjs-Message: 21:56:24.730: JS LOG: Promise status = 0
Gjs-Message: 21:56:24.730: JS LOG: return output
Gjs-Message: 21:56:24.730: JS LOG: status = 0
Gjs-Message: 21:56:24.730: JS LOG: stdout = Battery Health Charging: checking rules !
Battery Health Charging: checking ctl !
Battery Health Charging: installation is up to date 

Gjs-Message: 21:56:24.730: JS LOG: Main function completed
```

maniacx, 2024-08-23:
So, I fired up debian in Gnome-Boxes. use my dummy framework_tool.
Installed extension by modifying DMI vendor from Framework to QEMU.
Enabled the extension and I worked without any delay. 

```
 23 15:27:51.830898 debian gnome-shell[3691]: Battery Health Charging: ---- Extension Enabled ----
Aug 23 15:27:51.833059 debian gnome-shell[3691]: Battery Health Charging: ---- _getCurrentDevice ----
Aug 23 15:27:51.833603 debian gnome-shell[3691]: Battery Health Charging: ---- pre find_program_in_path ----
Aug 23 15:27:51.833682 debian gnome-shell[3691]: Battery Health Charging: ---- post find_program_in_path ----
Aug 23 15:27:51.833912 debian gnome-shell[3691]: Battery Health Charging: ---- Pre _checkInstallation ----
Aug 23 15:27:51.834074 debian gnome-shell[3691]: Battery Health Charging: ---- argv = pkexec,/usr/local/bin/batteryhealthchargingctl-maniacx,CHECKINSTALLATION,/home/maniacx/.local/share/gnome-shell/extensions/Battery-Health-Charging@maniacx.github.com/resources,maniacx ----
Aug 23 15:27:51.874431 debian gnome-shell[3691]: Battery Health Charging: ---- status = 0 ----
Aug 23 15:27:51.874514 debian gnome-shell[3691]: Battery Health Charging: ---- stdout = Battery Health Charging: checking rules !
                                                 Battery Health Charging: checking ctl !
                                                 Battery Health Charging: installation is up to date 
                                                  ----
Aug 23 15:27:51.874573 debian gnome-shell[3691]: Battery Health Charging: ---- stderr =  ----
Aug 23 15:27:51.874894 debian gnome-shell[3691]: Battery Health Charging: ---- Post _checkInstallation ----
Aug 23 15:27:51.875042 debian gnome-shell[3691]: Battery Health Charging: ---- setThresholdLimit ----
Aug 23 15:27:51.875177 debian gnome-shell[3691]: Battery Health Charging: ---- _setThresholdFrameworkTool ----
Aug 23 15:27:51.875496 debian gnome-shell[3691]: Battery Health Charging: ---- argv = pkexec,/usr/local/bin/batteryhealthchargingctl-maniacx,FRAMEWORK_TOOL_THRESHOLD_READ,/usr/bin/framework_tool,portio ----
Aug 23 15:27:51.909776 debian gnome-shell[3691]: Battery Health Charging: ---- status = 0 ----
Aug 23 15:27:51.909904 debian gnome-shell[3691]: Battery Health Charging: ---- stdout = Minimum 0%, Maximum 80%
                                                  ----
Aug 23 15:27:51.910002 debian gnome-shell[3691]: Battery Health Charging: ---- stderr =  ----
Aug 23 15:27:51.910892 debian gnome-shell[3691]: Battery Health Charging: ---- Start threshold panel ----
Aug 23 15:27:55.947495 debian gnome-shell[3691]: Battery Health Charging: ---- setThresholdLimit ----
Aug 23 15:27:55.947648 debian gnome-shell[3691]: Battery Health Charging: ---- _setThresholdFrameworkTool ----
Aug 23 15:27:55.947828 debian gnome-shell[3691]: Battery Health Charging: ---- argv = pkexec,/usr/local/bin/batteryhealthchargingctl-maniacx,FRAMEWORK_TOOL_THRESHOLD_READ,/usr/bin/framework_tool,portio ----
Aug 23 15:27:55.976733 debian gnome-shell[3691]: Battery Health Charging: ---- status = 0 ----
Aug 23 15:27:55.976816 debian gnome-shell[3691]: Battery Health Charging: ---- stdout = Minimum 0%, Maximum 80%
                                                  ----
Aug 23 15:27:55.976882 debian gnome-shell[3691]: Battery Health Charging: ---- stderr =  ----
Aug 23 15:27:55.977455 debian gnome-shell[3691]: Battery Health Charging: ---- argv = pkexec,/usr/local/bin/batteryhealthchargingctl-maniacx,FRAMEWORK_TOOL_THRESHOLD_WRITE,/usr/bin/framework_tool,portio,60 ----
Aug 23 15:27:56.003116 debian gnome-shell[3691]: Battery Health Charging: ---- status = 0 ----
Aug 23 15:27:56.003239 debian gnome-shell[3691]: Battery Health Charging: ---- stdout = Minimum 0%, Maximum 60%
                                                  ----
Aug 23 15:27:56.003372 debian gnome-shell[3691]: Battery Health Charging: ---- stderr =  ----
Aug 23 15:27:57.407149 debian gnome-shell[3691]: Battery Health Charging: ---- setThresholdLimit ----
Aug 23 15:27:57.407255 debian gnome-shell[3691]: Battery Health Charging: ---- _setThresholdFrameworkTool ----
Aug 23 15:27:57.407521 debian gnome-shell[3691]: Battery Health Charging: ---- argv = pkexec,/usr/local/bin/batteryhealthchargingctl-maniacx,FRAMEWORK_TOOL_THRESHOLD_READ,/usr/bin/framework_tool,portio ----
Aug 23 15:27:57.438319 debian gnome-shell[3691]: Battery Health Charging: ---- status = 0 ----
Aug 23 15:27:57.438403 debian gnome-shell[3691]: Battery Health Charging: ---- stdout = Minimum 0%, Maximum 60%
                                                  ----
Aug 23 15:27:57.438452 debian gnome-shell[3691]: Battery Health Charging: ---- stderr =  ----
Aug 23 15:27:57.438657 debian gnome-shell[3691]: Battery Health Charging: ---- argv = pkexec,/usr/local/bin/batteryhealthchargingctl-maniacx,FRAMEWORK_TOOL_THRESHOLD_WRITE,/usr/bin/framework_tool,portio,100 ----
Aug 23 15:27:57.464352 debian gnome-shell[3691]: Battery Health Charging: ---- status = 0 ----
Aug 23 15:27:57.464423 debian gnome-shell[3691]: Battery Health Charging: ---- stdout = Minimum 0%, Maximum 100%
                                                  ----
Aug 23 15:27:57.464475 debian gnome-shell[3691]: Battery Health Charging: ---- stderr =  ----
Aug 23 15:27:58.860774 debian gnome-shell[3691]: Battery Health Charging: ---- setThresholdLimit ----
Aug 23 15:27:58.860878 debian gnome-shell[3691]: Battery Health Charging: ---- _setThresholdFrameworkTool ----
[... cut ...]
XDG_DATA_DIRS=/usr/share/gnome:/usr/local/share/:/usr/share/
XDG_MENU_PREFIX=gnome-
XDG_SESSION_CLASS=user
XDG_SESSION_DESKTOP=gnome-xorg
XDG_SESSION_TYPE=x11
XMODIFIERS=@im=ibus
MANAGERPID=1310
INVOCATION_ID=e4de5c9246be4bd6a68f42b4e39959f0
JOURNAL_STREAM=9:3980
SYSTEMD_EXEC_PID=1583
MEMORY_PRESSURE_WATCH=/sys/fs/cgroup/user.slice/user-1000.slice/user@1000.service/session.slice/org.gnome.Shell@x11.service/memory.pressure
MEMORY_PRESSURE_WRITE=c29tZSAyMDAwMDAgMjAwMDAwMAA=
GJS_DEBUG_OUTPUT=stderr
GJS_DEBUG_TOPICS=JS ERROR;JS LOG

```

athomas-ccc, 2024-08-27:
My Framework 16 laptop is running GNOME 46 and Fedora 40. When I install the GNOME extension, I'm getting a "Missing dependencies" error. I have  `framework_tool` in my path, in `~/.local/bin/framework_tool`. @maniacx given that #102 and #103 have been merged, is there still more to be done before `framework_tool` is supported as a dependency? My apologies if this has already been covered somewhere above in this thread, or in a different place.

maniacx, 2024-08-27:
@athomas-ccc 

Could you do some test for me.
Copy all these command and run them in terminal and post the output here so that I know what system/ kernel and other information you are using.

```
echo "-- Kernel --"
uname -r
echo "-- Distro --"
lsb_release -a
echo "-- polkit  --"
pkexec --version
echo "-- dmi sys vendor--"
cat /sys/devices/virtual/dmi/id/sys_vendor
echo "-- PATH --"
echo $PATH
```
Install extension from this branch. (You may have to install gettext)

https://github.com/maniacx/Battery-Health-Charging/tree/debug-framework

Open terminal and run the below command and keep it open. This terminal will log info/error
```
journalctl -f -o short-precise /usr/bin/gnome-shell --priority=debug
```

Disable and Enable the extension.
Check if it the threshold changes. If it does, also report if it change instantly or take more than 5 seconds to change the threshold.

Post the output of log.

athomas-ccc, 2024-08-28:
@maniacx  here is the first set of output you requested:
```
$ echo "-- Kernel --"
uname -r
-- Kernel --
6.10.5-200.fc40.x86_64
$ echo "-- Distro --"
lsb_release -a
-- Distro --
LSB Version:	n/a
Distributor ID:	Fedora
Description:	Fedora Linux 40 (Workstation Edition)
Release:	40
Codename:	n/a
$ echo "-- polkit  --"
pkexec --version
-- polkit  --
pkexec version 124
$ echo "-- dmi sys vendor--"
cat /sys/devices/virtual/dmi/id/sys_vendor
-- dmi sys vendor--
Framework
$ echo "-- PATH --"
echo $PATH
-- PATH --
/home/<username>/.local/bin:/home/<username>/bin:/usr/local/bin:/usr/local/sbin:/usr/bin:/usr/sbin
```

athomas-ccc, 2024-08-29:
>Install extension from this branch. (You may have to install gettext)
https://github.com/maniacx/Battery-Health-Charging/tree/debug-framework

@maniacx to confirm, do you mean I should install it using the `install.sh` script from that branch?

maniacx, 2024-08-29:
Yes. To do this here are the steps..

1. Install gettext.
```
sudo dnf install gettext
```

2. Download the branch. Open webpage and got to `code` (green button) > `Download as zip`. This should download the file in Downloads folder

3. Right click on zip file and `Extract.` You will find another folder Created with the same name as the file.

4..Right click on file `install.sh` > `Properties` > Switch ON `Execute as program`.

5. Right click on file `Run as a Program`

6.  Logout and Login again. (Install extension wont show up until we do not restart Gnome shell so need to logout and login)

7. Now Test the installation.
If extension works well,it will ask you to install/update polkit rules. Go ahead and install it from extension preferences.
After installing, check if threshold changes. Also test if threshold changes instantly.

8. Open terminal and run the below command and keep it open. This terminal will log info/error
```
journalctl -f -o short-precise /usr/bin/gnome-shell --priority=debug
```

Disable and Enable the extension. change threshold and post the log.

paulvt, 2024-09-02:
A small update from my side. I've asked a friend who has done a really recent Debian trixie install on his new Framework. Same problem there unfortunately.

Even worse, we have noticed that `pkexec` is not only delaying the execution for 120 seconds, it's running at 100% CPU during those 120 seconds. `strace` reveals it is doing continuous `fcntl` calls on a bad file descriptor 😕...

I have replaced the calls to `pkexec` by `sudo` and whitelisted the helper and that works and has shown me what the experience can be :wink:. I will try to look depper into it what is going wrong when the extension calls the `pkexec` commands. (I have tried letting GNOME Shell also run the command via Alt+F2, that is also fine.)

maniacx, 2024-09-03:
Strange. Hopefully you find something, cause I am lost.

maniacx, 2024-09-05:
@paulvt  @athomas-ccc 

I have made updated changes in the GNOME45 branch.
So if you have any patch to be applied, at the updated GNOME45 branch

morph027, 2024-09-05:
Got a new Framework 13 w/ 7840U. Just built `framework_tool`, put it into `/usr/bin`, removed the extension and installed from GNOME45 branch.

First message was something timed out during check. Got the command from the journal.

```bash
sudo /usr/local/bin/batteryhealthchargingctl-$USER CHECKINSTALLATION /home/$USER/.local/share/gnome-shell/extensions/Battery-Health-Charging@maniacx.github.com/resources $USER
Battery Health Charging: checking rules !
Battery Health Charging: installation needs updating rules!
Battery Health Charging: installation needs updating ctl!
```

So i removed the privileges from the extension setting, installed again and it just works.

:rocket:

maniacx, 2024-09-07:
@morph027 
What is your system?
Can  give me the output of the following.
```
echo "-- Kernel --"
uname -r
echo "-- Distro --"
lsb_release -a
echo "-- polkit  --"
pkexec --version
echo "-- gnome-shell --"
gnome-shell --version
```

maniacx, 2024-09-07:
@paulvt 

Did you find anything?
Can you try the latest update on GNOME45 please?
I have added notifications incase the extension takes more than 3 seconds.
You can change timeout value as well
`GLib.timeout_add_seconds(GLib.PRIORITY_DEFAULT, 3`

If it still timeouts.

install this extension.

[test-pkexec.zip](https://github.com/user-attachments/files/16914051/test-pkexec.zip)

There is no GUI.
When enabled, it will directly run execCheck function and run CHECK_INSTALLATION command from batteryhealthchargingctl .
CHECK_INSTALLATION command by compareinstalled polkit rules and resource folder in Battery-health-charging extension and it will print logs the output if successfull or timeout or errored.

It will be easier to debug.

[7 more replies]

closed 2024-12-21 by maniacx