maniacx/Battery-Health-Charging #104: Feature Request: Support Chromebooks running coreboot to use Linux "MrChromebox firmware"

lmigfernandes, 2024-08-22:
Would be nice to have EC Chromebook “Adaptive Charging” since newer Chromebooks that use the UEFI custom firmware to run other OS use coreboot. I talk about Arch Linux. I can try another distros if necessary.
On Arch Linux, I use AUR "ectool for the Framework laptop” or the official repo source [here](https://gitlab.howett.net/DHowett/ectool).
To write I use “sudo ectool chargecontrol normal 75 80”

I find the Normal command of EC on official Chromium OS documentation https://chromium.googlesource.com/chromiumos/platform2/+/HEAD/power_manager/docs/adaptive_charging.md and found additional information [here](https://github.com/MrChromebox/firmware/issues/420#issuecomment-1827493960).

To read information about battery status:
`[arc@arcbook ~]$ whereis ectool 
ectool: /usr/bin/ectool
[arc@arcbook ~]$ sudo ectool chargecontrol
Charge mode = NORMAL (0)
Battery sustainer = on (75% ~ 80%)
[arc@arcbook ~]$ cat /sys/devices/virtual/dmi/id/sys_vendor
Google`

@maniacx Tried my best to help and describe what u asked me, since English is not my primary language.

Thanks :)

lmigfernandes, 2024-08-22:
Additional information for installing EC tool on Fedora. https://docs.chrultrabook.com/docs/installing/ectool.html

lmigfernandes, 2024-08-28:
"ectool” is a standalone file, can run like a AppImage.

`arc@arcbook:~/.local/bin> file ectool 
ectool: ELF 64-bit LSB executable, x86-64, version 1 (SYSV), dynamically linked, interpreter /lib64/ld-linux-x86-64.so.2, BuildID[sha1]=17f3683fce2eb6efc3b4cbc3484b20991a8b0630, for GNU/Linux 3.2.0, stripped`

maniacx, 2024-08-29:
Thank you @lmigfernandes 

I will try to implement this end of this week.

I just want to confirm some information again. Give me the output of these 2 command.

```
sudo ectool chargecontrol normal 75 80
```

```
sudo ectool chargestate show
```

Post output on a new line. 
Example. Do not post like this

`[arc@arcbook ~]$ whereis ectool  ectool: /usr/bin/ectool [arc@arcbook ~]$ sudo ectool chargecontrol Charge mode = NORMAL (0) Battery sustainer = on (75% ~ 80%) [arc@arcbook ~]$ cat /sys/devices/virtual/dmi/id/sys_vendor Google`

Post like this
```
[arc@arcbook ~]$ whereis ectool 

 ectool: /usr/bin/ectool
```

It helps me get the exact string output by the command.

I understand translation is difficult, and I apologise asking for too much details.

maniacx, 2024-09-03:
@lmigfernandes 

I am working on the extension.

It was difficult for me to find the source code, and I finally found here.

https://github.com/coreboot/chrome-ec/blob/main/util/ectool.cc

But I am still not sure if you are using the same one.

Kindly tell me which website did you download it from?

Also I want to know the range of acceptable threshold values , command  and exit codes, so I have created a script that contains 9 command that will change your threshold every 1 seconds.

download and extract the zip.
[ectooltest.zip](https://github.com/user-attachments/files/16841933/ectooltest.zip)

open terminal.

navigate to download folder which contain the bash script.

change script to executable
```
chmod +x ectooltest.sh
```

and run the script
```
sudo ./ectooltest.sh
```

Allow the script to complete. 

Post the output here.

lmigfernandes, 2024-09-04:
@maniacx The ectool blob is specific for Chromebooks. Normally I compile from source on Arch Linux AUR but can be found the .rpm on fedora Terra repository [here](https://fedora.pkgs.org/40/terra/chromium-ectool-0:55680f7-14526.B.fc40.x86_64.rpm.html) (normally I download the .rpm and extract installing dependencies).

The output from script u asked:

```
Running: ectool chargecontrol normal 75 80
Output: [Charge state machine is in normal mode with sustainer enabled.]
Exit code: [0]

Running: ectool chargecontrol
Output: [Charge mode = DISCHARGE (2)
Battery sustainer = on (75% ~ 80%)]
Exit code: [0]

Running: ectool chargecontrol normal 95 100
Output: [Charge state machine is in normal mode with sustainer enabled.]
Exit code: [0]

Running: ectool chargecontrol
Output: [Charge mode = NORMAL (0)
Battery sustainer = on (95% ~ 100%)]
Exit code: [0]

Running: ectool chargecontrol normal 98 100
Output: [Charge state machine is in normal mode with sustainer enabled.]
Exit code: [0]

Running: ectool chargecontrol
Output: [Charge mode = NORMAL (0)
Battery sustainer = on (98% ~ 100%)]
Exit code: [0]

Running: ectool chargecontrol normal 55 60
Output: [Charge state machine is in normal mode with sustainer enabled.]
Exit code: [0]

Running: ectool chargecontrol
Output: [Charge mode = DISCHARGE (2)
Battery sustainer = on (55% ~ 60%)]
Exit code: [0]

Running: ectool chargecontrol normal 50 55
Output: [Charge state machine is in normal mode with sustainer enabled.]
Exit code: [0]

Running: ectool chargecontrol
Output: [Charge mode = DISCHARGE (2)
Battery sustainer = on (50% ~ 55%)]
Exit code: [0]

Running: ectool chargecontrol normal 40 50
Output: [Charge state machine is in normal mode with sustainer enabled.]
Exit code: [0]

Running: ectool chargecontrol
Output: [Charge mode = DISCHARGE (2)
Battery sustainer = on (40% ~ 50%)]
Exit code: [0]

Running: ectool chargecontrol normal 40 42
Output: [Charge state machine is in normal mode with sustainer enabled.]
Exit code: [0]

Running: ectool chargecontrol
Output: [Charge mode = DISCHARGE (2)
Battery sustainer = on (40% ~ 42%)]
Exit code: [0]

Running: ectool chargecontrol normal -1 -1
Output: [ectool: invalid option -- '1'
ectool: invalid option -- '1'
Usage: ectool [--dev=n] [--interface=dev|i2c|lpc] [--i2c_bus=n] [--device=vid:pid] [--name=cros_ec|cros_fp|cros_pd|cros_scp|cros_ish] [--ascii] <command> [params]

  --i2c_bus=n  Specifies the number of an I2C bus to use. For
               example, to use /dev/i2c-7, pass --i2c_bus=7.
               Implies --interface=i2c.

  --interface Specifies the interface.

  --device    Specifies USB endpoint by vendor ID and product
              ID (e.g. 18d1:5022).

Use 'ectool help' to print a list of commands.]
Exit code: [1]

Running: ectool chargecontrol
Output: [Charge mode = DISCHARGE (2)
Battery sustainer = on (40% ~ 42%)]
Exit code: [0]

Running: ectool chargecontrol normal 75 80
Output: [Charge state machine is in normal mode with sustainer enabled.]
Exit code: [0]

Running: ectool chargecontrol
Output: [Charge mode = DISCHARGE (2)
Battery sustainer = on (75% ~ 80%)]
Exit code: [0]

```
Thank you.

maniacx, 2024-09-05:
@lmigfernandes 

You can try the  GNOME45 branch and let me if there are some issues.

https://github.com/maniacx/Battery-Health-Charging/commits/GNOME45/

lmigfernandes, 2024-09-22:
Hello, @maniacx
How [this](https://github.com/MrChromebox/firmware/issues/420#issuecomment-2308148375) Linux kernel patch will change the Chromebook ectool blob method.
Keep in mind, old Linux kernel's distributions like Debian still need to wait 1/2 years to get 6.11 release.
As soon Arch gets 6.11 I'm open to any additional help.

Thank you.

maniacx, 2024-09-23:
Yes, I saw the patch when I was finding out about ectools.
sysfs is already supported.

https://github.com/maniacx/Battery-Health-Charging/blob/GNOME45/devices/Chromebook.js#L9-L12

If ectool is installed and it runs older kernel, extension will use the ectool.
If ectool is not installed and it runs newer kernel to support charging threshold(sysfs) , it will use sysfs.

If running ectool and new kernel, you will get an option to choose what for changing threshold.

[image]

Althought I didnt add feature for charge_behavior.
I think it is something that never be used or will be used once in a while, for battery calibration, if battery shows wrong value. In that case user will have to calibrate the battery using command line, but it is quite rare that happens.

lmigfernandes, 2024-10-01:
Hey @maniacx ,

Looks like sysfs mode is working. I updated my Coreboot firmware and Linux Kernel.
Thank you.

maniacx, 2024-10-01:
Thank you for the update.
I just want fix some minor issues, and release this version, but too busy.
Give me two week, hopefully once the extension is finalize, I will request you to test again and release it on Gnome extensions.

lmigfernandes, 2025-01-06:
Hey @maniacx,

On all branches, I'm receiving this notification “Missing dependencies”. I don't know if this is affecting only the Chromebook integration.
``` 
[luis@arcbook ~]$ uname -r
6.12.8-arch1-1
```
Thank you,
Luis

maniacx, 2025-01-06:
On your laptop check if you  have sysfs path
Or post output of these commands.

```
ls - l /sys/class/power_supply/BAT0/charge_control_end_threshold

ls -l /sys/class/power_supply/BAT0/charge_control_start_threshold

cat /sys/devices/virtual/dmi/id/sys_vendor

```

lmigfernandes, 2025-01-06:
For some reason, I don't have charge_control_start_threshold

```
[root@arcbook ~]# ls -l /sys/class/power_supply/BAT0/charge_control_end_threshold
-rw-r--r-- 1 root root 4096 jan  6 19:35 /sys/class/power_supply/BAT0/charge_control_end_threshold
[root@arcbook ~]# ls -l /sys/class/power_supply/BAT0/charge_control_start_threshold
ls: cannot access '/sys/class/power_supply/BAT0/charge_control_start_threshold': No such file or directory
[root@arcbook ~]# cat /sys/devices/virtual/dmi/id/sys_vendor
Google
[root@arcbook ~]# 
```
Thank you, @maniacx

maniacx, 2025-01-06:
For some reason, you do not have start_threshold. That is the reason why it is asking for dependencies..

```
ls: cannot access '/sys/class/power_supply/BAT0/charge_control_start_threshold
```

Did you compile the kernel manually?
Or is it the default kernel that comes with MrChromebox firmware?

maniacx, 2025-01-06:
I have posted a question here
https://github.com/MrChromebox/firmware/issues/420#issuecomment-2574023038

Seems like there was some recent commit changes regarding cross_ec_charge.c that removes /hides charge_control_start_threshold

maniacx, 2025-01-09:
@lmigfernandes 

I have pushed new commit but I havent tested them.

This should show apply only end threshold for laptop using v2 command which I believe you laptop uses. 

Can you test them for me?

If you install ec-tools it will use ectools (with no option for sysfs)
But if un uninstall ectools, it should use sysfs with only end threshold.

maniacx, 2025-01-09:
Hello @lmigfernandes 
I have pushed new commit but I havent tested them.

This should show apply only end threshold for laptop using v2 command which I believe you laptop uses. 

Can you test them for me?

If you install ectool it will use ectools (with no option for sysfs)
But if you uninstall ectool, it should use sysfs with only end threshold.

maniacx, 2025-01-09:
test

lmigfernandes, 2025-01-09:
Hello @maniacx

I uninstalled ectools and all 3 default modes are working as expected.

Thank you,
Luis

closed 2025-01-25 by maniacx