maniacx/Battery-Health-Charging #105: Change Battery indication icon behavior & threshold not working

maniacx, 2024-09-11:
@MaximalCats

Can you give some details.

1. Make model of your laptop.
2. Distro name (example Ubuntu 24.04)
3. Kernel version
4. Gnome shell version

MaximalCats, 2024-09-12:
**Model:** ThinkPad P50
**Distro:** Void Linux
**Kernel:** 6.6.45_1
**GS version:** 46.2

maniacx, 2024-09-12:
Is this dual battery or single battery?

First let see what the problem with threshold is.

```
echo '--- polkit version ---'
pkexec --version
echo '--- writing threshold1---'
echo '55' | sudo tee /sys/class/power_supply/BAT0/charge_control_start_threshold
echo '60' | sudo tee /sys/class/power_supply/BAT0/charge_control_end_threshold
echo '--- reading threshold1---'
cat /sys/class/power_supply/BAT0/charge_control_end_threshold
cat /sys/class/power_supply/BAT0/charge_control_start_threshold
echo '--- writing threshold2---'
echo '100' | sudo tee /sys/class/power_supply/BAT0/charge_control_end_threshold
echo '95' | sudo tee /sys/class/power_supply/BAT0/charge_control_start_threshold
echo '--- reading threshold2---'
cat /sys/class/power_supply/BAT0/charge_control_end_threshold
cat /sys/class/power_supply/BAT0/charge_control_start_threshold
echo '--- writing threshold3---'
echo '75' | sudo tee /sys/class/power_supply/BAT0/charge_control_start_threshold
echo '80' | sudo tee /sys/class/power_supply/BAT0/charge_control_end_threshold
echo '--- reading threshold3---'
cat /sys/class/power_supply/BAT0/charge_control_end_threshold
cat /sys/class/power_supply/BAT0/charge_control_start_threshold
```

You might have to change your BAT0 to BAT1 is name assigned to the battery is BAT1
To find out use command
```
ls -l  /sys/class/power_supply/
```

MaximalCats, 2024-09-20:
Sorry; battery threshold seems to be working fine now. Icon is still behaving incorrectly.

maniacx, 2024-09-23:
Which icon are you talking about?

1 Extensions Threshold Icon
[image]

Or

2. Gnome system tray Battery level icon
[image]

Explain how is it behaving incorrectly?
And explain what is expected behavior?

maniacx, 2024-09-23:
Check this settings.
[image]

---

**With this settings disabled, it will use default Gnome behaviour.**
This extension doesnt control the icon, The icon is controlled by Gnome-shell

When battery is discharging, icon displayed is [image]

When battery is in threshold (reported as pending-charge in upower), icon displayed is [image]

When Battery is charging , icon displayed is [image]

---

**With this settings enabled, it will behaviour similar to most oem softwares in Windows.**
The icon is controlled by the extension.

When battery is discharging, icon displayed is [image]

When battery is in threshold (reported as pending-charge in upower), icon displayed is   [image]

When Battery is charging , icon displayed is [image]

---
To summarize, Gnome behavior, the icon reports more about battery status, whether charging, discharging or pending charge due to threshold.(while also reporting the battery level)

Well as the Windows,  the icon reports more about if charger is plugged-in or not (while also reporting the battery level)

Let me know if you are seeing a different other than what I have described above.

maniacx, 2024-10-16:
Closing, due to lack of feedback.

closed 2024-10-16 by maniacx