maniacx/Battery-Health-Charging #110: Different options on very similar machines

ViBE-HU, 2024-10-05:
maybe it is not an issue but would be good to clarify.

i have a Lenovo Legion 5 (AMD) machine and a Lenovo ThinkPad T14s Gen1 (Intel) a machine. both are from the same era. so i guess they shares platform. but somehow on the Legion machine i see less settings and options. are these machines really differs or it is a bug? the Legion machines have only the general tab and the ThinkPad have all tabs.

both machines running ubuntu 24.04 and everything updated.

maniacx, 2024-10-06:
No it's not a bug.

Lenovo ideapad, Legion series you can only enable or disable the charging limiter. Is it fixed threshold that is either 60 or 80%  set by the manufacturer and not cannot  customized. Hence there are settings available for Lenovo are according to ideapad / legion series

Lenovo thinkpad however, the threshold value can be customized. So it has additional settings tabs to customize the threshold

I have designed the extension in way that settings page will show settings available according to the laptop make/model.

ViBE-HU, 2024-10-06:
good to know. i thought that since the platform and all the features are shared the charging should be the same. but looks like it is dedicated to thinkpads.

closed 2024-10-06 by ViBE-HU