maniacx/Battery-Health-Charging #111: Battery Health Charging is taking too long to start on Ubuntu 24.10

danielbmeireles, 2024-10-11:
Description:

I’ve noticed that on Ubuntu 24.10, the Battery Health Charging feature is taking significantly longer to start. Previously, the feature would activate within a few seconds, but now it’s delayed by several minutes (I realize that because the quick settings button is also taking a long time to appear).

Environment:

- Ubuntu version: 24.10
- BHC version: 65

maniacx, 2024-10-11:
I will take a look at it.
Just curious which laptop do you have?

The delayed start, is something that another debian user was facing. Still couldn't find out what is the issue. Seems like something to do with polkit 

Did you make any changes in regard to sudo or polkit? Such as adding wheel or user to sudo etc?

maniacx, 2024-10-11:
Just for context look at the discussion at the here
https://github.com/maniacx/Battery-Health-Charging/issues/100

danielbmeireles, 2024-10-11:
> I will take a look at it. Just curious which laptop do you have?
>
I have a Lenovo T14. I can provide you with more info if you need it.
> 
> The delayed start, is something that another debian user was facing. Still couldn't find out what is the issue. Seems like something to do with polkit
> 
> Did you make any changes in regard to sudo or polkit? Such as adding wheel or user to sudo etc?
>
No, I upgraded my system to the newest version of Ubuntu, that's all.

maniacx, 2024-10-11:
I just installed ubuntu. I am having the same issue. 
One thing I found that running in X11 has no delays.
Running in Wayland causes the issue. 
I have to debug further, and check what has changed recently.

danielbmeireles, 2024-10-14:
I am also having issues to launch Warp terminal and seems the issue is related to Wayland as well: https://github.com/warpdotdev/Warp/issues/5367

maniacx, 2024-10-14:
@danielbmeireles @paulvt 

- So far from what I have found debian trixie (gnome 46 / polkit 125) and ubuntu 24.10 (gnome 47 / polkit 124) both have the  same issue of delay in command using pkexec in gnome extension. 
- And I found that the delay is only present when wayland and in Debian and ubuntu. 
- In Xorg mode there is no delay.
- Also Fedora 41 beta (gnome 47 / polkit 125) doesnt have this issue delay in both Xorg or Wayland.

As a test, I ran a simple command `pkexec uname -r` in the extension in ubuntu 24.10 in wayland and there a delay.

Using sysprof I found couple of logs regarding polkit
[image]

I have been comparing both Fedora and Debian and it seems, only difference I have noticed it Fedora has a group `wheel` which ubuntu and debian doest have, I believe.

maniacx, 2024-10-15:
> I am also having issues to launch Warp terminal and seems the issue is related to Wayland as well: [warpdotdev/Warp#5367](https://github.com/warpdotdev/Warp/issues/5367)

I think probably may or may not be related to polkit. However this comment made me laugh. Someone suggest use sudo to open wrap terminal.

[image]

So wrap terminal doesnt open at all or opens with delays takes time?

I have reported this issue to polkit. Let see if they give a response.

danielbmeireles, 2024-10-15:
> So wrap terminal doesnt open at all or opens with delays takes time?

The warp terminal opens every time, but with a huge delay (in my case, more than 5 minutes).

danielbmeireles, 2024-10-23:
Hello! The Warp issue was closed: https://github.com/warpdotdev/Warp/issues/5367

I could confirm the issue was fixed.

maniacx, 2024-10-24:
Is there a commit of what they fixed?
Also I think wrap issue is different from the one we have. The delay in lsof command may or may not be the same reason polkit command is suffering from delays. So I think they are not using lsof command anymore. But a commit will tell me what has been done, but I haven't seen anything submitted yet.

danielbmeireles, 2024-10-25:
I tried to find a commit or something like that, but they didn't publish anything about it.

danielbmeireles, 2024-12-02:
Quick update: I switched distros (from Ubuntu to Fedora) not only because of this issue, but it was something that I considered as a extra point to Fedora.

danielbmeireles, 2024-12-05:
No problem so far; the extension is working like a charm on Fedora.

maniacx, 2024-12-21:
Since you have already moved to Fedora, I will close this issue.
I have created a workaround to use sudo instead of pkexec for the delayed issue in wayland.

If you are still interested in using Ubuntu we can discuss this here

https://github.com/maniacx/Battery-Health-Charging/issues/122

closed 2024-12-21 by maniacx