maniacx/Battery-Health-Charging #112: Problem with kernel 6.11.3

AsusA15, 2024-10-15:
Hi, 
Prior to move to kernel 6.11.3, your extension worked, thank you!
[myKconf-6.11.3.txt](https://github.com/user-attachments/files/17382479/myKconf-6.11.3.txt)

But now i'm baffled. i don't know which options i've selected (maybe as module but not loaded) to break it.
With 6.11.3 (which i'm on), i do have ACAD and BAT1, but no 'charge_control_end_threshold'  
>> ls -ls  /sys/class/power_supply/BAT1 gives:

>> "alarm, capacity_level, charge_full_design, current_now, device, model_name, present, status, technology, uevent, voltage_now, capacity, charge_full, charge_now, cycle_count, manufacturer, power, serial_number, subsystem, type, voltage_min_design"

I've attached my .config.

So
1.What have i make wrongly? 
2.Can you do something for me?

I've spent hours configuring/compiling to get the kernel up and running to my expectations. I know gnu/linux is work in progress. Now i'm attacking to details!
I too am looking into BT batt meter too! (not working but that's for another day)
I own Rog Cetra True Wireless, working.

Regards.

Debian Bookworm 12
Gnome 43.9 / X11
Asus TUF A15 FA507NV

maniacx, 2024-10-15:
I have never had deal with kernel config in linux. but back in the day when, i was dealing with android, i used to compile kernel for the android.  I do not know much about it as I have never had to deal with it in linux.

Like you said, may be your modules are not loaded.
These are the two related to charging threshold as well as some addition function like keyboard backlight etc.
```
CONFIG_SENSORS_ASUS_WMI=m
CONFIG_SENSORS_ASUS_EC=m
```
either manual load the asus wmi module, or compile it with the kernel by changing it to yes.

```
CONFIG_SENSORS_ASUS_WMI=y
CONFIG_SENSORS_ASUS_EC=y
```

AsusA15, 2024-10-16:
ahah!! I'm not finished with it!
 Thanks for pointing me thoses variables. they are set as module, but not loaded! (There are others, like asus_laptop, xyz_asus, asus_xyz, etc.) I'm used to compile as modules to avoid problems. I am/was confident with the kernel to insmod every needed module, but ... not! Linux is still a work in progress, so i will have to investigate/work further. As i've stated, it works with stock kernel Debian Bookworm 6.1.0-25. (But i have hybrid graphics). 
If i have any success, i will report here. 
Again, Thank you for your time! :)

AsusA15, 2024-10-16:
/logoff

maniacx, 2024-10-16:
I think it doesn't matter much, compiling these in as kernel versus seperate modules. The saving in memory and boot time are in terms of kb and millisec. Unless you boot partition and compiling the kernel does not fit the partition, and you want to trim down the size of the kernel by a few MB, then you compile them as module otherwise it is a waste of time and effort. There are some large modules though that it is better to compile them as modules, such as wifi, etc, but most are the once are in kB's. But thats just my opinion. As long as whatever works for you, use it, coz every once usecase is different.

I will close this issue.

closed 2024-10-16 by maniacx