maniacx/Battery-Health-Charging #113: Feature request: display battery status in the extension icon

nexus-adspawn, 2024-10-25:
Right now, with the extension enabled, the status bar looks something like this
[image]

And the battery indicators are taking 50% of all the quick settings and is very redundant. It would be amazing if the battery indicator showed % of the battery that is charged, something like:

[image]
or maybe even this!
[image]

Then we could get rid of the system battery tray/separate percentage altogether and use this wonderful extension!

maniacx, 2024-10-25:
Quite an interesting idea.

But it can get quite complex. But we can discuss about it.

1. First issue is icon is only 16x16. Which would make it quite small. It will probably work well for display scaling factor of 200 - 300%. But on high resolution with 100% scaling, those 2 heart will just be visible as dots. Also the percentage text in the middle of the battery would be quite unreadable. 

2. Theming, some user may apply custom theme. Even some distro come shipped with their own theme such as yaru.
 
3. I would have to create 300 separate icons. For each percentage values and 3 threshold modes. But i think I could use st.drawingarea and to create this, so I do not have to create 300 icons, so I think it is achievable. 

I would suggest you the use the drawing you posted, convert it to svg, rename and replace the icon in extension folder and see, how it looks. If it is readable let me know.

nexus-adspawn, 2024-10-27:
Oh, i didn't know about the 16x16 limit. i thought it would proportionally scale to the horizontal size, as the "shield" icon for package manager is narrower than its width side. 

I have made some icons and screenshots with examples:
[image]

It looks a tad cluttered but the font size is the same as it's used with the "percentage" mode that shows max percentage limit. Also, i didn't scale the hearts down, so they should be at least as readable as they were before.

Also, charging indicator!
[image]

Charging with full capacity mode: 
[image]

Battery almost drained:
[image]

(i couldn't make it red, even though the icon is red, my theme probably just tints everything black/white depending on the color mode, i don't know how they made the original battery gauge icon red when it's drained and green when it's charging)

Here are the icons:
[image]
[image]
[image]
[image]
I also tweaked the battery icon a bit, making it take full canvas space (the original plugin icons were a it shifted from the corners).

However, i understand that the numbers inside the battery can look kind of cluttered (even though they are the same size as the plugin's smaller version of the numbers in the icon mode, they look same cluttered there). In such case, at least the percentual gauge (stepped to some few levels like the gnome original one) + charging indicator without the numbers would be helpful too