maniacx/Battery-Health-Charging #116: BHC doesn't keep setting (ubuntu 24.10 gnome 47 Wayland)

nicolastaffin, 2024-11-08:
Lenovo yoga slim 6, ubuntu 24.10 gnome 47 Wayland

BHC is sometimes there, sometimes not,
widget is sometimes highlighted, sometimes not, 
sometimes on max capacity, sometimes back to max lifespan. 
I can't understand the pattern. 

Also : when i click on max capacity, notifies a "failed setting" error then goes back to previous setting.

Not sure how to provide more information, but i'd be happy to if useful.

maniacx, 2024-11-08:
I apologize,  I am aware of the issue. However I cannot solve it.
The issue is that pkexec command when executed from gnome-extension, delays by 120 seconds.
This only affect Ubuntu and debian based systems running wayland and polkit 124 or above.
It works well in Xorg though in debian and Ubuntu system.

I have been trying for week to know, find out what is the issue.
I have reached out to gnome-extension matrix channel, raise and issue with polkit and also ubuntu forums. 
So far nobody has an idea, what is going on.

Fedora wayland or Xorg doesnt have this issue. I only found the issue in Debian and ubuntu and assuming the issue will be present for debian/ubuntu based distro.

https://github.com/polkit-org/polkit/issues/510#issuecomment-2451387828

Thank you for reporting.

Can you trying logging in Xorg once just only to confirm if it works.
And report back to me.

nicolastaffin, 2024-11-09:
Hi @maniacx you don't need to apologize (especially if you can't solve this)
So if i understand it's a delay, and we don't need an immediate battery policy, so despite it's a bit confusing on the feedback side, it should work.
I tested under X11 and it is reacting to my changes yes. However, in x11 also, the control widget is sometimes highlited, sometimes not, and i'm not sure what it  means

What is the difference between 
[image] and 
[image]

It looks like i'm on i'm off but not sure.
changing the policy in the menu, makes the widget go off. And often back to max lifespan policy.

nicolastaffin, 2024-11-09:
If i follow you the workaround for now is
- request what you want
- ignore any widget feedback (text and color) (the notification seems to be OK anyway)
- wait 120 s
- check later if you are good

is it ?

maniacx, 2024-11-10:
Let me have a look if anything else is changed with ubuntu.

maniacx, 2024-11-20:
I have tried everything. I am not sure. This only happens in ubuntu and Debian base system that uses wayland. Unfortunately, I do not know how to proceed further, but just wait till polkit team or ububtu forums respond, but there no response from them either.

nicolastaffin, 2024-11-20:
OK let's wait & see… It more or less works this way, meanwhile. Thanks

petr-fischer, 2024-11-24:
Also, next feedback: after going from the "screen blank/screensaver" state (screen blank activated with the `xdg-screensaver activate`), two `pkexec... CHEKINSTALLATION` were started
[image]

maniacx, 2024-11-25:
That happens when the extension is rebasing. 
When any extension is enable, it will enable and disable other extension. 
That is how gnome-extension works. So when laptop wake up from sleep state extension get repeatadly disable and enabled. earlier it was not noticable as pkexec executes instantly.

But due the the 120s delay in wayland. Now it is noticable. Only way I can avoid it is by using a delay execution of pkexec.
I think, I should work this out on the next release by adding a 1 second delay.

maniacx, 2024-12-21:
@petr-fischer 
@nicolastaffin 

I have created a script to use `sudo`  to be used as a workaround as I cannot solve the bugs with `polkit`. We can continue, our testing and discussion on this thread.

https://github.com/maniacx/Battery-Health-Charging/issues/122

I will close this issue.

closed 2024-12-21 by maniacx