maniacx/Battery-Health-Charging #117: Next release

taha-yassine, 2024-11-23:
Hi,
Sorry this isn't exactly an issue, but I was wondering if a new release is planned in the near future as the last one was in September.
Thanks!

maniacx, 2024-11-24:
Kinda stuck on an issue. 
pkexec (polkit) has a 120 second delay in Ubuntu and Debian based distro's using Wayland. On Xorg there is no issues.

But this is not the extension problem, and all extensions using pkexec and runing on ubuntu/debian with polkit version 104/105 will have this issue on wayland.

It has been a month, that I am trying to see what is the issue but still havent found what is causing this issues or any solution.

I have reported to polkit, ubuntu forum and in gnome-extension matrix, but so far nobody has a clue on this bug.

I am waiting for a response from polkit devs / Ubuntu. But i think I will release the new update in two week, even though this issue is not solved.

There are no feature updates.
But support new devices/configurations: Dell sysfs mode , chromebook sysfs/ectool, and framework_tool.

taha-yassine, 2024-11-25:
Understandable. Unfortunately I can't offer any help with this issue.
I'm interested in the `framework_tool` support that is why I initially asked. Will wait for the upcoming update!

taha-yassine, 2024-12-11:
Hi, I was wondering if there's been any progress on the issue you were facing and if a release is still planned.

maniacx, 2024-12-12:
I will try and release it start of next week. Sorry for the delay

closed 2025-01-06 by taha-yassine