maniacx/Battery-Health-Charging #122: [Ubuntu][Debian][Wayland][Delayed Timeout While Using pkexec]

maniacx, 2024-12-21:
## Bug with Polkit on Ubuntu/Debian running Wayland

- Users have reported a **2.5-minute delay** when setting the threshold using `pkexec`.
- Troubleshooting indicates this issue occurs on **Ubuntu (Wayland)** and **Debian (Wayland)** systems running **polkit version 124 or greater**.
- When the extension runs the `pkexec` command to set the threshold, the execution takes 2.5 minutes.
- This causes the extension to either start after a 2.5-minute delay or take the same time to apply the threshold. (Newer extension versions will display a timeout message after 5 seconds.)
- I have created a script to use sudo instead of pkexec command as a workaround for this issue.

<br>

## Systems Affected by the Delay in pkexec Execution

- ** Known Affected Systems:**
  - Ubuntu-based systems with `polkit >= 124` running Wayland.
  - Debian-based systems with `polkit >= 124` running Wayland.

- **Known Non-Affected Systems:**
  - Ubuntu-based systems with `polkit >= 124` running Xorg.
  - Debian-based systems with `polkit >= 124` running Xorg.
  - Fedora systems (both Wayland and Xorg).

<br>

## Workaround: Use sudo Instead of pkexec

To resolve this issue, replace `pkexec` with `sudo` for commands setting the threshold. This requires creating an exemption to allow `batteryhealthchargingctl` to run with `sudo` without requiring a password. This can be done **manually** or using the provided **bash script**.

<br>

## Workaround Using the Script

### <u>Download script</u>
[bhc_patch.zip](https://github.com/user-attachments/files/18217779/bhc_patch.zip)

### Note
Some systems, like Debian, do not add users to the `sudo` group by default. If prompted, use `su` instead of `sudo` to update the configuration.

### Steps
Run the script using:
```
bash bhc_patch.sh
```
If running the script for the first time, use **Option A** to perform all operations.

### Options Available
- **Option 1:** Install Polkit rules and `batteryhealthchargingctl` (if not already installed).
- **Option 2:** Modify the sudo configuration to allow `batteryhealthchargingctl` to run without a password. This creates the sudo configuration file `/etc/sudoers.d/batteryhealthcharging-{username}`.
- **Option 3:** Patch the Battery Health Charging extension by replacing `pkexec` with `sudo` in `driver.js` and `helper.js`.
- **Option A:** Perform all the above operations.
- **Option 4:** Remove the sudo configuration, deleting the `/etc/sudoers.d/batteryhealthcharging-{username}` file.

<br>

##  Workaround: Do it yourself

If you do no want to run the script, you can edit files yourself.

### Important Notes
- Replace `{USERNAME}` with your actual username in all the following steps below. example instead of 
- If unsure of your username, run the command `whoami` in the terminal.

### 1. Edit Sudoer Configuration
Rather than modifying `/etc/sudoers` directly, create a file in the `/etc/sudoers.d/` directory to leave the original file untouched for easier upgrades.

1. Create a sudo configuration file:
```
   sudo visudo -f /etc/sudoers.d/batteryhealthcharging-{USERNAME}
```

2. Add the following line to the file:
```
   %{USERNAME} ALL=NOPASSWD:/usr/local/bin/batteryhealthchargingctl-{USERNAME}
```
 Replace `{USERNAME}` with your username. 

This will allow `batteryhealthchargingctl` to execute with `sudo` without requiring a password.

### 2. Edit the Battery Health Charging Extension

Replace `pkexec` with `sudo` in the following files:
`~/.local/share/gnome-shell/extensions/Battery-Health-Charging@maniacx.github.com/lib/driver.js`

`~ /home/abc123/.local/share/gnome-shell/extensions/Battery-Health-Charging@maniacx.github.com/lib/helper.js`

#### Example: Modifying driver.js

Change this line:
```
const argv = ['pkexec', ctlPath, 'CHECKINSTALLATION', resourceDir, user];
```
To:
```
const argv = ['sudo', ctlPath, 'CHECKINSTALLATION', resourceDir, user];
```

maniacx, 2024-12-21:
@Umang-Chaurasia
@seffs 
@petr-fischer 
@nicolastaffin 
@paulvt 

Kindly test the script that changes the extension to use sudo and install configuration for sudo to run batteryhealthchargingctl without password.
Let me know how it goes.

Thank you.

zekrom-vale, 2026-09-03:
I have been having issues with opening up my laptop after it sleeps.  It quite literally locks up.  Ubuntu 24.04.4 LTS Dell Inc. Latitude 5520 Wayland Linux 7.0.0-30-generic gnome 46.
```
Sep 03 17:36:17 zekrom-Latitude systemd[1]: Starting fprintd.service - Fingerprint Authentication Daemon...
Sep 03 17:36:18 zekrom-Latitude dbus-daemon[1913]: [system] Successfully activated service 'net.reactivated.Fprint'
Sep 03 17:36:18 zekrom-Latitude systemd[1]: Started fprintd.service - Fingerprint Authentication Daemon.
Sep 03 17:36:22 zekrom-Latitude gdm-password][1336955]: gkr-pam: unlocked login keyring
Sep 03 17:36:25 zekrom-Latitude pkexec[1336975]: pam_unix(polkit-1:session): session opened for user root(uid=0) by zekrom(uid=1000)
Sep 03 17:36:25 zekrom-Latitude pkexec[1336975]: zekrom: Executing command [USER=root] [TTY=unknown] [CWD=/home/zekrom] [COMMAND=/usr/local/bin/batteryhealthchargingctl-zekrom CHECKINSTALLATION /home/zekrom/.local/share/gnome-shell/extensions/Battery-Health-Charging@maniacx.github.com/resources zekrom]
Sep 03 17:36:28 zekrom-Latitude gnome-shell[3924]: Launching DING process
Sep 03 17:36:30 zekrom-Latitude gnome-shell[3924]: Battery Health Charging: Exceeded timeout for command: pkexec /usr/local/bin/batteryhealthchargingctl-zekrom CHECKINSTALLATION /home/zekrom/.local/share/gnome-shell/extensions/Battery-Health-Charging@maniacx.github.com/resources zekrom
Sep 03 17:36:33 zekrom-Latitude gnome-shell[3924]: libinput error: event5  - DELL0A21:00 0488:1024 Touchpad: kernel bug: Touch jump detected and discarded.
Sep 03 17:36:33 zekrom-Latitude gnome-shell[3924]: See https://wayland.freedesktop.org/libinput/doc/1.25.0/touchpad-jumping-cursors.html for details
Sep 03 17:36:33 zekrom-Latitude gnome-shell[3924]: libinput error: event5  - DELL0A21:00 0488:1024 Touchpad: kernel bug: Touch jump detected and discarded.
Sep 03 17:36:33 zekrom-Latitude gnome-shell[3924]: See https://wayland.freedesktop.org/libinput/doc/1.25.0/touchpad-jumping-cursors.html for details
Sep 03 17:36:33 zekrom-Latitude gnome-shell[3924]: Launching DING process
Sep 03 17:36:34 zekrom-Latitude gnome-shell[3924]: libinput error: event5  - DELL0A21:00 0488:1024 Touchpad: kernel bug: Touch jump detected and discarded.
Sep 03 17:36:34 zekrom-Latitude gnome-shell[3924]: See https://wayland.freedesktop.org/libinput/doc/1.25.0/touchpad-jumping-cursors.html for details
Sep 03 17:36:34 zekrom-Latitude gnome-shell[3924]: libinput error: event5  - DELL0A21:00 0488:1024 Touchpad: WARNING: log rate limit exceeded (5 msgs per 24h). Discarding future messages.
Sep 03 17:36:37 zekrom-Latitude gnome-shell[3924]: Launching DING process
Sep 03 17:36:41 zekrom-Latitude gnome-shell[3924]: Launching DING process
Sep 03 17:36:42 zekrom-Latitude gnome-shell[3924]: libinput error: client bug: timer event5 hold: scheduled expiry is in the past (-118ms), your system is too slow
Sep 03 17:36:44 zekrom-Latitude systemd-journald[797]: Under memory pressure, flushing caches.
Sep 03 17:36:44 zekrom-Latitude gnome-shell[3924]: libinput error: event5  - DELL0A21:00 0488:1024 Touchpad: client bug: event processing lagging behind by 27ms, your system is too slow
Sep 03 17:36:44 zekrom-Latitude gnome-shell[3924]: libinput error: event5  - DELL0A21:00 0488:1024 Touchpad: client bug: event processing lagging behind by 41ms, your system is too slow
```

maniacx, 2026-09-04:
@zekrom-vale 

Yes That is expected on polkit 126 which i believe debian is still on 126. So you will have to use sudo patch until debian updates to polkit 127.

The bug is solved in polkit 127.

https://github.com/polkit-org/polkit/issues/572

https://github.com/polkit-org/polkit/commit/da4fb4d89f9f9ca840c4c4e0049952833f23926c