maniacx/Battery-Health-Charging #123: Chromebook undefined error, extension not working at all

mAmineChniti, 2024-12-29:
```
The settings of extension Battery-Health-Charging@maniacx.github.com had an error:

ReferenceError: Chromebook is not defined

Stack trace:

@/home/amine/.local/share/gnome-shell/extensions/Battery-Health-Charging@maniacx.github.com/lib/deviceList.js:44:5
@/home/amine/.local/share/gnome-shell/extensions/Battery-Health-Charging@maniacx.github.com/prefs.js:5:20
_init@resource:///org/gnome/Shell/Extensions/js/extensionPrefsDialog.js:23:33
ExtensionPrefsDialog@resource:///org/gnome/Shell/Extensions/js/extensionPrefsDialog.js:10:4
OpenExtensionPrefsAsync@resource:///org/gnome/Shell/Extensions/js/extensionsService.js:124:33
async*LaunchExtensionPrefsAsync@resource:///org/gnome/Shell/Extensions/js/extensionsService.js:110:14
_handleMethodCall@resource:///org/gnome/gjs/modules/core/overrides/Gio.js:368:35
_wrapJSObject/<@resource:///org/gnome/gjs/modules/core/overrides/Gio.js:401:34
run@resource:///org/gnome/Shell/Extensions/js/dbusService.js:186:20
main@resource:///org/gnome/Shell/Extensions/js/main.js:22:13
run@resource:///org/gnome/gjs/modules/script/package.js:206:19
start@resource:///org/gnome/gjs/modules/script/package.js:190:8
@/usr/share/gnome-shell/org.gnome.Shell.Extensions:1:17
```

Lucifer25x, 2024-12-29:
Same here, I'm on Gnome 42.9

midnightdoggo19, 2024-12-29:
[image]

maniacx, 2024-12-30:
Apologize for that. Already pushed corrected version. Thanks for reporting

closed 2024-12-30 by mAmineChniti