maniacx/Battery-Health-Charging #127: Question: For Dell laptops using Kernel 6.12+ Libsmbios and Dell Command Configure are no longer required?

logicito, 2025-01-03:
I am a little confused with the Readme, because it says no dependencies but then the dependencies (Libsmbios and Dell Command Configure) are listed, please could you clarify, thank you

maniacx, 2025-01-04:
In Kernel 6.12 , They have merged patch that will allow changing threshold through sysfs. 
Prior to kernel 6.12 Libsmbios or Dell command configure was needed. 

Since the supported was already added to change threshold using libsmbios (works for older dell laptops) and Dell command configure (works on newer laptop), I kept those options for laptops that run older kernel or incase sysfs doesnt work for some reason.

If running kernel 6.12, test if can change threshold using sysfs.

````
echo 'Custom' | sudo tee /sys/class/power_supply/BAT0/charge_types
echo '60' | sudo tee /sys/class/power_supply/BAT0/charge_control_end_threshold
echo '55' | sudo tee /sys/class/power_supply/BAT0/charge_control_start_threshold
``` 

If it doesnt work, you might have to try it will either one of the package (libsmbios or dell caommand configure) that works on your laptop.

logicito, 2025-01-04:
> In Kernel 6.12 , They have merged patch that will allow changing threshold through sysfs. Prior to kernel 6.12 Libsmbios or Dell command configure was needed.
>
> Since the supported was already added [...]

Thank you for the clarification, I was confused because it suddenly started working without having installed the smbios-utils-python package in Fedora 41, I am using a Dell, it was because I had upgraded to kernel 6.12.7, now I understand, thank you very much, I don't know what I would do without your extension, please keep supporting it.

closed 2025-01-25 by maniacx