maniacx/Battery-Health-Charging #128: Clevo Notebook comaptibility

GOMpt, 2025-01-06:
This is what I have been looking for for the past 2 months. Imagine my disappointment when I was told the extension is not compatible with my machine :-(  I am using Manjaro Gnome, Ver 47, on a Clevo, Notebook NS5x_NS7xPU, Intel i7120-1260P x 16, Kernel 6.6.65. Tried TLP, doesn't work. Tried Upower, doesn't work. Anybody got any Ideas? The objective is either; a) automatically limit charging to between 20% and 80% when plugged into AC
OR
b) give me flags / notifications when 20% or 80% are reached when unplugged or plugged in respectively
with the basic plan to limit the battery charging cycles and thus extend the lifespan of the battery. I have something similar on my Android phone that does the job - https://github.com/vbresan/HealthyBatteryCharging

Any help, suggestions, especially from maniac, would be great.
Thanks

maniacx, 2025-01-09:
1. Your laptop needs to support charging threshold. It is a hardware feature provided by manufacturers. It not a software feature.

How to know if your laptop model has hardware support for setting charging threshold? 

A) you will see an option in bios to set charging threshold or B) you will be provided a software by clevo to set charging threshold.

If you laptop has way to set charging threshold then in bios or windows that means your laptop hardware supports charging threshold.

Next would be linux kernel should support your laptop. If it supported , the  kernel will present a sysfs path to set charging threshold. If these paths are available, then software like TLP or this extension will be able to set charging threshold.

 Most likely clevo is not support natively.

 So, someone has to create a kernel module (driver) to support for clevo.

There are two 3rd party kernel drivers that are around that may or may not be compatible with your laptop.

1. Qc71 board type.

Use this command

```

cat "/sys/devices/virtual/dmi/id/board_name"

```

If output is different board name, or output shows some error, you board type is not **intel-qc71** and this kernel module will not work.

If the output is **intel-qc71**, 

You can install kernel module from here

https://github.com/pobrn/qc71_laptop

After installing if it works

You will see sysfs path to set threshold values.

```

echo 60 > /sys/class/power_supply/BAT0/charge_control_end_threshold

```

2. Tuxedo drivers

https://gitlab.com/tuxedocomputers/development/packages/tuxedo-drivers

Since tuxedo clevo use the barebones this may or may not work for you.

You will have to read and search what is compatible for your laptop.

If you have installed this you should see sysfs path and set charging mode

```

echo 'stationary' | sudo tee /sys/devices/platform/tuxedo_keyboard/charging_profile/charging_profile

```

Again these are 3rd party kernel module, and they might even break your linux system so use it at your own risk. Please do your research before installing

GOMpt, 2025-01-09:
Thanks @maniacx that's really helpful. I think you've confirmed my thoughts. I am not very competent when it comes to these things but after many hours of research, I came to that conclusion. What you've said at least gives me a way to check if my hardware is compatible. I'll look into that in the next 24-48 hours. Your help is much appreciated.

GOMpt, 2025-01-09:
> 1. Your laptop needs to support charging threshold. It is a hardware feature provided by manufacturers. It not a software feature.
>
>
> How to know if your laptop model has hardware support for setting [...]

A thought. Perhaps dumb. Could is be possible that this power management concept I am looking to achieve culd be managed through systemd? or udev?

GOMpt, 2025-01-09:
A thought. Perhaps dumb. Could it be possible that this power management concept I am looking to achieve culd be managed through systemd? or udev?

maniacx, 2025-01-10:
It starts with kernel or a package that reads and writes the embedded controller. (EC values)
This is a small microntroller on a laptop that is dedicated to these kinds of functions. Like fan curves, charging threshold, led lights, etc

The address and values that need to be read and written to do a specific function are reverse engineered. 

https://github.com/pobrn/qc71_laptop/blob/master/ec.h

Once you know the address and value that need to be changed, this is implemented in the kernel or package that can write the EC values.

Once that is achieved, then you can probably you can change it using systemd or udev rules, command line or script.

For asus, lenovo, thinkpad, samsung devs have already made these kernel module and submitted them to mainline kernel, so they work out of the box. On these laptops, you can change the charging threshold simply by writing to sysfs path, which in turn write the equivalent EC values. 

Some laptop developers and made installable packages to achieve this:

Chromebook: ectool
Framework laptop: framework_tool
Dell: Dell command center and libsmbios.

In your case, I would install dmidecode and get information on the board chipset and use that information to search github or gitlab. It could be someone has already made a driver, but nobody knows about it.

Example of how dmi is used to load some features on somemodelsl and disable it on other models

https://gitlab.com/tuxedocomputers/development/packages/tuxedo-drivers/-/blob/main/src/clevo_keyboard.h

[image]

[image]

As for this extension, all it does is provide a GUI and write the value on sysfs by echo values using commandline. Rest is handled by the kernel.

GOMpt, 2025-01-10:
@mainiacx

Thanks for the above and your time and help. Looking through all the above, the output of various things you and other have suggested, I am sadly coming to the conclusion that I don't have QC71 and thus cannot achieve what I was hoping to do. As a last resort, I plan to contact Clevo and see if a BIOS upgrade may help me. Unlikely but my last straw is clinging to the output of dmidecode as below which seems to indicate that ACPI is possible.

# dmidecode 3.6
Getting SMBIOS data from sysfs.
SMBIOS 3.3 present.
53 structures occupying 3406 bytes.
Table at 0x63088000.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
        Vendor: INSYDE Corp.
        Version: 1.07.09
        Release Date: 09/06/2022
        Address: 0xE0000
        Runtime Size: 128 kB
        ROM Size: 16 MB
        Characteristics:
                PCI is supported
                BIOS is upgradeable
                BIOS shadowing is allowed
                Boot from CD is supported
                Selectable boot is supported
                EDD is supported
                8042 keyboard services are supported (int 9h)
                CGA/mono video services are supported (int 10h)
                ACPI is supported
                USB legacy is supported
                BIOS boot specification is supported
                Targeted content distribution is supported