maniacx/Battery-Health-Charging #129: Dell XPS 15 Unexpected Error with New Installation of Fedora 41

xjia-wpi, 2025-01-11:
This is a fresh install of Fedora 41 with kernel 6.12.8 on Dell XPS 15 9570.
After installing this extension from extensions.gnome.org, I installed the polkit from extension settings as prompted, then the error appeared. Please see the screenshot of the error notificaiton and system details below.
[image]
[image]

maniacx, 2025-01-11:
Can you check if sysfs path for charging threshold exist and what are the values they read.

For reading copy paste these commands in terminal an post the output here

```
cat /sys/class/power_supply/BAT0/charge_types
cat /sys/class/power_supply/BAT0/charge_control_end_threshold
cat /sys/class/power_supply/BAT0/charge_control_start_threshold
```

Also try to set threshold custom threshold values using command line

```
echo 'Custom' | sudo tee /sys/class/power_supply/BAT0/charge_types
```

```
echo '60' | sudo tee /sys/class/power_supply/BAT0/charge_control_end_threshold
```

```
echo '55' | sudo tee /sys/class/power_supply/BAT0/charge_control_start_threshold
```

mansoormemon, 2025-03-02:
[image]

I am also getting an error similar to the one above. I am using Dell Insipiron too.

Here is output of the above commands:

```bash
$ cat /sys/class/power_supply/BAT0/charge_types
cat /sys/class/power_supply/BAT0/charge_control_end_threshold
cat /sys/class/power_supply/BAT0/charge_control_start_threshold
Trickle Fast Standard Adaptive Custom
cat: /sys/class/power_supply/BAT0/charge_control_end_threshold: Input/output error
cat: /sys/class/power_supply/BAT0/charge_control_start_threshold: Input/output error
```

maniacx, 2025-03-02:
I guess the new kernel module for dell, isn't working for all dell laptop.
Until somebody reports this, it wont get fixed.
So for now do not use sysfs for now as it seem the kernel is reporting errors.

Till then, I would suggest you try/install libsmbios.
And test it using command line, if you can set and apply 

if libsmbios doesnt work, then try dell command center.

If either one of the package (libsmbios or dell command center) works for you, then disable and renable the extension.
In extension prefs: choose libsmbios or dell command center

Check out the readme file and link to libsmbios or dell command center.
https://maniacx.github.io/Battery-Health-Charging/device-compatibility/dell

Also I am not sure where we can report bugs about kernel for dell, but you can find some email address for the author who submitted the patch
https://lore.kernel.org/lkml/20240720012220.26d62a54@5400/